Sloeber/arduino-eclipse-plugin #694: Library manager not refresh changing private library path/hardware path.

far5893, 2017-03-29:
I update workspace from mars to neon and arduino plugin from v3 to v4(whth toolchain update in project).
My Project did not found libraries.
I set Arduino private paths (library and hardware)
deselect all platforms/boards , but library manager show AVR/Arduino libraries and not ESP8266 libraries.
Compiler not found include.

jantje, 2017-03-29:
Please provide  following screenshots:
File browser showing the content of the private library used below
The arduino preferences where the private library is added
How you see the library manager is not refreshing

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

far5893, 2017-03-30:
[image]
[image]
[image]
[image]

jantje, 2017-03-30:
I think you misunderstood the idea behind the private libraries and hardware. It is a feature the arduino IDE doesn't have.
Also: It is an absolute bad idea to make the arduino manager maintained libraries/hardware available to sloeber.
I think this video should get you up and running https://www.youtube.com/watch?v=HE5iYxv-B-o&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF

closed 2017-03-30 by jantje

far5893, 2017-03-31:
My project link libraries from my folder  of esp8266 platform(downloade with arduino ide) ,
i remove all references  to libraries and your tool automagically reimport them from correct platform 
and now compile, so to pass from old to new all you need is to clean projects and let your plugin to work for you.