Sloeber/arduino-eclipse-plugin #695: Arduino tab missing

stanzlavos, 2017-03-30:
Hi

I followed the instruction and installed the arduino plugin with my eclipse. But unfortunately, the "Ardunio" section doesn't show up in the "Preferences" page. :(

What should I do ?

jantje, 2017-03-30:
Are you in arduino perspective?

stanzlavos, 2017-03-30:
Got it to work. But........ :(

* I created a project from existing folder
* Selected a folder that I was working on
* Code appeared
* I chose to delete it. Checkbox to delete from disk was unchecked
* Everything was fine
* Created a new project -> chose the blink example
* For some reason, the folder I chose earlier showed up as well (was even part of the compilation - throwed errors).
* I again, right clicked and deleted. Checkbox to delete from disk was unchecked
* Guess what. The folder along with its parent folder is missing !!!!

I mean how is that even possible ?????

The folder structure was something like :

Code/Arduino/<Folder under discussion>

And what I had opened in the IDE was <Folder under discussion>.

Now, even the Arduino folder got deleted !!!! I just lost all my work !!!!

closed 2017-04-01 by jantje