Sloeber/arduino-eclipse-plugin #696: Absolute vs Relative pathnames

hreintke, 2017-03-30:
Not a real issue now (I think) but like to understand.

I was cleaning up sloeber environments which resulted from working on the "clean project issue".
Tried to do that with "just copying" the sloeber directory from test/temp location to the one I always use.
Resulting in include files not found issue. Could not fix that with clean project and/or remove release directory and/or eclipse refresh.
This was due to the usage of absolute pathnames in the compile process.
Did not look into the details (yet) but can you give an indication when/where absolute pathnames are used and whether there is a way to circumvent that ?

jantje, 2017-03-30:
When doing things like upgrading copying things around and things do not work any more, the first thing to do is: to goto project properties->arduino see if all is ok. If not fix it and if all is ok select apply.
The reason for full path names has to do with the fact that sloeber needs to "expand" the recipes to do some post processing to make things work. This makes that the "expanded" recipes have full (os dependent) path names.
For more details see https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/Helpers.java#L1146

hreintke, 2017-03-31:
Hi,
Having trouble to get a real clean new sloeber environment.

What I have is : 
c:\userdata\esp8266\libraries -> private libraries
c:\userdata\esp8266\projects -> eclipse workspace with projects in it
c:\temp\sloeber -> sloeber environment, fully working

Then I do : 
pull latest git, mvn clean verify
copy product/... sloeber to c:\utilities
So I have c:\utilities\sloeber as the new environment
When I start that, I use the c:\userdata\esp8266\projects as the workspace 
Then I get the default download of arduino, which I expect from a clean install
In the windows->preferences->Arduino I see I already have the c:\userdata\esp8266\libraries as path defined -> must come from previous installs 
I add the esp8266 as board -> gets downloaded correctly.
When compiling projects I see the use of `"C:\temp\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++"` -> from the previous install

Checked several locations but cannot find where the "previous install" information is stored.
Can you tell me where that is ?

jantje, 2017-03-31:
>In the windows->preferences->Arduino I see I already have the c:\userdata\esp8266\libraries as path defined -> must come from previous installs

I think oomph is the "problem".
I have a love hate relation with oomph.
Check the log and try to see if oomph did something.
Oomph is installed with eclipse (so also with the product) and I didn't find a good way yet to disable/remove it.
I love the installation capabilities of oomph but I hate the recorder.

>When compiling projects I see the use of "C:\temp\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -> from the previous install

After doing :goto project properties->arduino see if all is ok. If not fix it and if all is ok select apply?

hreintke, 2017-03-31:
OK, next steps are done.

For oomph, can you tell me where to find the log and/or how to create that ?

For wrong compiler path. I found f.e. the C:\Userdata\ESP8266\Projects\Car\.settings\org.eclipse.cdt.core.prefs file. 
In there the paths for the compilers etc are stored. I would expect that to be a "sloeber generic definition".
Now the version/location of tools used is/can be different for each of the projects.
When updating sloeber/boards environment you need to go along all projects.

I did not notice this before as you mentioned : `After doing :goto project properties->arduino see if all is ok` Did not "apply" as all was OK. So you need to apply in all cases. Then it updates to the "new compiler paths"

jantje, 2017-03-31:
If you do apply you force all settings to be recalculated. If you click OK settings are only reset when there are changes (like going from Uno to yun).
You can easily see that as clicking ok will not enforce a complete rebuild and a apply does.
I'm trying to find ways to improve the "auto recalculation"

>For oomph, can you tell me where to find the log and/or how to create that ?

I have seen oomph logs in the standard log file. But they will only appear when the new workspace is created. 
Here you can see the settings changed by oomph <userdir>/.eclipse/org.eclipse.oomph.setup/setups/user.setup

hreintke, 2017-04-01:
When you do a `setGlobalValue(KEY_PRIVATE_LIBRARY_PATHS, String.join(File.pathSeparator, folderName));
`
Where/in which location is that value stored ?

jantje, 2017-04-01:
You can find it at [workspaces]\.metadata\.plugins\org.eclipse.core.runtime\.settings
Note that it is the eclipse api that saves these value, I advice not to mess with these files

hreintke, 2017-04-01:
I am not going to mess around with it.
for : 
>In the windows->preferences->Arduino I see I already have the c:\userdata\esp8266\libraries as path defined -> must come from previous installs

I think explains the reason. It is a workspace setting and I used the same workspace in \temp\.. and \utilities\..

Do you also save preferences/info outside the workspace

jantje, 2017-04-01:
everything in windows->preferences->Arduino is eclipse instance level

hreintke, 2017-04-25:
Yes,
I am still looking into this.
Will reopen or start a new when more info/ideas.

closed 2017-04-25 by hreintke