Sloeber/arduino-eclipse-plugin #698: Workspace deleted !!!

stanzlavos, 2017-03-31:
- I created a project from existing folder
- Selected a folder that I was working on
- Code appeared
- I chose to delete it. Checkbox to delete from disk was unchecked
- Everything was fine
- Created a new project -> chose the blink example
- For some reason, the folder I chose earlier showed up as well (was even part of the compilation - threw errors).
- I again, right clicked and deleted. (Checkbox to delete from disk was unchecked I guess - default ?)
- Guess what. The folder along with its parent folder is missing !!!!

The folder structure was something like :

Code/Arduino/Folder_Under_Discussion

And what I had opened in the IDE was Folder_Under_Discussion.

Now, even the Arduino folder got deleted !!!! I just lost all my work !!!! Let's assume it was my mistake somehow. In that case, "Folder_Under_Discussion" gets deleted (as that was the only folder I gave as input to the IDE). How did the parent directory get deleted ?

jantje, 2017-03-31:
First of all: You already reported this in issue #695. There is no need to report issues 2 times.
I assume #695 can be closed.

**This is my answer to your comments/questions:**
>Now, even the Arduino folder got deleted !!!!

I'm really sorry for that. This is not the user expectation we're aiming for.

>I just lost all my work !!!! 

This is why we always advice to use version control and make backups. 
When you start with a new tool or do a major upgrade always make sure you have a spare copy of your data. You and you alone are responsible for your data/hardware.

> Let's assume it was my mistake somehow. 

It is your mistake that you lost your data. You may have requested Sloeber to delete the data. It may be a bug in Sloeber. At this point we simply can not tell. But I'm happy you are willing to accept that :-)
 
>In that case, "Folder_Under_Discussion" gets deleted (as that was the only folder I gave as input to the IDE). How did the parent directory get deleted ?

Deleting "Folder_Under_Discussion" /.. would do so. But I guess your question is: Why did Sloeber delete the parent of a folder I provided to Sloeber. From the top of my head the only places where Sloeber deletes files/folders is in the [eclipse install]/arduinoPlugin folder and the .ino.cpp file in the project root.
There have been reports of issues with the project clean command trying to clean / but we have never been able to reproduce or identify the problem.

> note: I'm using the term Sloeber in 2 ways which is probably causing some confusion. I'm using Sloeber as the "solution" and Sloeber as the "plugin on top of CDT on top of Eclipse IDE on top of Eclipse". I'm only capable of modifying the plugin. So it is important for me but far less important for you.

As to the steps you did.
Only the step "Created a new project -> chose the blink example' is related to Sloeber. All other actions are CDT/eclipse. So I don't think this is a Sloeber issue.  

**The next steps**
If we want to get to the bottom of this issue, I need you to stick around for some time as this is going to take time and effort on your side. If not we can close this issue right now and continue our lives.
I need you because you are the only person having this issue' and willing to invest in it.

These are my first questions for you.
Where is your workspace located? 
Where did you provide "Folder_Under_Discussion" to the IDE?
Is this reproducible? 
How high would you rate your Sloeber knowledge? 
How high would you rate your Eclipse knowledge?
Which OS are you on?
Which version of Sloeber and Eclipse are you using?
Why didn't you install the product?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

stanzlavos, 2017-03-31:
First of all, please don't think that I am being hostile. That was not my intention. Well, I was a little pissed off which resulted in a few extra exclamation marks.... And I totally understand and value the contributions you guys make.

When I started issue #695, it was for something else. And that is the only reason I started a new thread. So yes, #695 can be closed.

And yes, luckily I had a backup of my workspace. But I did loose some work. My major concern was the code in "Folder_Under_Discussion". I was able to restore most of it from the backup.

If it was only "Folder_Under_Discussion" that got deleted, I wouldn't even have raised this issue. I would have assumed that I messed up ! But the parent directory itself got deleted and as a result a few other sub-directories got deleted as well.

I think I know what happened :

* The default workspace location set in Sloeber : "C:\Users\stanzlavos\My Documents\sloeber-workspace". But my code was present at : "D:\Code\Arduino\__FOLDER__" . 
* I used "File -> Open project from File system" to bring that into the project explorer. But then I deleted it (delete from disk was unchecked) and I actually made sure that the code was present in disk.
* After deleting my folder from the Project explorer, I opened the Blink example. And somehow, my folder was present as a sub-directory. I did not notice this initially. I just tried compiling the example, and errors were thrown. I was surprised to see my files in the compilation error. That was when I noticed my folder as one of the sub-directories. I went ahead and deleted it. **This is when it should have happened.** Like I did previously, I right clicked on the directory and chose delete. Since I had done this once before and verified that the files won't get deleted, I might have a blindly clicked OK.
* I was just trying to reproduce the issue. And I noticed that when I choose delete of a sub-directory in the project, it is actually asking if I want to delete the files from disk. :( And i gave OK !!!

Now, how in the world my folder ended up in the blink example project is beyond me. And even if somehow I managed to do it, how the parent directory got deleted is a mystery. 

* I have not been able to reproduce the issue.
* Using Sloeber for the very first time.
* Have used Eclipse in the past. But I haven't worked on it very much.
* Windows 10
* I installed the product first. Could not get it to work and so deleted it. Then I manually installed Eclipse and tried installing the plugin in two ways : 1) using the Eclipse marketplace and 2) using "install new software". But, the Arduino Tab was not showing up. That was when I raised #695.
  So, I uninstalled the plugin. Again, in two ways : 1) from marketplace and 2) installed softwares. I then downloaded the product. And the rest of the story happened on the product.

stanzlavos, 2017-03-31:
And btw, I haven't got the blink example to compile either. It just say no rule to make target.

"make: *** No rule to make target 'C:/Users/stanzlavos/My', needed by 'test.elf'.  Stop."

Don't know why the full path is not shown as well. Is it somehow coz there is a space in the file path ?

jantje, 2017-03-31:
http://eclipse.baeyens.it/faq.shtml#/troubleshooting

jantje, 2017-03-31:
I had a look here at the import project (this is eclipse code not sloeber code and should not be used except for importing sloeber created projects).
Import project only imports "eclipse projects" that means there has to be a .project file in the folder. That should not be the case with standard arduino projects. The fact it worked is strange... are there .project files on your backups?
The open project works a bit different but very similar to the import.
The import does not move files around and the delete actually does a delete of the files in the original location. It should not delete the parent though.
Note that when you do the delete it tells you the exact folder it proposed to delete.

>, I opened the Blink example

How did you open that?

stanzlavos, 2017-04-03:
I opened the example like this : 

File -> New -> Arduino Sketch

and then followed the wizard.

jantje, 2017-04-03:
The wizard makes a copy of the code (unless you clicked make a link)
When you created the blink sketch was that with the same name as the imported project?

> That was when I noticed my folder as one of the sub-directories. I went ahead and deleted it. 

When you delete a project eclipse asks whether you want to delete the contents on disk and you can say no. In this case the project is still on disk (can be imported) but is no longer shown in eclipse.
However when you delete a folder from a project the content is always erased from disk.
So here you actually deleted the files from disk.
Was this the Code/Arduino/Folder_Under_Discussion folder that you deleted or Code/Arduino?

stanzlavos, 2017-04-03:
1) the project name for the blink skwtch should have been "test"
2) "Arduino" folder was never added to the IDE.
3) Still no idea how my folder showed up in the test project. And as I
said, let's say I somehow managed to do it. Even then why the parent folder
got deleted is the question.

jantje, 2017-04-03:
>Even then why the parent folder got deleted is the question.

That is what I'm trying to find out. But I really have no clue.The appearing folder is strange and should not happen. Unless you would have had the same name for the 2 projects.
But still that would not explain why the parent folder got deleted.

stanzlavos, 2017-04-04:
I guess you can close  the ticket for the time being. Maybe, if I see the
issue again, I will raise a new ticket.

jantje, 2017-04-04:
close as "one off".

closed 2017-04-04 by jantje