Sloeber/arduino-eclipse-plugin #699: recipe for target '.ino.cpp.o' failed

unisorsebastian, 2017-03-31:
Hi, I am trying to compile an arduino sketch but it seems that is not able to find the source. 

The message i get :

>
[subdir.mk.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/886444/subdir.mk.txt)

20:10:20 **** Build of configuration Release for project firstBoard ****
"C:\\eArduino\\arduinoPlugin\\tools\\make\\make" all 
subdir.mk:24: recipe for target '.ino.cpp.o' failed
process_begin: CreateProcess(NULL, echo "Building file: ../.ino.cpp", ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [.ino.cpp.o] Error 2
20:10:21 Build Finished (took 297ms)

Any suggestion to fix this?

I am running Windows 10 64bit

Thanks

jantje, 2017-03-31:
Did you install sloeber in "C:\eArduino"? That is the first time I see this.
Are you on the nightly? There have been quite some fixes in this area.

unisorsebastian, 2017-04-03:
Installation location is "C:\eArduino".
I've also tried to install it my PC from my work place, but without admin rights. I get similar result.

> "C:\\eArduino\\arduinoPlugin\\tools\\make\\make" all 
subdir.mk:24: recipe for target '.ino.cpp.o' failed
process_begin: CreateProcess(NULL, echo "Building file: ../.ino.cpp", ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [.ino.cpp.o] Error 2

I have tried also a nightly build from the below location

https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.0.1-SNAPSHOT/io.sloeber.product-4.0.1-20170319.225420-30-io.sloeber.product.executable.win32.win32.x86_64.zip

jantje, 2017-04-03:
It is the echo command that fails on your system.
Not sure why.
You do not need admin rights to install or run Sloeber

unisorsebastian, 2017-04-03:
I've reinstalled in a new location...same problme.
However part of the subdir.mk file looks odd to me.

Location to avr-g++ is not proper formated i think, it uses backslash and forward slash **\\** and **/**
"C:\sloeb**er\\/ard**uinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\variants\standard" -MMD -MP -MF"$(@:%.o=%.d)" -MT"$@" -D__IN_ECLIPSE__=1 -x c++ "$<"  -o  "$@"

>.ino.cpp.o: ../.ino.cpp
	@echo 'Building file: $<'
	@echo 'Starting C++ compile'
	"C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\variants\standard" -MMD -MP -MF"$(@:%.o=%.d)" -MT"$@" -D__IN_ECLIPSE__=1 -x c++ "$<"  -o  "$@"
	@echo 'Finished building: $<'
	@echo ' '

unisorsebastian, 2017-04-03:
I think i understand...so the problem is  **@echo 'Building file: $<'**. After $< signs the location of the file should be appended ?

jantje, 2017-04-03:
yes. As the first file to compile is .ino.cpp you get:
`process_begin: CreateProcess(NULL, echo "Building file: ../.ino.cpp", ...) failed.`
So make tries to run the command
`echo "Building file: ../.ino.cpp`
and that failed

jantje, 2017-04-12:
Have you fixed this issue yet?

unisorsebastian, 2017-04-13:
no, i have not solve it...i will give another try this w-end. If I find a solution i will post it. Thanks

jantje, 2017-04-13:
other people have errors on the echo in the makefile so I was hoping :-)

jantje, 2017-04-25:
If you still havn't fixed and are still interested we could consider some remote debugging

adminvas, 2017-04-27:
same problem here guys. W10 x64, something goes wrong with \/ chars. I would appreciate a fix on this issue when your time allows it. Cheers.

jantje, 2017-04-27:
@adminvas are you on V4 or the nightly? lots of / \ issues have been fixed in the nightly

adminvas, 2017-05-01:
Sorry for my late response. I'm on V4

jantje, 2017-05-01:
I think it is a good idea to try the nightly

pprotschka, 2017-05-20:
Same issue, also on WIN10 64bit. Any solution on this?

jantje, 2017-05-20:
@pprotschka Did you try the nightly?

adminvas, 2017-05-20:
yes I did try it but had similar resullts. I just quit for the moment due to other responsibilities.

jantje, 2017-05-20:
has anyone tried to compile outside eclipse with the generated makefile?
Gpint to the [workspacefolder]/[project name]/release and run
"[eclipse install folder]\arduinoPlugin\tools\make\make" all

jantje, 2017-05-20:
Can you try to add cmd.exe /c in front of the make command in project properties like below?
[image]

jantje, 2017-05-22:
Just wondering, which java do you use?

unisorsebastian, 2017-05-22:
in case it was java8

jantje, 2017-05-22:
oracle? did the cmd.exe /c help?

adminvas, 2017-05-22:
java 8.xx 
Which is the recommended java version ?
I just saw your messages. I will try your recommendations and let you know.

adminvas, 2017-05-22:
good news jante !
I tried your advice for cmd.exe /c and I just managed to comile the test eeprom_write.ino file from the samples.

Thank you very much. I really appreciate it. I the next couple of days I will try to compile a project I created with codeblocks to see what happens.

Thanks again.
Vassilis

adminvas, 2017-05-22:
yep!
All test code so far from the ready examples compiles without problems.
So to my feeling, **w10 + Oracle java 8** runs perfect **with the tip from jantje**.

jantje, 2017-05-22:
I'm glad we found a solution :-)
Oracle java 8 and W10 is what I use as well so I have no clue as to the root cause.
I tried the 32 and 64 bit version and I do not need the cmd /c fix.
Anyways I'm happy to know this workaround

pprotschka, 2017-05-23:
thanks for the tip but my console is showing:

01:22:30 **** Incremental Build of configuration Release for project TEST1 ****
cmd.exe /c "C:\\Users\\XXX" "XXX\\eclipse\\cpp-neon\\eclipse\\/arduinoPlugin/tools/make/make" all 
'C:\Users\XXX' is not recognized as an internal or external command,
operable program or batch file.

[image]

jantje, 2017-05-24:
do not use spaces in the path

jantje, 2017-05-25:
On some windows machines this happens though the root cause has not yet been identified.
Workaround add cmd.exe /c to the build command as shown here https://github.com/Sloeber/arduino-eclipse-plugin/issues/699#issuecomment-302886239

closed 2017-05-25 by jantje

fozc, 2017-07-28:
Hello everyone,
First off all, thanks for the pulling  @jantje.

I downloaded the 64 bits bundle package from http://eclipse.baeyens.it, unzip the C:\slaeber.
After that the IDE downloaded the arduino things. 

But I cant compile any thing. I think I have same issue. I tried to add "cmd.exe /c" but it didnt works.here is my error logs.

[image]

14:37:38 **** Build of configuration Release for project kolp ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\.ino.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  ".ino.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
subdir.mk:24: recipe for target '.ino.cpp.o' failed
make: [.ino.cpp.o] Error 1 (ignored)
'Finished building: ..\.ino.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\CDC.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\CDC.cpp.d" -MT"core\CDC.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\CDC.cpp"  "core\CDC.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
core\core\subdir.mk:98: recipe for target 'core\CDC.cpp.o' failed
make: [core\CDC.cpp.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\CDC.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\HardwareSerial.cpp.d" -MT"core\HardwareSerial.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial.cpp"  "core\HardwareSerial.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
core\core\subdir.mk:105: recipe for target 'core\HardwareSerial.cpp.o' failed
make: [core\HardwareSerial.cpp.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial0.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\HardwareSerial0.cpp.d" -MT"core\HardwareSerial0.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial0.cpp"  "core\HardwareSerial0.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
core\core\subdir.mk:112: recipe for target 'core\HardwareSerial0.cpp.o' failed
make: [core\HardwareSerial0.cpp.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial0.cpp'
' 
...................................
......There are more same things..
...................................

'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\wiring_shift.c'
'Starting C compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\wiring_shift.c.d" -MT"core\wiring_shift.c.o" -D__IN_ECLIPSE__=1 "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\wiring_shift.c"  "core\wiring_shift.c.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
core\core\subdir.mk:266: recipe for target 'core\wiring_shift.c.o' failed
make: [core\wiring_shift.c.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\wiring_shift.c'
' '
'Starting archiver'
C:/cccc/workspace/kolp/Release/arduino.ar  .\core\CDC.cpp.o .\core\HardwareSerial.cpp.o .\core\HardwareSerial0.cpp.o .\core\HardwareSerial1.cpp.o .\core\HardwareSerial2.cpp.o .\core\HardwareSerial3.cpp.o .\core\IPAddress.cpp.o .\core\PluggableUSB.cpp.o .\core\Print.cpp.o .\core\Stream.cpp.o .\core\Tone.cpp.o .\core\USBCore.cpp.o .\core\WInterrupts.c.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\hooks.c.o .\core\main.cpp.o .\core\new.cpp.o .\core\wiring.c.o .\core\wiring_analog.c.o .\core\wiring_digital.c.o .\core\wiring_pulse.S.o .\core\wiring_pulse.c.o .\core\wiring_shift.c.o 
makefile:73: recipe for target 'C:\cccc\workspace\kolp\Release\arduino.ar' failed
process_begin: CreateProcess(NULL, C:/cccc/workspace/kolp/Release/arduino.ar .\core\CDC.cpp.o .\core\HardwareSerial.cpp.o .\core\HardwareSerial0.cpp.o .\core\HardwareSerial1.cpp.o .\core\HardwareSerial2.cpp.o .\core\HardwareSerial3.cpp.o .\core\IPAddress.cpp.o .\core\PluggableUSB.cpp.o .\core\Print.cpp.o .\core\Stream.cpp.o .\core\Tone.cpp.o .\core\USBCore.cpp.o .\core\WInterrupts.c.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\hooks.c.o .\core\main.cpp.o .\core\new.cpp.o .\core\wiring.c.o .\core\wiring_analog.c.o .\core\wiring_digital.c.o .\core\wiring_pulse.S.o .\core\wiring_pulse.c.o .\core\wiring_shift.c.o, ...) failed.
make (e=2): Sistem belirtilen dosyayı bulamıyor.

make: *** [C:\cccc\workspace\kolp\Release\arduino.ar] Error 2

14:37:43 Build Finished (took 4s.365ms)

 "-E" not found in PATH 

What is your suggestions ? Thank you very much.

Mesyanzhin, 2018-01-23:
I have the same error. Adding cmd.exe /c to the build command did not help. Windows 7 64bit.
Message in locale is: The system can not find the specified path.

> 07:45:50 **** Incremental Build of configuration Release for project Test ****
> cmd.exe /c "D:\\sloeber\\/arduinoPlugin/tools/make/make" all 
> 'Building file: ..\sloeber.ino.cpp'
> 'Starting C++ compile'
> "/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/include" "-IC:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include" "-IC:/Users/user/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-ID:/Artem/sloeber-workspace/Test/Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536  -DARDUINO=10802 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DESP8266  -I"C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" -I"C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
> Системе не удается найти указанный путь.
> subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
> make: *** [sloeber.ino.cpp.o] Error 1
> 
> 07:45:52 Build Finished (took 1s.704ms)

uzi18, 2018-01-23:
Had same on linux 32bit on latest nightly.
I remember this path is wrong: /bin/xtensa-lx106-elf-g++

Mesyanzhin, 2018-01-23:
I found xtensa-lx106-elf-g++.exe in the path:
C:\Users\user\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\bin\
How can I add this path to the sloeber so that it finds it?

Mesyanzhin, 2018-01-23:
I found that the build variable list does not contain the variable A.RUNTIME.TOOLS.XTENSA-LX106-ELF-GCC.PATH, and the variable A.COMPILER.PATH is marked in red. 
I added the variable A.RUNTIME.TOOLS.XTENSA-LX106-ELF-GCC.PATH with needed value to the list, after that the A.COMPILER.PATH ceased to be red. However, the project is still not compiled and produces the same error as before.

jantje, 2018-01-23:
You did not install the esp8266 platform properly in sloeber