Sloeber/arduino-eclipse-plugin #700: Uploading sketch may corrupt bootloader

PropagandaP4nd4, 2017-04-03:
I am using the Sloeber plugin for Eclipse Neon on Windows 10. Whenever I upload sketches with varying complexity with Eclipse to my Arduino Zero(Firmware 1.6.10), no error pops up in the console, however my Arduino does not send anything over the Serial port.
When I then burn the bootloader or upload a sketch via the Arduino IDE(1.8.2), it shows an error:
`Error: at91samd21g18.cpu -- clearing lockup after double fault.
Polling target at91samd21g18.cpu failed, trying to reexamine`
Uploading a sketch via Arduino IDE works perfectly, but after uploading via Eclipse(and then Serial doesn't work in either) uploading via Arduino IDE finishes, but Serial still does not work. However, burning the bootloader resolves this, hence my assumption that the bootloader gets corrupted.

In short: Uploading via Arduino IDE works fine, uploading via the Eclipse plugin possibly corrupts the bootloader or something else so that it is impossible to send data over Serial.

jantje, 2017-04-04:
I have a zero (from the beta test) and I run windows 10 and I never noticed this.
Can you provide following info:
The steps to reproduce the problem.
The upload command in Sloeber and arduino IDE
The sloeber version you are running.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

PropagandaP4nd4, 2017-04-05:
1. I start eclipse & select my workspace
2. I create a new Arduino Sketch with following settings: 
- Project name: "Test"
- Platform Folder: E:\...\eclipse\arduinoPlugin\packages\arduino\hardware\samd\1.6.12\cores
- Board: Arduino/Genuino Zero (Programming Port)
- Upload Protocol: Atmel EDBG
- Port: COM3
-Default ino file

3. I insert following code:
`#include "Arduino.h"
void setup() {
	Serial.begin(9600);
}
void loop() {
	Serial.println("Another loop!");
}`

4. I save and click on the "Upload sketch Button"
4.1 Then command-line output appears, which is shown below my post
5. I then switch to the Serial monitor view and click on the plus to add a Serial port listener with following settings: 
-Port: COM3
-Baudrate: 9600
-Set DTR and RST: yes
6. "Connect to Serial portCOM3 at9600" is shown in the Serial monitor, but then nothing.

4.1 The build and upload log:
`Starting upload
using arduino loader

LaunchingE:\\...\eclipse\/arduinoPlugin/packages/arduino/tools/openocd/0.9.0-arduino5-static/bin/openocd.exe -s E:\\...\eclipse\/arduinoPlugin/packages/arduino/tools/openocd/0.9.0-arduino5-static/share/openocd/scripts/ -f E:\\...\eclipse\/arduinoPlugin/packages/arduino/hardware/samd/1.6.12/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; program {{E:\\...\eclipse-workspaces\arduino\Test/Release/Test.elf}} verify reset; shutdown 
Output:
Open On-Chip Debugger 0.9.0-g688b0b6 (2017-01-12-15:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 02.01.0157
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x00000614 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 32768 bytes from file E:\...\eclipse-workspaces\arduino\Test/Release/Test.elf in 3.850952s (8.310 KiB/s)
** Programming Finished **
** Verify Started **
verified 17992 bytes in 1.524276s (11.527 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
/arduinoPlugin/packages/arduino/tools/openocd/0.9.0-arduino5-static/bin/openocd.exe finished
upload done
`
, where "..." is always the same folder

My Sloeber version is 4.0.0.201701171403, installed from the eclipse Marketplace as plugin.

So, when I now upload anything (even the same sketch) via the Arduino IDE, it uploads successfully, but with an error like the one in my first post. Additionally, the freshly uploaded sketch doesnt send anything over the Serial port. When i instead/after that burn the bootloader via the IDE, the boatloader gets burned succassfully with the same error, but then, after uploading any sketch, the Serial port recieves something again.
Tomorrow I'm gonna try to control an LED to see if everything stops working or just the Serial port. Unfortunately, I don't own an Oscilloscope or any other equipment that could help with finding the error.

Thanks for your time, I will look into the details of eclipse, Arduino and microcontrollers in general, but I think I am of no good use.

PropagandaP4nd4, 2017-04-05:
Oh, I forgot...
I am using version 1.6.12 for the SAMD Boards, because v. 1.6.13 apparently has an error built in, which I pointed out. It got fixed rightaway into v. 1.6.14. I can't see this version in eclipse, is this because you/I manually add new versions or have I set anythin up badly?

jantje, 2017-04-07:
Why don't you use the default upload protocol?

jantje, 2017-04-07:
>I can't see this version in eclipse, is this because you/I manually add new versions or have I set anythin up badly?

Normally you should see these after a restart but I also noticed this doesn't work as good as it should.
At this point the best way to make sure you have the latest version is deleting the file [eclipse install]/arduinoPlugin/[the json file to update].json
In your case that would be E:...\eclipse\arduinoPlugin\package_index.json

PropagandaP4nd4, 2017-04-09:
How do I use the default protocol and what did I do in order to not use it? O.o
I'm sorry, I am not very experienced with Sloeber nor internal mechanics of such software.

jantje, 2017-04-09:
You stated
>Upload Protocol: Atmel EDBG

You can change it to default in the project properties->arduino

PropagandaP4nd4, 2017-04-10:
Errrrm, yeah I thought I had to use that since I used Atmel EDBG as programmer in the Arduino IDE, but now, with "Upload protocol = Default", it works, so I was obviously wrong there. Thank you for your time and patience and I'm sorry having bothered you. :/

jantje, 2017-04-10:
I think the Arduino IDE is confusing on behalf of this.
Happy to see you got things going.
Please show your appreciation in hard cash ;-)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-04-10 by jantje