Sloeber/arduino-eclipse-plugin #701: Serial monitor uses 100% of one CPU if there is not device plugged

Altair31, 2017-04-05:
When I have the serial monitor connected to a COM interface that is no more connected, the serial monitor uses a CPU to 100%.

How to reproduce:
Tested with esp8266
- Plug a device
- Start monitor on the associated COM port
- Unplug the device

jantje, 2017-04-05:
I will have  a look

jantje, 2017-04-24:
I tried to reproduce this here and I did not see this happen.
I tested with a wi mos

rlogiacco, 2017-04-25:
@Altair31 I assume you are on Windows because of your _COM port_ definition, but is my assumption right? And which version of Windows is that? The serial communication partially relies on the OS facilities so that might play a role...

Altair31, 2017-04-28:
@rlogiacco Your assumption is correct.
I'm on Windows 7 Home with Eclipse Neon.3

My version is not the latest (but the previous one) as each times I update, I cannot work on my projects (but this should be another ticket).
Can you tell me the name of the plug-in with the monitor if you want the exact version number?

rlogiacco, 2017-05-08:
If I remember correctly the plugin name was already Sloeber in the previous version 😄 

I'm using a Win 10 64 bit with Oracle JDK 1.8, Eclipse Neon 3 and Sloeber V4 with 6 serial lines all streaming data and my CPU is under 10%: disconnecting the lines doesn't produce the CPU rise you are experiencing.

Which model of ESP8266 do you have? I'm asking because most ESP8266 don't have a TTL to USB converter, so rather than focusing on the MCU I would try to focus on the USB chip...

I would suggest to:

1. ensure you are running an updated version of Java
2. compare the CPU usage with and without Eclipse to ensure it's not related to the USB drivers
3. update to Sloeber 4, this is probably not going to solve your issue per se, but it will make our life a lot easier as we don't keep each and every version installed for debugging 😄

jantje, 2017-05-09:
Sounds like you have a really old version of Sloeber. I strongly advice to upgrade as many issues have been fixed and plenty of improvements have been made.
This indeed caused some incompatibilities. 
To upgrade from a old version like this I advice to copy your code to a safe place. install a brand new version (wait for 4.1 to release) and a new workspace. 
Install the boards and libraries you use.
Create new projects and copy your code into these projects.
This is not so much work (If you do not wait for the downloads) and should work all the time with default settings. Well you may have to add some libraries to your projects before they compile.
New in 3.1 https://www.youtube.com/watch?v=MGAyIOC24lU&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=2
V3.0 single install https://www.youtube.com/watch?v=HE5iYxv-B-o&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=6
V3.0 new and noteworthy part 2 https://www.youtube.com/watch?v=x_JKcvqpxq8&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=7
V3.0 new and noteworthy part 3 https://www.youtube.com/watch?v=quT-5SSj-Gg&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=8

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-05-25:
@Altair31 Is it ok to close this issue?

bergmannm, 2017-06-09:
I'm having the same issue with 4.1, which is latest stable version of sloeber. It's quite annoying because high cpu usage caused by sloeber makes my laptop fan to kick into higher gear :-( .

There are two separate issues i was able to detect.

First one is related to serial console. To replicate it, upload any sketch with serial console output, then in serial monitor view connect to serial port and finally disconnect USB cable. This causes one CPU core to work overtime. Here is backtrace showing where CPU time is being spent.

"jssc.SerialNativeInterface.waitEvents(long)"
"jssc.SerialPort.waitEvents()"
"jssc.SerialPort.access$200(SerialPort)"
"jssc.SerialPort$EventThread.run()"

After removing serial port from monitor view CPU usage goes down a bit, but still doesn't go down to zero, even when I don't interact with sloeber UI in any way and it's window is minimized. A little bit of profiling discovered, that sloeber is spending a lot of time doing some network checking . I'm not sure what's the purpose of that ...

"javax.jmdns.impl.NetworkTopologyDiscoveryImpl.getInetAddresses()"
"cc.arduino.packages.discoverers.NetworkChecker.run()"
"java.util.TimerThread.run()"

Altair31, 2017-06-09:
@jantje Sorry I didn't had time to test.
BTW, I'm not able for the moment to install the Eclipse repository for v4.
I'll try again as soon as I had time for my personal projects.

jantje, 2017-06-09:
@bergmannm 
The network checking is for network upload. Sloeber polls from time to time to see whether new devices have appeared.
This is a feature for yun, yun-shield and esp. (esp calls it OTA or over the air)
Thinking about this. The discovery will not work on windows when bonjour is not installed https://support.apple.com/kb/DL999?viewlocale=en_US&locale=en_US.
It might be related to the high CPU usage you see in jssc.

@all
I just experienced #716 again and I need to do some serious upload tests before I can release my lib improvements so I will be in this area "soon" so any info that can lead to a root cause analysis is welcomed.

bergmannm, 2017-06-11:
I've looked at how to fix high CPU usage after USB disconnecting. The issue is caused by unfinished error handling in jssc - see FIXME at line 1113 in https://github.com/scream3r/java-simple-serial-connector/blob/2.8.0/src/java/jssc/SerialPort.java . The only way to fix it is to make a few changes in jssc. I've a bit experience in Java development, so I can easily fix the source code, but I'm not sure how to manage the rest.  JSSC development seems to be no longer active, submitted pull request are not being processed since 2014. What's the correct procedure in such case? Should I fork it, rename the package to avoid any conflicts and somehow try to release it under different name?  
Another way how to handle it would be including modified jSSC source code into Sloeber.

jantje, 2017-06-11:
JSSC seems dead since 2014 indeed. Even Arduino proposed changes which are not accepted.
Not sure how to handle this.
@wimjongman @rlogiacco 
Any input on how to handle this?

wimjongman, 2017-06-12:
I guess forking and fixing it. The code seems simple with a few C++ native
libraries that do not seem all too big.

Altair31, 2017-06-12:
I think that the best idea should be to fork Arduino fork and send them a pull request with your fix

jantje, 2017-06-12:
It is not really a Arduino fork. Cmagly made a pull request.
There are quite some forks and there are some active forks as well.
As far as I can see https://github.com/aploese/spsw has done quite some eclipse related work. Like a pom

bergmannm, 2017-06-12:
I don't know, maybe I'm looking at it the wrong way, but it seems to me, that https://github.com/aploese/spsw  is based on completely different project than jssc. See commit https://github.com/aploese/spsw/commit/77ece77f8c62ba4a44a9eb4b2398f1adc16b8007 where he deleted all jssc codebase and imported different project from  https://svn.code.sf.net/p/atmodem4j/code .

jantje, 2017-06-12:
I didn't look at the content. I only looked at the activity. I do not feel qualified to judge the content :-s

jantje, 2017-07-08:
JSSC is also the root cause of #783 and #252 
Unfortunately I can't reproduce any of these issues.

jantje, 2017-12-11:
Is there still interest in a fix for this issue?
If not please close.

uzi18, 2017-12-12:
@bergmannm what changes do You need to fix usb?

Altair31, 2017-12-12:
Hi,
In fact I had to change my HDD and I'm not able to use Sloeber anymore since V3 is not available anymore and V4 does not works at all.
So now I switched to PlatformIO that has less footprint on my Eclipse IDE.

uzi18, 2017-12-12:
Why V4 does not work for You?

Altair31, 2017-12-12:
I'm now on Ubuntu and my jvm refuse to download the esp8266 build chain due to unknown CA (toolchain is hosted on GitHub).
So I decided to stop fighting...

uzi18, 2017-12-12:
Update ca-certificates...

jantje, 2017-12-12:
@Altair31 As you seem to have used platformio and Sloeber I would be interested in some comparison experiences from you. (Though V3 and V4.2 are seriously different)

>Sloeber anymore since V3 is not available anymore and V4 does not works at all.

V3 is still downloadable from the website.

Altair31, 2017-12-12:
@uzi18 Be sure I tried to solve this problem but without success.
@jantje I'm used to use CMake at work for more standard C++ development and PlatformIO mimics it on some points.
First, it generates a standard C++ projects that can be simple makefile projects (for shell addicts) or Eclipse ones (and may some other IDE)
It also manages board toolchain download.
On the underlying makefile, it generates some targets to either build, clean, upload,... binaries.
As Eclipse Oxygen now includes natively a serial console (which have the same problem this ticket points for Sloeber), I have all I need for my IOT development.

When I said V3 is no more available, I should have told V3 update repository is no more available. I personally prefer to use plugins than a full Eclipse environment.

jantje, 2017-12-12:
The V3 update site should still be available at http://www.baeyens.it/eclipse/V3/. I didn't check but there has never been a intent to remove it.

I don't understand your comments about platformio further than: it fits more with what I do at work and it has this issue as well.

uzi18, 2017-12-12:
In my opinion platformio has got more disk space overhead and don't support
32bit system.

Overtime eclipse/sloeber also allocate lots of memory, indexer after some
time forgets everything and marks files with red cross (not sure icon)

jantje, 2017-12-12:
If I understood correctly platformio does not have the indexer functionality which means:
* less good context sensitive colouring, 
* no drill down, 
* no call trees,
 ....

Altair31, 2017-12-12:
@jantje In fact, PlatformIO is just a project generator for some platforms.
I, personally, choose to use Eclipse as IDE but you can choose between lots of well-known IDE (see http://docs.platformio.org/en/latest/ide.html#ide ).
For the indexer part, it relies on the IDE one. It just add a parser for library detection.

IMHO it fits me my needs than Sloeber.

And @uzi18, we almost are in 2018 so let RIP  the 32 bits OS to the 2000 years 😉 (or I haven't understood what you said)
For the disk overhead, I'll let IDE-less developers (they still exist) compare an IDE-integrated solution with a makefile builder...
Everyone has its point of view and this is not a place to debate about (just to remember, it is a big tracker).
If anyone needs this fix, raise a new ticket as I close this one.

closed 2017-12-12 by Altair31

jantje, 2017-12-12:
Thanks for the input. 
And I you change your mind about platformio ... welcome back