Sloeber/arduino-eclipse-plugin #702: V4 Windows Install Contains Spaces in Default Workspace Name

mrwgx3, 2017-04-05:
### Basic Infos

#### Hardware

```
Hardware:     ESP12, Adafruit Huzzah Breakout and Feather Boards  
Core Version: 2.3.0
Computer:     Dell, Intel Core i7 CPU, 870 @ 2.93Ghz,  16.0 GB RAM
```

#### OS, IDE

``` 
OS:   Windows 10 Professional and Linux Ubuntu 16.04LTS   
IDE:  V4.0_win64.2017-01-17_15-14-48 (stable release)
      io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64 (nightly build)
```

### Description

After downloading and doing the default install of V4 for Windows 10, I found the 'spaces-in-path' issue subsequently described in ['Install Advice' under topic 'Windows Comments'][spaces_in_path]:
```
'spaces Windows users keep in mind that spaces cause problems.
 So do not install the arduino IDE eclipse or Sloeber in "program files". 
 Don't create the workspace in "my documents".'
```
For those of us which jump first and ask questions later, it would be nice if you could change the default workspace path in future builds such that they don't contain spaces (the workspace shown for my install was 'C:\Users\DELL\My Documents\sloeber-workspace')

Thanks for a job well done!

[spaces_in_path]: http://eclipse.baeyens.it/installAdvice.shtml

### Requested Install Information
```
  C/C++ Autotools support	9.2.1.201703062208	org.eclipse.cdt.autotools.feature.group	Eclipse CDT
  C/C++ Common GDB Support	9.2.1.201703062208	org.eclipse.cdt.gdb.feature.group	Eclipse CDT
  C/C++ Development Platform	9.2.1.201703062208	org.eclipse.cdt.platform.feature.group	Eclipse CDT
  C/C++ Development Tools	9.2.1.201703062208	org.eclipse.cdt.feature.group	Eclipse CDT
  C/C++ DSF GDB Debugger Integration	9.2.1.201703062208	org.eclipse.cdt.gnu.dsf.feature.group	Eclipse CDT
  C/C++ GNU Toolchain Build Support	9.2.1.201703062208	org.eclipse.cdt.gnu.build.feature.group	Eclipse CDT
  C/C++ GNU Toolchain Debug Support	9.2.1.201703062208	org.eclipse.cdt.gnu.debug.feature.group	Eclipse CDT
  ECF Core Feature	1.3.0.v20160823-2221	org.eclipse.ecf.core.feature.feature.group	Eclipse.org - ECF
  ECF Core SSL Feature	1.1.0.v20160823-2221	org.eclipse.ecf.core.ssl.feature.feature.group	Eclipse.org - ECF
  ECF Filetransfer Feature	3.13.2.v20160823-2221	org.eclipse.ecf.filetransfer.feature.feature.group	Eclipse.org
  ECF Filetransfer SSL Feature	1.1.0.v20160823-2221	org.eclipse.ecf.filetransfer.ssl.feature.feature.group	Eclipse.org - ECF
  ECF Httpclient4 Filetransfer Provider	3.13.2.v20160823-2221	org.eclipse.ecf.filetransfer.httpclient4.feature.feature.group	Eclipse.org - ECF
  ECF Httpclient4 Filetransfer SSL Provider	1.1.0.v20160823-2221	org.eclipse.ecf.filetransfer.httpclient4.ssl.feature.feature.group	Eclipse.org - ECF
  Eclipse 4 Rich Client Platform	1.5.3.v20170228-0512	org.eclipse.e4.rcp.feature.group	Eclipse.org
  Eclipse Help System	2.2.2.v20170301-0400	org.eclipse.help.feature.group	Eclipse.org
  Eclipse Platform	4.6.3.v20170301-0400	org.eclipse.platform.feature.group	Eclipse.org
  Eclipse RCP	4.6.3.v20170301-0400	org.eclipse.rcp.feature.group	Eclipse.org
  EMF - Eclipse Modeling Framework Core Runtime	2.12.0.v20160420-0247	org.eclipse.emf.ecore.feature.group	Eclipse Modeling Project
  EMF Common	2.12.0.v20160420-0247	org.eclipse.emf.common.feature.group	Eclipse Modeling Project
  Equinox p2, backward compatibility support	1.2.203.v20170131-1444	org.eclipse.equinox.p2.extras.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, Discovery UI support	1.0.401.v20160901-1335	org.eclipse.equinox.p2.discovery.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, headless functionalities	1.3.203.v20170131-1444	org.eclipse.equinox.p2.core.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, minimal support for RCP applications	1.2.203.v20170131-1444	org.eclipse.equinox.p2.rcp.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, Provisioning for IDEs.	2.2.203.v20170131-1444	org.eclipse.equinox.p2.user.ui.feature.group	Eclipse.org - Equinox
  Git integration for Eclipse	4.6.1.201703071140-r	org.eclipse.egit.feature.group	Eclipse EGit
  Java implementation of Git	4.6.1.201703071140-r	org.eclipse.jgit.feature.group	Eclipse JGit
  Marketplace Client	1.5.4.v20170222-1941	org.eclipse.epp.mpc.feature.group	Eclipse Marketplace Client
  Mylyn Commons	3.21.0.v20160707-1856	org.eclipse.mylyn.commons.feature.group	Eclipse Mylyn
  Mylyn Commons Connector: Discovery	3.21.0.v20160729-1739	org.eclipse.mylyn.discovery.feature.group	Eclipse Mylyn
  Mylyn Commons Connector: Monitor	3.21.0.v20160630-1702	org.eclipse.mylyn.monitor.feature.group	Eclipse Mylyn
  Mylyn Commons Identity	1.13.0.v20160630-1702	org.eclipse.mylyn.commons.identity.feature.group	Eclipse Mylyn
  Mylyn Commons Notifications	1.13.0.v20160721-2347	org.eclipse.mylyn.commons.notifications.feature.group	Eclipse Mylyn
  Mylyn Commons Repositories	1.13.0.v20160630-1702	org.eclipse.mylyn.commons.repositories.feature.group	Eclipse Mylyn
  Mylyn Context Connector: Eclipse IDE	3.21.0.v20160912-1820	org.eclipse.mylyn.ide_feature.feature.group	Eclipse Mylyn
  Mylyn Context Connector: Team Support	3.21.0.v20160701-1337	org.eclipse.mylyn.team_feature.feature.group	Eclipse Mylyn
  Mylyn Task List	3.21.0.v20160914-0252	org.eclipse.mylyn_feature.feature.group	Eclipse Mylyn
  Mylyn Task-Focused Interface	3.21.0.v20160815-2336	org.eclipse.mylyn.context_feature.feature.group	Eclipse Mylyn
  Mylyn Tasks Connector: Bugzilla	3.21.0.v20160909-1813	org.eclipse.mylyn.bugzilla_feature.feature.group	Eclipse Mylyn
  Mylyn WikiText	2.10.1.v20161129-1925	org.eclipse.mylyn.wikitext_feature.feature.group	Eclipse Mylyn
  Nebula Oscilloscope Widget	1.2.0.201703302228	org.eclipse.nebula.widgets.oscilloscope.feature.feature.group	Eclipse Nebula
  Remote System Explorer End-User Runtime	3.7.2.201610260947	org.eclipse.rse.feature.group	Eclipse TM Project
  Sloeber	4.0.1.201704012122	io.sloeber.feature.feature.group	Jan Baeyens and Others
  Sloeber, the Eclipse Arduino IDE	4.0.0.201704012122	io.sloeber.product	null
```

jantje, 2017-04-05:
Indeed. Using Windows my documents as workspace is not a great idea

closed 2017-05-25 by jantje

lietu, 2018-01-28:
I don't get it, what part of this is fixed? I just installed Sloeber 4.2 and it still complains about spaces in the installation path. Saying "it's not a great idea" is not a fix, and I can't find anywhere an explanation on what exactly doesn't work and why, and what is required to be solved for the issue to be fixed - fixed here obviously meaning that spaces in paths will NOT cause issues to Sloeber.

jantje, 2018-01-28:
It is not a great idea to not support spaces (forced due to make which is out of sloeber control) and then have the default workspace contain a space.
So the thing fixed here is the default workspace no longer contains spaces in windows

lietu, 2018-01-28:
> So the thing fixed here is the default workspace no longer contains spaces in windows

That is at best a workaround, it makes it run, with limitations. A fix would be to make spaces work.

Ok, so you're saying `make`, specifically in Windows, does not support spaces in filenames? Correct me if I'm wrong, but isn't `make` generally considered an ancient blast from the past even in C programming world? Wouldn't something like CMake be better, and could it work for this kind of projects?

jantje, 2018-01-28:
You start sounding like a troll
Make on any os has space issues.
Make is a old and tried program still used in C/C++ world
CDT the basis of Sloeber is working on cmake support. Even if cmake support is there I will have to modify a lot of stuff to make Sloeber work with cmake. With the current support very unlikely it will happen.

>That is at best a workaround,

No. Read the title. That one is fixed.

lietu, 2018-01-28:
> You start sounding like a troll

I really don't care what you think I sound like

> Make on any os has space issues.

You're saying this is a problem that has nothing to do with the OS in question, but you only focus on Windows with the "fix" and all other comments/workarounds?

--- 

> Windows Install Contains Spaces

> 'spaces Windows users keep in mind that spaces cause problems.
> So do not install the arduino IDE eclipse or Sloeber in "program files". 
> Don't create the workspace in "my documents".'

> Using Windows my documents as workspace is not a great idea

> So the thing fixed here is the default workspace no longer contains spaces in windows

--- 

> Make is a old and tried program still used in C/C++ world

"old and tried" and "still used" are not good things - make is pretty awful, and if it truly has universal issues with spaces in filenames that's just one more nail in the coffin. The C/C++ world isn't known for being progressive, or caring about the developer experience, or most things in general, so I understand why ancient remnants like that stick around. That does not mean they are good tools, and that you should not look into alternatives.

In case you are unaware of why the `make` tool is flawed, I would suggest you read up on e.g. http://www.conifersystems.com/whitepapers/gnu-make/

> CDT the basis of Sloeber is working on cmake support. Even if cmake support is there I will have to modify a lot of stuff to make Sloeber work with cmake. With the current support very unlikely it will happen.

Right. So basically this boils down to

1. CDT only supports Make
2. Make has issues with spaces

Considering Eclipse is also not known for being progressive, I'm not holding high hopes for them focusing effort on CMake, SCons, waf, or similar more modern options, so I guess that means Sloeber will have issues with spaces on all OSes for the foreseeable future.

The page mentioned in the description ( http://eclipse.baeyens.it/installAdvice.shtml ) however currently mentions nothing about spaces. Under some very poorly designed button you find some mention of spaces under the "All os comments" or "Windows comments" -options. You should seriously reconsider redesigning that page, and putting critical information like that on the download page, just above the download buttons. Here's my design suggestion:

[image]

Anyway, I don't think there's anything more to talk about here for me.

uzi18, 2018-01-28:
@jentje what about cmake4eclipse from marketplace?

jantje, 2018-01-28:
@uzi18 
as stated above
>CDT the basis of Sloeber is working on cmake support. Even if cmake support is there I will have to modify a lot of stuff to make Sloeber work with cmake. With the current support very unlikely it will happen.

Having users that state
>I really don't care what you think I sound like

Is not helping.