Sloeber/arduino-eclipse-plugin #703: Add support for BBC micro:bit board

mjmeijer, 2017-04-07:
Sloeber can provide tools for all open hardware boards, and I think the micro:bit is one that is well supported in education, but only offers an online tool set using mbed architecture.

All about micro:bit http://microbit.org
Getting Started With C/C++ On The Micro:bit http://bit.ly/2oMGuvp

jantje, 2017-04-07:
Feel free.
If you want to make it a arduino board the core will be the hardest.

mjmeijer, 2017-04-09:
More info on using mbed with Eclipse here:
https://developer.mbed.org/cookbook/eclipse-for-building-and-debugging

jantje, 2018-03-06:
Just tested the board explicitly and it works fine.
The json is also part of the buildandcompile test which tested the board under linux and windows.
Waiting for a mac tester ;-)

closed 2018-03-06 by jantje

liebman, 2018-03-06:
I should have one by this weekend, I'll give it a try.

jantje, 2018-03-06:
:-) then the upload can be tested as well :-)

liebman, 2018-03-13:
confirmed: Both compilation and upload work from MacOS

jantje, 2018-03-13:
@liebman thanks :-)