Sloeber/arduino-eclipse-plugin #704: Serial port needs to be constantly added - every single upload!

andrew-davie, 2017-04-07:
Great plugin. Thanks. I have just got my project compiled/running and I think I have found an issue.

I am using an Arduino Micro, which I understand has a software serial port that is removed and re-added every time you reset the device (or it is reset by a download).  That's an issue for Sloeber because it means the serial port changes every compile. I have to manually go to the serial window view and click the add button and add the new port before my program will run.

Why?  Because I have to wait for the serial port to ACK/be ready before printing to it. So it totally kills Sloeber's usability if I'm printing to the serial debug from the Arduino. Hope this can be fixed or there's a workaround because otherwise Sloeber is fantastic.

jantje, 2017-04-07:
I'm sorry I can't follow here.
Which os are you using?
I assume your arduino sketch waits for the serial connection before it continues.

andrew-davie, 2017-04-08:
Here is a minimum code example; the BLINK (which, by the way, you have to add the #include <Arduino.h> for it to compile)....

All I have added here is a Serial.begin(9600) and then a Serial.println("Blink") in the loop.
Now if you attach the serial port in the "serial monitor view" you would expect to see "Blink" printed every few seconds.  You do NOT see this.  Because every time you go through the compile/download cycle, the serial port is removed and re-added (this is how the Micro works).  But sloeber doesn't get this and so there's no output.  What you have to do is compile/download and THEN go to "serial monitor view" and add the serial port AGAIN.  Every single time.

This does not happen in the Arduino IDE;  it survives the Micro removing/adding the serial port, I assume by some sort of polling or checking.

So this makes sloeber problematic if, as instructed, you yield/wait for the serial port communications. It means that you can't setup the project and compile/download/test/compile/download/test -  there's now a new step in there to manually add the serial port back in the view. Every. Single. Time.

I hope that's clearer!

`/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the Uno and
  Leonardo, it is attached to digital pin 13. If you're unsure what
  pin the on-board LED is connected to on your Arduino model, check
  the documentation at http://www.arduino.cc

  This example code is in the public domain.

  modified 8 May 2014
  by Scott Fitzgerald
 */

#include <Arduino.h>

// the setup function runs once when you press reset or power the board
void setup() {

  Serial.begin(9600);

  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second

  Serial.println("Blink");
}
`

andrew-davie, 2017-04-08:
I am using OSX. I do not believe this is an OS issue specifically. It's more to do with the difference between the Arduino Micro (which uses the ATMega32U4 to do the COM in software, rather than having on-board hardware to do it).

uzi18, 2017-04-08:
Did You set serial port in Project properties/Arduino?

andrew-davie, 2017-04-08:
Yes, I set the serial port. It is shown beside the board selection on the project explorer title correctly. This is an issue with the Arduino Micro which detaches and then reattaches the serial port after a download. This causes sloeber to think there is no connection, even though the (reattached) device has the same name. You have to go to the view page and add the connection manually once again. If you can find an Arduino Micro you will see the same behaviour. It's well documented that the Micro does this. I just think sloeber is having issues with the port disappearing and reappearing.

jantje, 2017-04-08:
The yun, leonardo and due have the same behaviour of ports disappearing and reappearing. I have these here and tested with previous versions, they all worked fine. What I'm trying to say is that this is not a missing functionality.

>Because every time you go through the compile/download cycle, the serial port is removed and re-added (this is how the Micro works). But sloeber doesn't get this and so there's no output.
 
Sloeber disconnects the serial connection before the upload and auto reconnects after the upload. The upload output even tells you  this for boards that use a different upload port.
I am not saying that you are not experiencing a problem.  I am saying that Sloeber is catering for this behaviour and has been doing this way before the Arduino IDE did. 
It could be a timing issue related to the micro or a mac issue or a regression in the code.

>you have to add the #include <Arduino.h> for it to compile)....

No you don't. See "Eclipse tells me there are bugs in my code but it compiles ok." here http://eclipse.baeyens.it/faq.shtml#/troubleshooting

>THEN go to "serial monitor view" and add the serial port AGAIN. 

Did you know that when you are not yet connected to the serial port you can simply click the serial port view in the toolbar and it will automatically connect to the serial port with the correct baud rate?

Which version of Sloeber are you running?
Are there errors/warnings in the error log?
Can you make a video demonstrating the issue?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

andrew-davie, 2017-04-08:
Jantje thanks for your analysis and comments.

I have done a video of the workflow with the simple case failure I have described.  Here it is: https://youtu.be/urRtUSqycuM  - As you can see, I go through a number of tests - in all cases I do not get a display on the serial monitor view window UNTIL I manually re-add the port.

Also, you said "No you don't. See "Eclipse tells me there are bugs in my code but it compiles ok." here http://eclipse.baeyens.it/faq.shtml#/troubleshooting" - and the irony is that the very link you pointed me towards says...

"When using ino files and no headers are included include "Arduino.h" to make them go away. This is caused by the solution used to support .ino files."

Since one of the recommended solutions is EXACTLY what I said - I'm kind of feeling that my comment was valid on this one. I do understand the issue and options.

To answer your specific questions:

Sloeber version 4.0.0.201701171403
Lots of "port in use" and "port not found" errors in the log.
Video is at [https://youtu.be/urRtUSqycuM)]

I just wanted to say thank you for your patience and support - I hope what I'm doing is not "stupid user" error - I am an experienced software engineer, not a newbie.

jantje, 2017-04-08:
@andrew-davie don't pick a fight. Check your information and watch out with your language. You are close to a ban on my repositories.
You state:
>you have to add the #include <Arduino.h> for it to **compile**)....

I say: 
>no you don't

The you say
>Also, you said "No you don't. See "Eclipse tells me there are bugs in my code but it **compiles ok**." here http://eclipse.baeyens.it/faq.shtml#/troubleshooting" - and the irony is that the very link you pointed me towards says...

This is basic stuff I do not want to spend time on this.

From your video:
You have 3 connections to the same usb port. You should only have 1. The serial monitor in Sloeber supports multiple connections (Don't start a discussion it doesn't!!!!). Apparently the micro usb driver on your os allows you to connect multiple times. You should only connect once. 
Sloeber only disconnects 1 connection so having multiple will cause issues.

andrew-davie, 2017-04-09:
I don't want to be seen to be picking fights, so sorry about being nit-picky on the include issue. I appreciate the work that's gone into this great tool and have said so several times, and also your responsiveness to this issue.  I see the same issue with only one connection - so it's not user error. I am now thinking it's the arduino that is not presenting a serial port to the computer after a detach/reattach either in sufficient time, or properly. It is pretty full-up with code/RAM usage and the arduino IDE compile does warn "Low memory available, stability problems may occur."  If I find out that this is the issue I'll try to come back here and report.

jantje, 2017-04-09:
ok, I assume the basic blink works with one port open on the serial monitor.
In which case this issue can be closed.

andrew-davie, 2017-04-09:
No, unfortunately the basic blink does not work with for me either with only one port open on the serial monitor. When I first add the port, I see the output "blink" repeating as expected. When I do another compile/download I no longer see any text output. If I then remove the existing port (x button) and then add (+ button) I again see the output text. I have done another video to demonstrate - hopefully as clear as possible.

My comment about RAM being full was for my "real" project I'm trying to port over, not the blink example. I'm sorry I should have made that more clear. Neither work for me (in terms of the serial view)... but to keep things simple I just re-did the blink example to test.

I've done another video trying to keep things as plain and simple as possible, to show the issue...

https://youtu.be/0rQ5UZlNvNw

Here's a step-by-step of what i was doing...
I have just booted started sloeber, with the arduino unplugged, then...
0:00 - 0:10 shows the project has an attached serial port and there is no port selected in serial view.
0:10 - 0:35 I plug in the arduino and do a compile/download.  it succeeds. Arduino LED blinking as expected.
0:35 - 0:50 I connect to a serial port and I start seeing the serial output ("blink")
0:50 - 1:00 I again compile/download. It succeeds.
1:00 - 1:15 i switch to serial view and see that the blink message is no longer being updated, yet the arduino is definitely running the blink program (LED flashing).
1:15 - 1:27 I manually ("X") remove the connection to the serial port.
1:27 - end... I manually reconnect ("+") to the serial port and scroll down (because the view doesn't appear to autoscroll) - I see the blink coming in again every second.

So that's the minimal failure case as I see it.
That is, to see serial output after a compile/download, I have to manually reconnect the port.
I suspect this is the arduino doing something weird, but in any case on my OS/system at least and assuming I haven't misunderstood how things are supposed to work, I think this demonstrates the serial view not working on a recompile/download unless you re-add the port manually.

I don't mind if you close the ticket - hardly worth your while as I'm obviously the only user seeing this issue. I did want to show what's happening, so that I can be corrected if I was doing something stupid.

Thanks for your time.

jantje, 2017-04-09:
As the blink example shows the same issue we should keep this issue open.
From the video I see that things are not working as expected. I also see that the disconnect message is not logged when you press the disconnect button. Which indicates that something went wrong during the disconnect reconnect during upload.

Thinking about this. It seems like the port remains available.
You can see in your video at 0:28 that sloeber reset the micro and it reappears with the same name.
Maybe sloeber connects to fast to the port.
Some actions you could do to research.
1) Check for errors in the error log view.
2) Add a serial print to the setup so we see whether the board gets reset when the serial monitor connects (Sloeber normally does not reset the serial monitor when connecting)
3) experiment with toggling the dtr rst option. I thought default should be off (tested here seems it is not).

Note 
1 clicking the serial monitor button on the toolbar is a quicker way to connect.
2 There is a auto scroll on off button in the serial monitor

andrew-davie, 2017-04-10:
When I use the blink project...
1) Added serial print "setup", but NEVER see this message.  I think this is because the port is not attached when I first run and the board seems never to be reset when connection made. And I am guessing messages are not buffered, or cleared on port open, so only what comes in after the port is opened are shown.
2) Errors are present in the log view. I was simply compiling/upload - just doing that was sufficient to bring up errors "port not found" and "port in use". I saw both.  When a compile/upload failed (and I have just clicked to close the error message window), I still see "blink" coming over serial - indicating the port was not closed and the machine was not reset (because I don't see an intervening connecting message, or the "setup" message).
However the console did show it looping, showing port status and eventually...
PORTS { /dev/cu.usbmodem1421,} / { /dev/cu.usbmodem1421,} => {}
PORTS { /dev/cu.usbmodem1421,} / { /dev/cu.usbmodem1421,} => {}
PORTS { /dev/cu.usbmodem1421,} / { /dev/cu.usbmodem1421,} => {}
Comport is not behaving as expected
Using comport /dev/cu.usbmodem1421 from now onwards
Ending reset

Launching/Users/boo/Downloads/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude -C/Users/boo/Downloads/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino8/etc/avrdude.conf -patmega32u4 -cavr109 -P/dev/cu.usbmodem1421 -b57600 -D -Uflash:w:/Users/boo/Documents/sloeber-workspace/blinkx/Release/blinkx.hex:i 
Output:
avrdude: ser_open(): can't open device "/dev/cu.usbmodem1421": Resource busy

avrdude done.  Thank you.

avrdude finished
upload done

3) I toggled DTR/RST to off, and when it connected I still see the blink message - no change to behaviour on that part. A subsequent compile/upload shows no serial messages until I manually connect the port, as before.

Seems to be slightly different - now I get failures just doing successive compiles/uploads IFF the serial port is open and displaying "blink". Port not found or port busy. Suggests that the port handling code might have a problem?

4) If I have no serial port open, then the compile/upload always seems to work (but of course no serial output until manually selected).
5) This appears to be different behaviour than before, where I could always compile/upload but I never saw the port open.  Now I'm never seeing the port CLOSE.  Bizarre.  The consistency seems to be something to do with serial port handling...?  Yesterday you weren't seeing disconnect messages (but it was disconnecting and not reconnecting). Today it's not actually disconnecting and instead staying connected, failing with "port not found" or "port busy".

I don't quite know what to make of it.
Happy to test a diagnostic version if you want to analyse with some debug info.

Notes:
1. OK thanks I also ACK your first message about this.  I was being explicit about what I was doing in the video but do use the toolbar button for regular testing.
2. TY, now using this.

jantje, 2017-04-10:
Add a delay of about 3000 in the setup before serial begin. This should solve delays in opening closing ports making the message visible.
FYI I always have a delay of about 3000 before serial begin just to fix communication problem in general. It is one of the good uses of delay ;-)

Have you tried serial connection open -> upload-> reset in serial monitor?

>Happy to test a diagnostic version if you want to analyse with some debug info.

I'm considering a "closing the serial monitor connection" and "opening the serial monitor connection message" together with their outcome to the upload console. This doesn't have to be a special debug version because it is interesting for anyone having problems in this area.
But it'll take some time as I need to fix some other stuff I'm currently working on.

jantje, 2017-04-21:
Any update here?

andrew-davie, 2017-04-23:
No, the problem was not solved and I have had no luck at all - original problem still happening. Thank you for the followup/support - I am no longer in need of a fix for this.

jantje, 2017-04-23:
I'll close the issue for now. If you do need a fix later we can reopen the issue.

closed 2017-04-23 by jantje