Sloeber/arduino-eclipse-plugin #705: Long Install Directory Pathname Creates Issues for Compiler and Make (ESP8266)

mrwgx3, 2017-04-07:
### Basic Infos

#### Hardware

```
Hardware:     ESP12, Adafruit Huzzah Breakout and Feather Boards  
Core Version: 2.3.0
Computer:     Dell, Intel Core i7 CPU, 870 @ 2.93Ghz,  16.0 GB RAM
```

#### OS, IDE

``` 
OS:   Windows 10 Professional and Linux Ubuntu 16.04LTS   
IDE:  V4.0_win64.2017-01-17_15-14-48 (stable release)
      io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64 (nightly build)
```

### Description

During my testing while re-discovering the 'spaces-in-path' issue [(see 'Install Advice' under topic 'Windows Comments')][spaces_in_path], I accidentally forgot to rename the unzip directory before performing the install. The    long pathname created issues for both the compiler and 'make', even without spaces present:

[spaces_in_path]: http://eclipse.baeyens.it/installAdvice.shtml

#### Problems Output
```
Description	Resource	Path	Location	Type
fatal error: bits/c++config.h: No such file or directory    ntp_test    line 48, external location: c:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoplugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional	C/C++ Problem
make: *** [.ino.cpp.o] Error 1                              ntp_test    C/C++ Problem
recipe for target '.ino.cpp.o' failed                       subdir.mk   /ntp_test/Release	line 24     C/C++ Problem
```

#### Console Output
```
hr:mn:ss **** Build of configuration Release for project ntp_test ****
"C:\\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\.ino.cpp'
'Starting C++ compile'
"C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-IC:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-IC:\Users\DELL\SloeberWork\Wksj\ntp_test/Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC  -DARDUINO=10609 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_ESP12\"" -DESP8266  -I"C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" -I"C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\variants\adafruit" -I"C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi" -I"C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" -I"C:\Huzzah\Hzsa\libraries\LiquidCrystal_I2CX" -I"C:\Huzzah\Hzsa\libraries\TimeLibX" -I"C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"
In file included from C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27:0,
                 from C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,
                 from C:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,
                 from ..\.ino.cpp:10:
c:\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\arduinoplugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:48:28: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
                            ^
compilation terminated.
subdir.mk:24: recipe for target '.ino.cpp.o' failed
make: *** [.ino.cpp.o] Error 1

```

### Requested Install Information
```
  C/C++ Autotools support	9.2.1.201703062208	org.eclipse.cdt.autotools.feature.group	Eclipse CDT
  C/C++ Common GDB Support	9.2.1.201703062208	org.eclipse.cdt.gdb.feature.group	Eclipse CDT
  C/C++ Development Platform	9.2.1.201703062208	org.eclipse.cdt.platform.feature.group	Eclipse CDT
  C/C++ Development Tools	9.2.1.201703062208	org.eclipse.cdt.feature.group	Eclipse CDT
  C/C++ DSF GDB Debugger Integration	9.2.1.201703062208	org.eclipse.cdt.gnu.dsf.feature.group	Eclipse CDT
  C/C++ GNU Toolchain Build Support	9.2.1.201703062208	org.eclipse.cdt.gnu.build.feature.group	Eclipse CDT
  C/C++ GNU Toolchain Debug Support	9.2.1.201703062208	org.eclipse.cdt.gnu.debug.feature.group	Eclipse CDT
  ECF Core Feature	1.3.0.v20160823-2221	org.eclipse.ecf.core.feature.feature.group	Eclipse.org - ECF
  ECF Core SSL Feature	1.1.0.v20160823-2221	org.eclipse.ecf.core.ssl.feature.feature.group	Eclipse.org - ECF
  ECF Filetransfer Feature	3.13.2.v20160823-2221	org.eclipse.ecf.filetransfer.feature.feature.group	Eclipse.org
  ECF Filetransfer SSL Feature	1.1.0.v20160823-2221	org.eclipse.ecf.filetransfer.ssl.feature.feature.group	Eclipse.org - ECF
  ECF Httpclient4 Filetransfer Provider	3.13.2.v20160823-2221	org.eclipse.ecf.filetransfer.httpclient4.feature.feature.group	Eclipse.org - ECF
  ECF Httpclient4 Filetransfer SSL Provider	1.1.0.v20160823-2221	org.eclipse.ecf.filetransfer.httpclient4.ssl.feature.feature.group	Eclipse.org - ECF
  Eclipse 4 Rich Client Platform	1.5.3.v20170228-0512	org.eclipse.e4.rcp.feature.group	Eclipse.org
  Eclipse Help System	2.2.2.v20170301-0400	org.eclipse.help.feature.group	Eclipse.org
  Eclipse Platform	4.6.3.v20170301-0400	org.eclipse.platform.feature.group	Eclipse.org
[... 3 lines cut ...]
  Equinox p2, backward compatibility support	1.2.203.v20170131-1444	org.eclipse.equinox.p2.extras.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, Discovery UI support	1.0.401.v20160901-1335	org.eclipse.equinox.p2.discovery.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, headless functionalities	1.3.203.v20170131-1444	org.eclipse.equinox.p2.core.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, minimal support for RCP applications	1.2.203.v20170131-1444	org.eclipse.equinox.p2.rcp.feature.feature.group	Eclipse.org - Equinox
  Equinox p2, Provisioning for IDEs.	2.2.203.v20170131-1444	org.eclipse.equinox.p2.user.ui.feature.group	Eclipse.org - Equinox
  Git integration for Eclipse	4.6.1.201703071140-r	org.eclipse.egit.feature.group	Eclipse EGit
  Java implementation of Git	4.6.1.201703071140-r	org.eclipse.jgit.feature.group	Eclipse JGit
  Marketplace Client	1.5.4.v20170222-1941	org.eclipse.epp.mpc.feature.group	Eclipse Marketplace Client
  Mylyn Commons	3.21.0.v20160707-1856	org.eclipse.mylyn.commons.feature.group	Eclipse Mylyn
  Mylyn Commons Connector: Discovery	3.21.0.v20160729-1739	org.eclipse.mylyn.discovery.feature.group	Eclipse Mylyn
  Mylyn Commons Connector: Monitor	3.21.0.v20160630-1702	org.eclipse.mylyn.monitor.feature.group	Eclipse Mylyn
  Mylyn Commons Identity	1.13.0.v20160630-1702	org.eclipse.mylyn.commons.identity.feature.group	Eclipse Mylyn
  Mylyn Commons Notifications	1.13.0.v20160721-2347	org.eclipse.mylyn.commons.notifications.feature.group	Eclipse Mylyn
  Mylyn Commons Repositories	1.13.0.v20160630-1702	org.eclipse.mylyn.commons.repositories.feature.group	Eclipse Mylyn
  Mylyn Context Connector: Eclipse IDE	3.21.0.v20160912-1820	org.eclipse.mylyn.ide_feature.feature.group	Eclipse Mylyn
  Mylyn Context Connector: Team Support	3.21.0.v20160701-1337	org.eclipse.mylyn.team_feature.feature.group	Eclipse Mylyn
  Mylyn Task List	3.21.0.v20160914-0252	org.eclipse.mylyn_feature.feature.group	Eclipse Mylyn
  Mylyn Task-Focused Interface	3.21.0.v20160815-2336	org.eclipse.mylyn.context_feature.feature.group	Eclipse Mylyn
  Mylyn Tasks Connector: Bugzilla	3.21.0.v20160909-1813	org.eclipse.mylyn.bugzilla_feature.feature.group	Eclipse Mylyn
  Mylyn WikiText	2.10.1.v20161129-1925	org.eclipse.mylyn.wikitext_feature.feature.group	Eclipse Mylyn
  Nebula Oscilloscope Widget	1.2.0.201703302228	org.eclipse.nebula.widgets.oscilloscope.feature.feature.group	Eclipse Nebula
  Remote System Explorer End-User Runtime	3.7.2.201610260947	org.eclipse.rse.feature.group	Eclipse TM Project
  Sloeber	4.0.1.201704012122	io.sloeber.feature.feature.group	Jan Baeyens and Others
  Sloeber, the Eclipse Arduino IDE	4.0.0.201704012122	io.sloeber.product	null
```

jantje, 2017-04-07:
A check is done whether the install path is no longer than 100 chars here https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/Activator.java#L222
With the long package naming 100 seems still to long :-(.
I'll reduce it to 40

dpeddi, 2017-04-15:
This fix brakes my installation:
"The path D:/PortableApps/EclipsePortable/App/Eclipse/arduinoPlugin is to long"

I haven't the described issue here

jantje, 2017-04-15:
@dpeddi 
you will get problems if you install and use the xtensa-lx106-elf-gcc stuff.
It is a tough call but or I test and limit everybody to the largest still working size or I don't test and get lots of issues.
Given the patreon support it is actually not really a tough choice.
And after all it is a windows problem

dpeddi, 2017-04-16:
I use that and compiled some project with esp8266 and that setup with
success. Now i get the path to long error. Build still work but upload to
flash not.

Regards

dpeddi, 2017-04-30:
I compared my path with the path of the guy which opened this case:

C:\\io.sloeber.product-4.0.1-20170401.212426-33-win32.win32.x86_64\\arduinoPlugin\\tools\\
D:\\PortableApps\\EclipsePortable\\App\\Eclipse\\arduinoPlugin\\tools\\

I still using Sloeber 4.0 (Release) without any issue.. I think that you could increase the path limit to 72 or 65 depending if we have to consider the \\ as a double char..

jantje, 2017-04-30:
The max path length for the slçoeber installation = max path length allowed by windows - the eclipse install path - "arduinoPlugin" - "packages" - whatever could be in the zip file delivered by a third party.

The last part is out of control of sloeber and can change on a day to day basis. 

Therefore I advice to go for [Drive letter]:\Sloeber and I use a "safe 40"

closed 2017-05-25 by jantje

bill-orange, 2017-06-12:
V4.1 fails to install in Neon 4.6.3 #760

Can you provide explicit instructions on how to make the install work in a Windows 10 environment?

jantje, 2017-06-12:
@bill-orange 
Can you create your own issue and not post unrelated stuff in issues?
Can you read and refer to the already available information like http://eclipse.baeyens.it/installAdvice.shtml and state what you are missing?

bill-orange, 2017-06-12:
I will move on to a different product and leave it at that.

grompes, 2017-06-17:
RUssia
Буквально сегодня скачал заново Eclipse и "The Arduino Eclipse blabla **(Product)** v4" и после перезагрузки сраже же выскакивает эта ошибка =((
Придётся удалить плагин и пользоваться стандартной arduino ide

jantje, 2017-06-17:
@grompes 
Just install Sloeber in the root of your system

greeran, 2017-06-26:
hi
after reading all the comments it still dont understand how to install Sloeber as a plugin on my neon-eclipse
thanks

greeran, 2017-06-26:
i solved it with shortening the path to my eclipse
thanks

jantje, 2017-06-26:
@greeran 
That is the solution. How would you propose to change the description so it is easier to come to this conclusion?

greg-ware, 2017-10-13:
Why don't you just let the path point to anywhere in the file system rather than insist on having it in this arduinoPlugin subdirectory? I used to be able to just point to my existing Arduino installation and things would be ok, this does not seem to be the case anymore?

tyblu, 2017-10-24:
This means that Windows 10 users can't use the Eclipse Marketplace to install Sloeber. Don't really understand why this is closed.

wimjongman, 2017-10-24:
I like @greg-ware 's idea.

jantje, 2017-10-24:
@wimjongman 
I originally implemented the overwrite, quote from my patreon post:

> I added this functionality. You can specify the location on the command line as follows -manager_path:XXXXX where XXXXX is the full path to the different location.

I however fear this got lost as I can't find manager_path in the Sloeber code

jantje, 2017-10-24:
@tyblu 
How do you come to this conclusion?

shutterbug-photo, 2018-01-07:
To avoid the issue with Eclipse-Portable (or any other case where Eclipse is otherwise in a "non-40-safe" folder), simply make a batch file that reads like the following:
```
subst m: e:\dropbox\eclipse-portable
m:
EclipsePortable.exe
rem these last two lines simply remove the subst and are optional
c:
subst m: /d
```
Of course, change the path and drive letter as desired and possibly the name of the .exe file. Then run that batch file to start Eclipse. Works fine and no error msg when uploading sketches (that I've found yet, anyways.)

wimjongman, 2018-01-09:
Nice one. Thanks!

tkamaal, 2018-05-08:
Hi all, 

I ran into same issue on windows 7 platform. I'm using eclipse oxygen. After installation an error message popped up, asking me to look into this issue for a fix. 

But I still don't understand what is the fix. @shutterbug-photo Can you please elaborate on your comment (made on Jan 7) ? @jantje : Kindly request your inputs for a fix. 

Thank you guys!

shutterbug-photo, 2018-05-13:
Open a command prompt as administrator. In there, navigate to the folder where eclipse.exe exists (or where eclipse-portable.exe exists, if using the portable version.) Then type in the following five lines:

subst m:  %cd% > run-eclipse.bat

m: >> run-eclipse.bat

EclipsePortable.exe >> run-eclipse.bat

c: >> run-eclipse.bat

subst m: /d >> run-eclipse.bat

This will result in a batch file called run-eclipse.bat. Then you run that file instead of eclipse.exe. The m: can be changed to any unused drive letter.

shutterbug-photo, 2018-05-13:
Grrrrr... It's dropping the line breaks. I'll redo it later when on my computer (using my phone right now.)

Vargnath, 2019-02-04:
It seems you can solve this issue on Windows 10 since version 1607 by disabling the maximum path length limitation. You'll still receive the error message that the path is too long, but I haven't encountered any strange results yet.

You can either enable the registry key `HKLM\SYSTEM\CurrentControlSet\Control\FileSystem LongPathsEnabled (Type: REG_DWORD)` or set it via Group Policy at `Computer Configuration > Administrative Templates > System > Filesystem > Enable NTFS long paths`.
A reboot is recommended for all applications to apply the new behavior.

Source: https://docs.microsoft.com/en-us/windows/desktop/FileIO/naming-a-file#maximum-path-length-limitation

jantje, 2019-02-04:
Thanks for the info.

MarekBujko, 2020-02-08:
> It seems you can solve this issue on Windows 10 since version 1607 by disabling the maximum path length limitation. You'll still receive the error message that the path is too long, but I haven't [...]

Enabled NTFS long paths on Win 10 1909. 
Eclipse problem: Due to issues with long pathnames on Windows, the plugin installation path must be less than 40 characters.

wimjongman, 2020-02-08:
Thanks Marek.

> Eclipse problem: ...
It is not an Eclipse issue but an issue of the make tooling iirc.

ROVguy, 2022-09-25:
I'm having an issue trying to get Sloeber working and get the error that is discussed here. The difference is that my issue is with the plugin and not the actual .exe file.
I did enable NTFS LongPath and made a batch file but neither works since the error is for the plugin.

"Due to issues with long pathnames on Windows, the Sloeber installation path must be less than 40 characters. 
Your current path: C:/Programs_MY/EclipseArduno_4.4.1_win64.2022-09-05_06-48-31/Sloeber/arduinoPlugin is too long and the plugin will no longer function correctly for all boards.
Please visit issue #705 for details. https://github.com/Sloeber/arduino-eclipse-plugin/issues/705"

wimjongman, 2022-09-25:
Install sloeber in C:\Sloeber or c:\Programs_My\Sloeber

grompes, 2022-09-28:
RU
Сколько лет уже прошло с открытия этой темы обсуждения, а "Sloeber" до сих пор неумеет работать в длинном пути =))

wimjongman, 2022-09-28:
> Сколько лет уже прошло с открытия этой темы обсуждения, а "Sloeber" до сих пор неумеет работать в длинном пути =))

The forum language is English. If you have something to say, do it in English, otherwise don't bother to comment.