Sloeber/arduino-eclipse-plugin #707: Homie framework fails with exception

DavidR1953, 2017-04-08:
Jan - I started this question on Patreon, where you kindly offered to look at the build output.
The Homie framework (https://github.com/marvinroger/homie-esp8266) builds OK in Sloeber but crashes with an exception. Using Sloeber 4.0.0.201701171403
The same code builds and runs with PlatformIO
Here are the two files:
[PlatformIoHomie.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/907581/PlatformIoHomie.txt)

[SloeberHomie.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/907582/SloeberHomie.txt)

and this is what's on the serial monitor:
Exception (28):
epc1=0x4021316d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000226 depc=0x00000000

ctx: cont 
sp: 3fff1390 end: 3fff15c0 offset: 01a0

>>>stack>>>
3fff1530:  3ffef3c0 3ffe86b2 00000003 3ffe86b2  
3fff1540:  3ffef3c0 00000000 00000000 4020b125  
3fff1550:  3fff2d0c 00000000 00000000 4021b508  
3fff1560:  40206fa8 00000000 4021ad5c 40206f14  
3fff1570:  40206fa8 3fff1590 3ffef600 402148bb  
3fff1580:  3fffdad0 00000000 3ffef600 40207205  
3fff1590:  402070dc feefeffe 4021ad84 4021ad70  
3fff15a0:  feefeffe feefeffe 3fff058c 40219e14  
3fff15b0:  feefeffe feefeffe 3fff05a0 40100714  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3ffe9127
~ld

Please let me know if you need more info.

jantje, 2017-04-08:
Scanning the differences I found that sloeber only compiles 1 assembly file and platformio does 2 
cores\esp8266\cont.S (both)
esp8266\setjmp.S (platfrmio only)

DavidR1953, 2017-04-09:
Thanks for looking. The comment in setjmp.S is "Required by PlatformIO unit testing feature" which I'm not using, so I'm guessing that's not the problem.
I do not know how to proceed. Should I close this?

jantje, 2017-04-09:
I'll check whether I see other differences

jantje, 2017-04-12:
I have seen that the linking happens with another executable.

DavidR1953, 2017-04-12:
Could you be more specific 😀

I'm afraid I had to give up on Sloeber/Homie.
I also tried some of the other ESP8266 frameworks - Sonoff/Tasmota,
ESPEasy, all of which work in PlatformIO, but not in Sloeber.

I'm sure it's down to my incompetence, but I cannot seem to progress.

jantje, 2017-04-12:
Snoff/tasmota seems to have a problem with the Arduino IDE varsion provided with sloeber https://github.com/Sloeber/arduino-eclipse-plugin/issues/708#issuecomment-293719342

jantje, 2017-05-29:
@DavidR1953
Did you see the comments on the homie issue?

DavidR1953, 2017-05-30:
@jantje 
Yes I had a look. Where do you think the problem is?

jantje, 2017-05-30:
> sounds like config is not instantiated

jantje, 2018-04-20:
Can this issue be closed?

DavidR1953, 2018-04-20:
I haven't used Sloeber for over a year, but Homie is on GitHub, so you can
give it a try.

Kind regards,

David Read.

jantje, 2018-04-20:
I used it and worked fine so closing

closed 2018-04-20 by jantje