Sloeber/arduino-eclipse-plugin #708: ARDUINO defined as 10609

DavidR1953, 2017-04-11:
Sloeber V4 defines ARDUINO as 10609. This means that code which checks for a later version of the Arduino IDE fails to build. (Specifically Sonoff/Tasmota - https://github.com/arendst/Sonoff-Tasmota)

jantje, 2017-04-12:
What is the latest number in Arduino ide right now?

DavidR1953, 2017-04-12:
I don't actually use the Arduino IDE. The latest version is 1.8.2 according
to their release notes, but Google didn't help to find what the define was.
I think gcc's -dM gives a list of macros. I'll install the IDE and try it,
but it will have to be tomorrow.

jantje, 2017-04-15:
Do you have a update?

DavidR1953, 2017-04-15:
Looks like it's 10802.
Not sure if setting it to that is a long term solution, but it's your
call 🙂

Have you got any further with Homie?
Have you tried any other ESP8266 frameworks and got them working?

I can get several of the single example sketches to run, e.g. ArdunoOTA,
WebServer etc. but no more.

Kind regards,

David Read.

jantje, 2017-04-18:
Homie is not on my todo list.

closed 2017-05-25 by jantje