Sloeber/arduino-eclipse-plugin #709: Do not try to create symlinks on windows OS

jantje, 2017-04-11:
Seems trying to create symlinks makes the installation fail.

jantje, 2017-04-11:
it seems the symlinks are needed.

hreintke, 2017-04-12:
FYI (don't know if it helps you but anyway),

Running on windows 8.1, I am using a symlink in the following way.
I want to run the ESP8266 github master version instead of the 2.3.0.
I install Arduino plugin and add ES8266 2.3.0 support.
I delete the 2.3.0 directory
For that I make a symlink 2.3.0 -> my local git Arduino core workspace

This functions without issues.

jantje, 2017-04-12:
Thanks for the input.
The problem with symlink in windows 10 home edition (?) is that even when you are admin you do not have the rights to so so.
You need to install software to set the rights so that the install in Sloeber will succeed.
Sounds like to much hassle.
I think I'll take a look at how Arduino IDE fixed the issue. 

If your usage is to have the core (and not the toolchain) I would advice the usage of private hardware like I do here https://youtu.be/ZN-rLSyhBN8. This allows you to have 2.3.0 next to the latest and greatest and you can easily switch between the 2 for workarounds or trouble shooting.

closed 2017-05-25 by jantje