Sloeber/arduino-eclipse-plugin #710: V4 Fails: only error in makefile at line "@echo 'Starting combiner"

Mingybear, 2017-04-12:
I know this has come up before but it is making me crazy. I have a simple ESP/Arduino project which compiles with no errors but won't make. In the past I have managed to get it to work by repeated re-indexing, project clean, deleting the makefile, etc., but nothing works anymore.

My project consists of a simple .ino file which calls .cpp files in order to ensure proper type checking, make sure all the declarations are OK, etc..

I have tried downloading the most recent Sloeber nightly, reinstalling, etc., etc..

My current configuration is 
  Sloeber	4.0.1.201704020116	io.sloeber.feature.feature.group	Jan Baeyens and Others

Any help would be appreciated. The code was mostly working until this showed up.

jantje, 2017-04-12:
You provide me very little information to help you out.
If possible make a video of what you are doing and what is going wrong.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Mingybear, 2017-04-12:
Understood though I do not see what the benefit of a video would be.

I have a fresh install of Sloeber into an empty directory call sloeber on a Windows10/64 laptop.

I have these files as a project
[I2CTest.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/918058/I2CTest.zip)

in a workspace called ArduinoWork which is in my documents directory. The project is configured to target WeMos D1 Mini 80 MHz.

When I compile I get no compile errors or warnings but the file will not make.

[image]

The only evident error is in makefile 
[image]

Mingybear, 2017-04-12:
Sorry this is the error in the makefile
[image]

Mingybear, 2017-04-12:
This is the error in the makefile

[image]

Mingybear, 2017-04-13:
Additional information:

1) When I scrape all the code ( .cpp, .h) into a single .ino file it compiles OK. Of course, this makes it horrific to try and debug;

2) I have the Sloeber product as described but I also have Eclipse Neon CDT installed on my system. I do this so I have consistent look/feel when developing code for application on the PC. When I want to work on my ESP/Arduino project I load Sloeber, when I do C development for PC I load Eclipse Neon. The two platforms are not running at the same time.

jantje, 2017-04-13:
I compiled the project here and it fails as well but with a real error in the console:
```
'Starting combiner'
"D:\embedded\V4\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -Wall -Wextra -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LD:\embedded\V4\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib" "-LD:\embedded\V4\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/ld" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -o "D:/workspaces/embedded/issue_710/Release/issue_710.elf" -Wl,--start-group    .\.ino.cpp.o .\I2CDriver.cpp.o .\MyLCD.cpp.o   D:\workspaces\embedded\issue_710\Release\arduino.ar   "D:/workspaces/embedded/issue_710/Release/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group "-LD:/workspaces/embedded/issue_710/Release" D:/workspaces/embedded/issue_710/Release/arduino.ar
.\MyLCD.cpp.o:(.bss.LCDCONTROLCACHE+0x0): multiple definition of `LCDCONTROLCACHE'
.\.ino.cpp.o:(.bss.LCDCONTROLCACHE+0x0): first defined here
.\MyLCD.cpp.o:(.data.LCD_SETUP_ARRAY+0x0): multiple definition of `LCD_SETUP_ARRAY'
.\.ino.cpp.o:(.data.LCD_SETUP_ARRAY+0x0): first defined here
.\MyLCD.cpp.o:(.data.CONTRAST+0x0): multiple definition of `CONTRAST'
.\.ino.cpp.o:(.data.CONTRAST+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
makefile:82: recipe for target 'issue_710.elf' failed
make: *** [issue_710.elf] Error 1
```
This error is caused by having the arduino.ar 2 times in the combine command. 
Don't you see this info in the console output?

Mingybear, 2017-04-13:
I will retry and let you know. I just discovered that if I rename all the .cpp files .ino the problem disappears and was going to post that.

I did not see any error flagged in the error output, nor did any of the files indicate an error. I have never seen a situation in Eclipse where there were errors produced in the console output but otherwise not flagged.

I will try and replicate and get back to you.

jantje, 2017-04-13:
The error is not cause by having multiple archives in the link command but by the definition of variables in the header file.
This problem is a standard C/C++ coding issue as explained here http://stackoverflow.com/questions/36209788/gcc-multiple-definition-of-error

closed 2017-04-13 by jantje

Mingybear, 2017-04-14:
Thanks. I am still rather surprised Eclipse itself didn't flag an error but that is clearly the issue.

Sorry for the bother.

Mingybear, 2017-04-14:
I did have guards on the definitions though. There must be something else I am doing wrong.

jantje, 2017-04-14:
guards alone won't help you. putting everything in 1 file does.
If you want to use a variable in multiple source files you have to declare them in the header file as extern and define them in 1 .cpp file.

Mingybear, 2017-04-14:
Well, thanks again for your help and I'm sorry to bother you - especially since it was my fault.

I'll be sending a donation over the next day or two.

Mingybear, 2017-04-14:
Jantje

 

Is there any way I can send you a one time donation? 

 

I am unfamiliar with Patreon but it seems to only permit monthlies and that is not practical in my situation

jantje, 2017-04-14:
Patreon is a monthly subscription to help me going on this. Patreon is currently my only income.
I prefer patreon. If that is not a option for you send me a pm jan@baeyens.it