Sloeber/arduino-eclipse-plugin #712: Compilation Bug

guillotl, 2017-04-13:
Hi, 
When I try to compile an arduino sketch on the eclipse plugin while it compiles cleanly on the arduino IDE, I have the following error:
-----------------------------------------------------------------------------------------------------------
12:03:45 **** Build of configuration Release for project Nano_Gateway ****
"C:\\Eclipse_Arduino\\eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"C:\Eclipse_Arduino\eclipse\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" -I"C:\Eclipse_Arduino\eclipse\arduinoPlugin\libraries\MySensors\2.1.1" -I"C:\Eclipse_Arduino\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\libraries\SPI" -I"C:\Eclipse_Arduino\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\libraries\SPI\src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"
'Finished building: ../.ino.cpp'
' '
'Building file: C:/Eclipse_Arduino/eclipse/arduinoPlugin/libraries/MySensors/2.1.1/examples_linux/mysgw.cpp'
'Starting C++ compile'
"C:\Eclipse_Arduino\eclipse\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs" -I"C:\Eclipse_Arduino\eclipse\arduinoPlugin\libraries\MySensors\2.1.1" -I"C:\Eclipse_Arduino\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\libraries\SPI" -I"C:\Eclipse_Arduino\eclipse\arduinoPlugin\packages\arduino\hardware\avr\1.6.18\libraries\SPI\src" -MMD -MP -MF"libraries/MySensors/examples_linux/mysgw.cpp.d" -MT"libraries/MySensors/examples_linux/mysgw.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Eclipse_Arduino/eclipse/arduinoPlugin/libraries/MySensors/2.1.1/examples_linux/mysgw.cpp"  -o  "libraries/MySensors/examples_linux/mysgw.cpp.o"
C:/Eclipse_Arduino/eclipse/arduinoPlugin/libraries/MySensors/2.1.1/examples_linux/mysgw.cpp:20:20: fatal error: iostream: No such file or directory
 #include <iostream>
                    ^
compilation terminated.
libraries\MySensors\examples_linux/subdir.mk:18: recipe for target 'libraries/MySensors/examples_linux/mysgw.cpp.o' failed
make: *** [libraries/MySensors/examples_linux/mysgw.cpp.o] Error 1

12:03:49 Build Finished (took 3s.161ms)
-----------------------------------------------------------------------------------------------------------

Thanks

jantje, 2017-04-13:
this is probably a duplicate of #688

jantje, 2017-04-14:
Please reopen if this is not a duplicate

closed 2017-04-14 by jantje