Sloeber/arduino-eclipse-plugin #713: building/run a java project start to build an arduino one

dpeddi, 2017-04-13:
On my eclipse I've multiple project.

sometime i need to develop java project. By building/run a java project I see that eclipse try to build arduino project on same workspace.

A workaround to this i to close all arduino projcets.

It seems like that arduino plugin bind to run event but filtering needs to be improved.

jantje, 2017-04-13:
Just to make sure I understand correctly.
In a workspace you have one or more java projects and one or more sloeber projects
When you build/run the java project all the sloeber projects are build.

dpeddi, 2017-04-14:
Exactly.

jantje, 2017-04-14:
I'll have a look but it is low priority.

jantje, 2017-04-24:
I just tried it here with a plugin project and I could not reproduce the issue.
Are you sure you didn't create project dependecies somewhere?

dpeddi, 2017-04-24:
Please, consider that I've just rolled back to v4.0 stable:

tomcat plugin installed...
I'm on a Jave EE project, the only implicit dependency is the "Server"
dependency
-> switched to perspective "Java EE project" an closed all other perspective
-> run as Apache Tomcat v8 server at localhost

on Eclipse progress bar I can see
c++ indexer...
 [...] arduinoPlugin/tool/make/make all...
and finally eclipse try to start the project

On the console i can see:
22:00:18 **** Incremental Build of configuration Release for project
arduino_olimex_SST_e05_ESP8266_SST_webconf ****
"[...]\\Eclipse\\arduinoPlugin\\tools\\make\\make" all

Regards

jantje, 2017-05-09:
The whole build cycle is standard eclipse/CDT and and always has been so V4 or nightly should not behave differently on what you see as issue.

jantje, 2017-05-25:
Closing this issue as non reproducible. If you upgrade to the latest stable and still have the issue you can reopen it.

closed 2017-05-25 by jantje