Sloeber/arduino-eclipse-plugin #714: Cannot upload to board

klyman, 2017-04-13:
I have issue with uploading sketches to my STM32F103C8T6 based board.

First of all, I have installed generic bootloader and checked with Arduino IDE - and THIS IS WORKING. I am able to upload sketches successfully.

An issue occurs when I try to do the same in Sloeber. All of configuration seems to be fine. I am able to compile sketches, but uploading gives me following output in the console:

```
Starting upload
using arduino loader
Starting reset using DTR toggle process
Toggling DTR
Continuing to use COM5
Ending reset

Launching 
Output:
failed to upload
```

It looks like Sloeber cannot extract correct upload tool. Any ideas why is it happening or how to fix it?

I am using:
Arduino_STM32 https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/e3753df280123427a07d74f5b142d927a5546cf7
Arduino IDE 1.8.2
Sloeber 4.0.0

jantje, 2017-04-13:
Are you using the default uploader?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

klyman, 2017-04-14:
[image]
This is my configuration. I could actually try to debug it myself. Does Sloeber create any logs which could give me a hint where is a problem ?

jantje, 2017-04-14:
The image looks ok. There is the eclipse error log that may contain info but in this case I doubt it.
Seems like the json file you use is not at https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls so I do not have it on my system to have a look.
Can you provide the json url?

jantje, 2017-04-21:
any update here?

jantje, 2017-04-24:
closing due to no response

rlogiacco, 2017-04-25:
If we will ever wish to tackle this I believe the hardware definitions should be available at https://github.com/rogerclarkmelbourne/Arduino_STM32

jantje, 2017-04-25:
https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json is in the unit test but There is no hardware so no upload test.

buaus, 2017-05-13:
I just have exactly the same upload problem as mentioned above today. I tried also to use my ST Link, but unfortunately its failed as well.  Within Ardunio 1.8.1 itself both are working properly.
I downloaded  and using the 64 bit package of http://eclipse.baeyens.it/stable.php?OS=Windows.
I added the URL https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json  into the Third Party URL.
Even with new start the problem keeps  unfortunately the same. 
@jantje Do you have got any advice?
Thanks buau

jantje, 2017-05-13:
@buaus 
can you provide the board you use?

buaus, 2017-05-14:
@jantje 
It is the STM32F103C8
buau
[image]

klyman, 2017-05-14:
@buaus check this response https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/266#issuecomment-300219055

@victorpv has posted a workaround

jantje, 2017-05-14:
If it really boils down to environment variables not being set then
A) the nightly does a far better job
B) opening the project properties and selecting apply and then ok should doe the trick

buaus, 2017-05-14:
@jantje 
Hi,
is the page http://www.baeyens.it/eclipse/download.php currently working properly? I tried to open with firefox and windows edge, but nothing selectable. Or where to download the nightly version?

thanks buau

jantje, 2017-05-14:
that is a very old page that was still on the sever. I deleted it.
Goto http://eclipse.baeyens.it/ and follow the menu

buaus, 2017-05-17:
Hi @jantje ,
many thanks for the advise. I downloaded the nightly io.sloeber.product-4.0.1-20170512.010734-58-win32.win32.x86_64.tar.gz and installed it at my windows 10 64bit, Java 8 131 64bit

This seems better, but I still facing problems.

I create with following conditions a project:
[image]

[image]
The third party url file are installed by the sloeber to 
C:\sloeber\arduinoPlugin\packages\STM32\hardware
but it seems to search for it at
C:\Users\Andrea\Arduino\hardware
Even I copy the files also to this folder I get 
Upload will fail due to missing upload parameters

The default variable is not countable. Failed expression: 
org.eclipse.core.internal.expressions.CountExpression@fc8b68b6

Do you have got any additional advice, or should I use better an other version? 
Thanks buau

jantje, 2017-05-17:
As you can see in the error log it states

> Upload will fail due to missing upload parameters

I assume it has to do with the fact that the upload method needs to be selected.
I'll have a look

jantje, 2017-05-17:
I succeeded in running the upload command but that upload command fails :-(
```
D:\embedded\V4\sloeber\arduinoPlugin\packages\STM32\tools\STM32Tools\2017.1.19\tools\win>D:\embedded\V4\sloeber\arduinoPlugin\packages\STM32\tools\STM32Tools\2017.1.19\tools\win\serial_upload.bat D:\workspaces\embedded\stmtest\Release\stmtest.bin
stm32flash 0.4

http://stm32flash.googlecode.com/

Intel HEX ERROR: System Error
/=\: No such file or directory

D:\embedded\V4\sloeber\arduinoPlugin\packages\STM32\tools\STM32Tools\2017.1.19\tools\win>
```
To make it work I added the following to the programmers.txt

```
serial_upload.name=serial_upload
serial_upload.program.tool=serial_upload
serial_upload.tools.serial_upload.program.pattern={tools.serial_upload.upload.pattern}
```
To make sure it works you need to select serial upload in both the uploader and the options.

I don't know whether there is something wrong in the bat file or the command is wrong. As I don't have this board .....

jantje, 2017-05-17:
I did some more investigations. 
I made a fix to sloeber and you need to add this to your programmers.txt file

serial_upload.name=serial_upload
serial_upload.program.tool=serial_upload
serial_upload.tools.serial_upload.program.pattern={tools.serial_upload.upload.pattern}
serial_upload.tools.serial_upload.upload.altID=no_altID
serial_upload.tools.serial_upload.upload.usbID=no_usbID

After having upgraded (tomorrow) you should get
```
Starting upload
using arduino loader

LaunchingD:\embedded\eclipse\/arduinoPlugin/packages/STM32/tools/STM32Tools/2017.1.19/tools/win/serial_upload.bat COM3 no_altID no_usbID D:\workspaces\runtime-arduino.product\stm/Release/stm.bin 
Output:
stm32flash 0.4

http://stm32flash.googlecode.com/

Using Parser : Raw BINARY
Interface serial_w32: 230400 8E1

Failed to init device.
/arduinoPlugin/packages/STM32/tools/STM32Tools/2017.1.19/tools/win/serial_upload.bat finished
upload done
```
Without the upgrade the com port is missing

buaus, 2017-05-24:
@jantje 
Hi Jan,
I tried your recommondation today.  I updated the version and added the items to the programmers.txt.
I still face similar trouble.   Where the file "programmers.txt " have to be located?
[image]

In my case it is in the file C:\sloeber\arduinoPlugin\packages\STM32\hardware\stm32f1\2017.1.20,
but actually sloeber is still looking at other position. Even I copy the hardware folder to the folder mentioned in the error log it still claims "Upload will fail due to missing upload parameters"
Thanks buau

jantje, 2017-05-24:
The platform.txt must be in the same folder the boards.txt is in and that is selected in project properties->arduino
each time you change a file you need to select project properties->arduino->apply->ok
did you do so?

buaus, 2017-05-25:
@jantje 
Yes I have done like this. As mentioned above currently I have this in two folders.  See below. The content of the files is the original. The programmers.txt file consist your mentioned items only.
[image]
[image]

I tried this morning again the procedure of apply and OK in order to be sure.  At doing this I saw at doing the right click on project I get also the error Problems occurred when invoking code from plug-in: "org.eclipse.ui.navigator". I inserted the copied error log below

> > eclipse.buildId=unknown
> java.version=1.8.0_131
> java.vendor=Oracle Corporation
> BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=de_DE
> Framework arguments:  -perspective io.sloeber.application.perspective
> Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective
> 
> org.eclipse.ui.navigator
> Error
> Thu May 25 08:47:55 CEST 2017
> Problems occurred when invoking code from plug-in: "org.eclipse.ui.navigator".
> 
> org.eclipse.core.runtime.CoreException: The default variable is not countable. Failed expression: 
> org.eclipse.core.internal.expressions.CountExpression@fc8b68b6
> 	at org.eclipse.core.internal.expressions.Expressions.getAsICountable(Expressions.java:256)
> 	at org.eclipse.core.internal.expressions.CountExpression.evaluate(CountExpression.java:103)
> 	at org.eclipse.core.internal.expressions.CompositeExpression.evaluateAnd(CompositeExpression.java:53)
> 	at org.eclipse.core.internal.expressions.EnablementExpression.evaluate(EnablementExpression.java:55)
> 	at org.eclipse.ui.internal.navigator.NavigatorPlugin$Evaluator.run(NavigatorPlugin.java:247)
> 	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
> 	at org.eclipse.ui.internal.navigator.NavigatorPlugin.safeEvaluate(NavigatorPlugin.java:262)
> 	at org.eclipse.ui.internal.navigator.wizards.CommonWizardDescriptor.isEnabledFor(CommonWizardDescriptor.java:138)
> 	at org.eclipse.ui.internal.navigator.wizards.CommonWizardDescriptorManager.getEnabledCommonWizardDescriptors(CommonWizardDescriptorManager.java:152)
> 	at org.eclipse.ui.navigator.WizardActionGroup.setContext(WizardActionGroup.java:195)
> 	at org.eclipse.ui.internal.navigator.resources.actions.NewActionProvider.fillContextMenu(NewActionProvider.java:109)
> 	at org.eclipse.ui.navigator.NavigatorActionService$2.run(NavigatorActionService.java:225)
> 	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
> 	at org.eclipse.ui.navigator.NavigatorActionService.addCommonActionProviderMenu(NavigatorActionService.java:219)
> 	at org.eclipse.ui.navigator.NavigatorActionService.fillContextMenu(NavigatorActionService.java:174)
> 	at org.eclipse.ui.navigator.CommonNavigatorManager.fillContextMenu(CommonNavigatorManager.java:267)
> 	at org.eclipse.ui.navigator.CommonNavigatorManager$3.menuAboutToShow(CommonNavigatorManager.java:283)
> 	at org.eclipse.jface.action.MenuManager.fireAboutToShow(MenuManager.java:339)
> 	at org.eclipse.jface.action.MenuManager.handleAboutToShow(MenuManager.java:470)
> 	at org.eclipse.jface.action.MenuManager.access$1(MenuManager.java:465)
> 	at org.eclipse.jface.action.MenuManager$2.menuShown(MenuManager.java:497)
> 	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:256)
> 	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
> 	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4418)
> 	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1079)
> 	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1103)
> 	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1084)
> 	at org.eclipse.swt.widgets.Control.WM_INITMENUPOPUP(Control.java:5160)
> 	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4828)
> 	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:359)
> 	at org.eclipse.swt.widgets.Decorations.windowProc(Decorations.java:1656)
> 	at org.eclipse.swt.widgets.Shell.windowProc(Shell.java:2199)
> 	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5110)
> 	at org.eclipse.swt.internal.win32.OS.TrackPopupMenu(Native Method)
> 	at org.eclipse.swt.widgets.Menu._setVisible(Menu.java:262)
> 	at org.eclipse.swt.widgets.Display.runPopups(Display.java:4277)
> 	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3818)
> 	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1121)
> 	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
> 	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1022)
> 	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:150)
> 	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:693)
> 	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
> 	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
> 	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
> 	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:138)
> 	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
> 	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
> 	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
> 	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
> 	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
> 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
> 	at java.lang.reflect.Method.invoke(Unknown Source)
> 	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
> 	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
> 	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
> 
 buau

jantje, 2017-05-25:
Did you select uploader serial?

buaus, 2017-05-25:
@jantje yes

jantje, 2017-05-25:
maybe I did some changes in the library improvement thread that affect this. I will take a look

buaus, 2017-06-09:
Hi Jantje,

cool, the serial upload works now for the blue pill libary. I tried also 
to use also my STlink, but this still fails with the log

"Failed to run the Upload recipe
Cannot run program "" (in directory 
"C:\Users\Andrea\Documents\sloeber-workspace"): CreateProcess error=87, 
Falscher Parameter"

I used for this Upload protokoll: Default and Upload method: STLink

buau

buaus, 2017-06-14:
@jantje Hi,
I updated this evening. No unfortunately at serial upload and STlink I get only the message "Starting upload" after building, but nothing additionally happens, so no upload  anymore :-(.
buau

jantje, 2017-06-14:
Lots has changed and upload is not seriously tested on the nightly right now.
I'll try and see what it does here.

jantje, 2017-06-14:
I did some fidling around and now I get

> Starting upload
> using arduino loader
> Starting reset using DTR toggle process
> Toggling DTR
> Continuing to useCOM11
> Ending reset
> 
> 
> LaunchingD:\embedded\eclipse\/arduinoPlugin/packages/STM32/tools/STM32Tools/2017.1.19/tools/win/serial_upload.bat COM11 no_altID no_altID D:\workspaces\runtime-arduino.product\714/Release/714.bin 
> Output:
> stm32flash 0.4
> 
> http://stm32flash.googlecode.com/
> 
> Using Parser : Raw BINARY
> Interface serial_w32: 230400 8E1
> 
> Failed to init device.
> /arduinoPlugin/packages/STM32/tools/STM32Tools/2017.1.19/tools/win/serial_upload.bat finished
> upload done

Does this look good to you?

jantje, 2017-06-14:
I checked in the code that generated the command above There is no need to adapt the programmers as I explained. This is "out of the box"

jantje, 2017-06-15:
Strange I fixed this issue but the nightly seems not to contain the fix

chhartmann, 2017-06-17:
I had the same problem (STM32F103C8-board and ST-Link).
When i configure "STM Nucleo F103RB (STLink)" instead, it works with the STM32F103C8-Board without problems (at least i had no side effects so far).
Maybe this will help to fix the problem or this workaround might help until the problem is fixed.

jantje, 2017-06-17:
With the latest nightly:
Bluepill upload command
[image]

D:\embedded\V4\sloeber\/arduinoPlugin/packages/STM32/tools/STM32Tools/2017.1.19/tools/win/serial_upload.bat com4 no_altID no_altID D:\workspaces\embedded\test/Release/test.bin 
Adafruit wiced upload command
[image]

 D:\embedded\V4\sloeber\/arduinoPlugin/packages/adafruit/hardware/wiced/0.6.2/tools/win32-x86/feather_dfu/feather_dfu.exe -u D:\embedded\V4\sloeber\/arduinoPlugin/packages/adafruit/hardware/wiced/0.6.2/tools/win32-x86/dfu-util/dfu-util-static.exe featherlib_upgrade D:\embedded\V4\sloeber\/arduinoPlugin/packages/adafruit/hardware/wiced/0.6.2/featherlib/featherlib.bin 
[image]
ESP8266
[image]
D:\embedded\V4\sloeber\/arduinoPlugin/packages/esp8266/tools/esptool/0.4.9/esptool.exe -cd nodemcu -cb 115200 -cp com4 -ca 0x00000 -cf D:\workspaces\embedded\test/Release/test.bin 

D:\embedded\V4\sloeber\/arduinoPlugin/packages/STM32/tools/STM32Tools/2017.1.19/tools/win/stlink_upload.bat com4 no_altID no_altID D:\workspaces\embedded\test/Release/test.bin 

This is with the nightly of today.
I think all known upload problems are fixed now. If you think not please be very precise what json file what board what is the expected upload command what is the actual upload command

jantje, 2017-06-17:
f you have a upload problem please reopen and be very precise what json file what board what is the expected upload command what is the actual upload command.
Sloeber supports more than 400 boards I can't guess which board you have a problem with.

closed 2017-06-17 by jantje