Sloeber/arduino-eclipse-plugin #717: Upload Console font is not configurable.

alex-vas, 2017-04-14:
The Upload Console view font is seemingly not using the user preference set at "preferences>colors and fonts>console font" like other console views like C++, Java, SVN, Maven, etc.

[image]

Would be nice if the Upload Console view would respect one of the existing font settings like "Debug>Console font" or have its own one.

Cheers

jantje, 2017-04-14:
I would have expected it to follow the C/C++ build console font but apparently it is not.
Any urgent reason to need this?

alex-vas, 2017-04-16:
No urgency, it's just I like my consoles to be in a small fonts, so it can fit more, and this one stands out now. Thanks!

jantje, 2018-05-16:
Can you give this a test?

alex-vas, 2018-05-16:
Looks good. Thank you!

Three smaller observations:

1. Spelling of the Configuration item should start with capital "U" to be consistent with other settings.
2. The value for the font should inherit from "Terminal Console Font" like other console-like views do.
3. The view should update once the setting is "applied". At the moment setting only effects new created views.

[image]

jantje, 2018-05-17:
I'll do 1 and 2 but I have no clue how to do 3.
The same issue exists for the serial monitor

jantje, 2018-05-17:
As to 1. made all first characters capital
As to 2 Apparently the default font should be "text font"
As to 3... you can have a listener and the listener can change the font but..that would not work with the colours as they are decided on based on the input to the text field.
So the font could change on apply but the colours not. 
This is why I prefer to change nothing on apply
The change should be available tomorrow on the update site.

alex-vas, 2018-05-17:
> This is why I prefer to change nothing on apply
Fair enough. 
Looks good!
Thank you.

jantje, 2018-05-18:
closing as it is fixed in the nightly

closed 2018-05-18 by jantje