Sloeber/arduino-eclipse-plugin #719: V4 with Adafruit Boards Package problems

nonsintetic, 2017-04-18:
I'm using a fresh install of v4 with the Adafruit Boards package and a Adafruit Feather M0 board based on the AT SAM D21g micro.

The **first problem** I've had is the plugin wasn't linking the CMSIS includes properly. I fixed that by adding them by hand in the compile settings. They were listed as `compiler.arm.cmsis.c.flags="-I{runtime.tools.CMSIS-4.5.0.path}/CMSIS/Include/" "-I{runtime.tools.CMSIS-Atmel-1.1.0.path}/CMSIS/Device/ATMEL/"` in the platform.txt file, and they got added AS IS in the commandline `"-I/CMSIS/Include/"` instead of `-I"<maybe full path?>/CMSIS/Include/"` (the quotes were before the -I flag causing them not to be read). 

Now **I'm getting the following error** and I'm not sure how to fix it:

Console:
```
12:22:44 **** Incremental Build of configuration Release for project Blink ****
"E:\\ARDUINO\\Programs\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
Building file: ../.ino.cpp
Starting C++ compile
"E:\ARDUINO\Programs\sloeber\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -Wall -Wextra -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DF_CPU=48000000L -DARDUINO=10609 -DARDUINO_SAMD_FEATHER_M0 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x800B -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0\"" "-I/CMSIS/Include/" "-I/CMSIS/Device/ATMEL/"  -I"E:\ARDUINO\Programs\sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.0.15\cores\arduino" -I"E:\ARDUINO\Programs\sloeber\arduinoPlugin\packages\adafruit\hardware\samd\1.0.15\variants\arduino_zero" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"  -I"C:\Users\Neo\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0\CMSIS\Device\ATMEL" -I"C:\Users\Neo\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\"
/usr/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: syntax error: unexpected end of file
subdir.mk:24: recipe for target '.ino.cpp.o' failed
make: *** [.ino.cpp.o] Error 1

12:22:45 Build Finished (took 502ms)
```

Board json: `https://adafruit.github.io/arduino-board-index/package_adafruit_index.json`

jantje, 2017-04-18:
I tried here with the nightly and that worked fine.

jantje, 2017-04-24:
Did you get it to working?

nonsintetic, 2017-04-24:
no, couldn't make it work so I went with a different solution

jantje, 2017-04-24:
ok I guess I close the issue then?

closed 2017-05-09 by jantje