Sloeber/arduino-eclipse-plugin #720: Working with SAMD/Zero and Atmel ICE

tuxedo0801, 2017-04-19:
When using latest Arduino IDE, I can upload sketches via openocd and my Atmel ICE. The IDE will execute f.i. the following:

```
/home/achristian/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static/bin/openocd -d2 -s /home/achristian/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static/share/openocd/scripts/ -f /home/achristian/.arduino15/packages/arduino/hardware/samd/1.6.14/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; program {{/tmp/arduino_build_795299/Appl_RotoDFF_8BtnLed.ino.bin}} verify reset 0x2000; shutdown 
Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-11:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: JTAG Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 01.21.0076
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000614 msp: 0x20007ffc
** Programming Started **
auto erase enabled
Info : SAMD: partitioning the first flash page into 8 subpages
Info : SAMD MCU: SAMD21G18A (256KB Flash, 32KB RAM)
wrote 40960 bytes from file /tmp/arduino_build_795299/Appl_RotoDFF_8BtnLed.ino.bin in 4.593703s (8.708 KiB/s)
** Programming Finished **
** Verify Started **
verified 28120 bytes in 2.298414s (11.948 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

```

When trying exactly the same (with same sketch) with sloeber, the following command is executed:

```
Starting upload
using arduino loader

Launching/bin/openocd -s /share/openocd/scripts/ -f /home/achristian/.arduino15/packages/arduino/hardware/samd/1.6.14/variants/arduino_zero/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; program {{/home/achristian/Documents/sloeber-workspace/Appl_RotoDFF_8BtnLed/Release/Appl_RotoDFF_8BtnLed.elf}} verify reset; shutdown 
Output:
failed to upload 
```

So there's something going wrong:

* sloeber is using wrong openocd scripts folder
* parameter "-d2" is missing
* argument "0x2000" is missing

Setup in sloeber:

[image]

[image]

I have no idea how to solve/overcome this issue.
I've downloaded latest SAMD-Core version with arduino IDE + using Arduino IDE 1.6.13

Any ideas?

jantje, 2017-04-19:
Do not use the Arduino IDE installed packages

tuxedo0801, 2017-04-19:
Even if I don't, it does not change the described behavior :-(

The reason I did used the packages was, that it wasn't working. So I specified the packages, but this actually did not help.

jantje, 2017-04-19:
I have been in the beta testing of the zero and I have done debugging with the zero.
The zero did not come up in the last unit test (which ran about 490 boards of which about 80 failed) and as far as I remember the zero ice was not there as well.
So the zero works from before it became public and worked with the last nightly.

tuxedo0801, 2017-04-20:
Hmm, ok. Maybe I try to start from scratch with a fresh install and try again.
Did you test with windows or linux or both?

jantje, 2017-04-20:
V4 is tested with windows/linux and mac
The nightly only with windows
As there are a lot of windows related issues solved in the nightly I advice to take the nightly.
Note that oomph may have recorded settings and set them in a new installation which may cause issues.

jantje, 2017-04-24:
Did you get it to working?

jantje, 2017-05-09:
Closing due to no response of OP

closed 2017-05-09 by jantje