Sloeber/arduino-eclipse-plugin #722: Update to 4.0.1.201704171301 caused a 'path too long' error

Normalised, 2017-04-20:
Just updated to 4.0.1.201704171301 from 4.0.1.201704020116 and I now have an error :

The path C:/dev/apps/eclipse/cpp-neon/eclipse/arduinoPlugin is to long

Which prevents the Upload Sketch command from starting.

(I've reverted my install so I can keep working)

jantje, 2017-04-20:
This is not a issue but a design decision #705

Normalised, 2017-04-20:
Ok, but leaving a previously working install in a broken state with no explanation doesn't seem like the best way of implementing this design decision.

At least a dialog or warning explaining that 'The installation path is too long and the plugin will no longer function correctly' would go a long way to helping.

jantje, 2017-04-20:
Didn't you get a dialog box?

Normalised, 2017-04-20:
Yes, but it just says there's an error and that path is too long. Not friendly or helpful.

I suggest: 

> Due to issues with long pathnames on Windows, the plugin installation path must less than 40 characters. Your current path: PLUGIN_PATH is too long and the plugin will no longer function correctly
> Please visit https://github.com/Sloeber/arduino-eclipse-plugin/issues/705 for details.

jantje, 2017-04-20:
I hope you like the new message more

Normalised, 2017-04-21:
Fantastic 👍

closed 2017-05-25 by jantje

MrAmericanMike, 2018-03-05:
I can't find a clear fix for it. Not even on the https://github.com/Sloeber/arduino-eclipse-plugin/issues/705 that we are pointed to. At least not a clear fix that I can see.
What is the fix for this? I have seen suggestions and some people mentioning some solutions, but not saying how to achieve those, and nobody confirms the suggested solutions works neither.
The only "Clear one" is to change the path to our Eclipse. I guess that means install all over again?