Sloeber/arduino-eclipse-plugin #723: Add an "Arduino Library" project template

rlogiacco, 2017-04-23:
I suggest to add a project template beside the existing _Arduino Sketch_ for those who wish to create a library project.

The project template should provide a template `library.properties` file, an `examples` directory and should also work as an Arduino Sketch template, meaning you should be able to include other libraries.

jantje, 2017-04-24:
I have been thinking about this but could never make it fully work in my head.
The reason why I think about this is because in C/C++ oriented IDE's you can typically create a library and a executable project. For typical C/C++ project this makes sense as the library creates a archive file which can be used in the executable project which is managed as "project dependencies". 
V1 of Sloeber worked this way with the cores.

Where this scheme goes wrong with arduino is that Arduino always uses the code and not the archives. This is done because defines are used to identify the hardware and this requires a rebuild of the code.
This is also true for the core since arduino IDE 1.5.X beta which made me make V2.

Another problem I see is "What to compile". I already created the "link to code" to make library example development easier because I don't see how to solve the 1 project multiple executables (1 for each example)  problem.

Can you elaborate on how you see the use case for the "Arduino library" project?

PS: As a step-up I see more benefit in a library.properties editor and a keywords.txt generator/editor.

rlogiacco, 2017-04-25:
Ok, so, here is the use case I see.

I start working on an Arduino _sketch_ and then I need to use a device or an algorithm or a data structure for which there isn't a library... so I decide to make a library out of my need: what should I do?

Sure, I can create a folder in my sketch project and start working my library in there, add a few files in the folder, an `include` in my sketch and work my way out.

Now that my library is completed I wish to publish it, but to do so I need:

- create an independent github repository to host my library
- move my code
- add the `library.properties`
- create the examples

Now that my code is outside of the project I also need to somehow verify I'm not missing some dependency, so I need to create a new project to at least compile the examples and verify my library is ready to go...

So, in my mind, it would be easier if I could fire a new library project where I can somehow decide which example to build (if many) and verify my library works as intended...

I don't know if that makes sense, mine might be a distorted vision :-P

jantje, 2017-04-25:
Some thoughts..
Do you want a github repository for each library? I have a github repository for all my libraries.
This also means I have 1 project for all my libraries.
There is no need to make the library in a different folder. In most cases I started off with a class and then moved the .h and cpp to a folder in my library project
.
>So, in my mind, it would be easier if I could fire a new library project where I can somehow decide which example to build (if many) and verify my library works as intended...

I agree that this is a easier way of testing a library. But how are you going to tell cdt which example to build? The only thing I can think of is resource exclusion.

rlogiacco, 2017-04-26:
I know you have 1 repo for all your libraries, but that doesn't play nicely if you want to publish your libraries so to have them visible into the Arduino Library manager. This is the reason why I stole your SimPlot library and reworked it into the [PlotPlus github repo](https://github.com/rlogiacco/PlotPlus) (btw, I believe I forgot to add credits there... my bad, I'll remedy, I promise)

But that's not exactly the reason for my request, that can be worked around.

I was thinking about using the _Run As_ command in conjuction with a custom operation like _"Create new library example"_ which creates the `.ino` file and also the _Run As_ configuration which can fire the compiler replacing the appropriate `.ino` file.

Another option would be to create one single _Run As_ command which asks the user to select a file under the examples folder, or actually any file he wishes...

Obviously the best solution would be to run a series of unit tests on the code, but apparently I'm unable to find a decent solution for that...

jantje, 2018-03-06:
...> but that doesn't play nicely if you want to publish your libraries so to have them visible into the Arduino Library manager
. 
As you know... there are many ways to skin a cat.
To add libraries to ArduinoIDE you have to have the source code. If you dump that source code in "a supported folder" you are ok. However this is "advanced usage". For normal usage you need a zip file (in the github zip format) then you can do sketch->include library->add.zip library.
More advanced you  can get your library "arduino approved" so you can add the library with sketch->include libraries->manage libraries. Which points to the same zip file but adds some kind of release management.

From where I am: I don't want to discuss with the arduino people to get "my libraries" on the arduino IDE. I do want to get them in Sloeber. As Sloeber supports multiple library json files, this can easily be done. And I did so for my libraries.
So people wanting to use my libraries can easily use the zip "release files" I need to make for my json file. 
Available on the releases page https://github.com/jantje/ArduinoLibraries/releases.
Sloeber users will not notice a difference between the Arduino pushed libraries and Libraries pushed by others (like me ;-) )

One of the nice features is that the zip file format is the file format provided by git to pack a repository. Making it possible to provide a url that automagically always provides the latest version of the library.  
And guess what...the github zip url gives you the latest code but no update possibilities :-s 
I think that if you want the latest code you also want to be able to update. So my advised way of working would be to import the github  repository (or  your clone) and not the zip url.

The only remaining issue is... how can one easily create a zip file in Sloeber.
I use a external command
[image]

So for me a "arduino library project still does not make sense". I hope you can agree so we can close this issue :-)

jantje, 2018-04-20:
@rlogiacco Do you want to persue this? Or do we close it?

maxela, 2018-06-24:
I vote for the feature request "Arduino library". Without any doubt it makes sense to have a seperate project for eac library. Sorry to say so, but clean and tidy structures are vital for any software project. When it comes to publishing this is even a must.
But why would it help to have a library project template?
First: A library is not for a certain hardware. It is meant to work on an Arduino Nano as well as on an Uno or any other Arduino device. But when I start my library with an Arduino Sketch project, I have to select a device, port, etc. ... :-/
Second: Why don't I use a CDT C++ library project? (...oh btw: Why does CDT have a library project template?) I don't use it, because I would have to add all the dependencies to the Arduino standard libraries myself... :-/
Third: Publishing

jantje, 2018-06-24:
All my libraries are in standard eclipse projects. No cdt no Sloeber just a collection of files.
Why?
Because all my libraries are "upgraded classes". Which means I have some code for which I think "Maybe it is a good idea to make a library out of this".
I use one library project but if you want to have a project per library create a project and drag and drop the code to the library project.
Mark the root of the project as a private library and Sloeber links your library back in (If you used the correct names)
Because the code is linked, any updates done in the original code, are reflected in the library. 
Note that Sloeber allows for linking to the example code (in the new sketch wizard) so you can easily maintain your example code in a "separate project".

CDT has a library project because in the non-arduino world libraries are compiled to archives (.ar, .dll, ...) and shared like that. This is why project dependencies exists.
The first versions of sloeber used 2 project for a sketch, one for the core (archive) and one for the sketch (depended on the core) Due to changes in arduino IDE 1.5 and later this was no longer workable and I modified Sloeber to build a archive and the sketch in 1 project which is pretty uncommon in CDT world.

closed 2019-01-09 by jantje