Sloeber/arduino-eclipse-plugin #725: Multiple .ino files

JohnInWI, 2017-04-24:
First, a big thank you to the developers of this project, Eclipse is a vastly superior IDE vs Arduino IDE. I am using V4.0 and downloaded the product bundle.

I am working on a project with multiple .ino files (a mod of https://github.com/goebish/nrf24_multipro) 
I can work on and build the main .ino file, but the moment I open any of the accompanying .ino files or add a new one, multiple errors are thrown. The project still seems to build with the Arduino plugin, however errors remain. When project is opened in Arduino or other IDE, I have no issues with the same multiple .ino files.

I'm a noob to Eclipse, so perhaps my lack of understanding of how work-spaces are handled, etc are contributing to my issue.

jantje, 2017-04-24:
Given that you are not experienced in this are I think you are referring to FAQ http://eclipse.baeyens.it/faq.shtml#/troubleshooting Eclipse tells me there are bugs in my code but it compiles ok.
Basically said: I consider ino files a bad way of working. Support for ino files is there to get the examples to work and to compile projects. Forcing the indexer to look at the .ino files as one big file is a lot harder.
As to me I think you have 2 options
1) ignore the indexer generated warnings
2) create a header file like it should be.

1 is easy but you will lose lots of functionality. You can even disable the indexer in the project properties but then you will lose even more functionality.
2 will take more time as you will need to add #include "header.h" to all ino files and need to create the header file. However after you have done so you can rename the .ino files to .cpp files and make libraries out of them.
I tested this for some files so I'll share the header I have created here to get you going.
```
#ifdef __IN_ECLIPSE__
#ifndef __HEADER_H__
#define __HEADER_H__

#include "Arduino.h"
#include <util/atomic.h>
#include <EEPROM.h>
#include "iface_nrf24l01.h"
uint32_t process_Bayang() ;
void Bayang_init() ;
void Bayang_bind() ;
void Bayang_send_packet(u8 bind) ;
uint32_t process_CG023() ;
void CG023_init() ;
void CG023_bind() ;
void CG023_WritePacket(uint8_t init) ;
uint32_t process_CX10() ;
void CX10_init() ;
void CX10_bind() ;
void CX10_Write_Packet(uint8_t init) ;
uint32_t process_FQ777124() ;
void FQ777124_init() ;
void FQ777124_bind() ;
uint16_t nrf_crc(uint8_t data[], uint8_t len, uint16_t crc) ;
void ssv_pack_dpl(uint8_t addr[], uint8_t pid, uint8_t* len, uint8_t* payload, uint8_t* packed_payload) ;
void FQ777124_send_packet(u8 bind) ;
void H7_initTXID() ;
void H7_init() ;
void H7_bind() ;
uint8_t H7_calcChecksum() ;
void H7_WritePacket() ;
uint32_t process_H7() ;
uint32_t process_H8_3D() ;
void H8_3D_bind() ;
uint8_t  H8_3D_checksum() ;
void H8_3D_send_packet(uint8_t  bind) ;
void H8_3D_init() ;
uint32_t process_HiSky() ;
void HiSky_init() ;
void HiSky_calc_fh_channels() ;
void HiSky_build_binding_packet(void) ;
void HiSky_build_ch_data() ;
void kn_start_tx(u8 bind_yes) ;
u32 process_KN() ;
void kn_init(u8 tx_addr[], u8 hopping_ch[]) ;
void kn_calculate_freqency_hopping_channels(u32 seed, u8 hopping_channels[], u8 tx_addr[]) ;
void kn_calculate_tx_addr(u8 tx_addr[]) ;
void kn_bind_init(u8 tx_addr[], u8 hopping_ch[], u8 bind_packet[]) ;
void kn_send_packet(u8 packet[], int32_t rf_ch) ;
void kn_send_init(u8 tx_addr[], u8 packet[]) ;
void kn_update_packet_send_count(u8 packet[], int32_t packet_sent, int32_t rf_ch_idx ) ;
void kn_update_packet_control_data(u8 packet[], int32_t packet_count, int32_t rf_ch_idx) ;
void kn_read_controls(u16* throttle, u16* aileron, u16* elevator, u16* rudder, u8* flags) ;
u16 kn_convert_channel(u8 num) ;
u8 mjx_checksum() ;
u8 mjx_convert_channel(u8 num) ;
u8 mjx_pan_tilt_value() ;
void mjx_send_packet(u8 bind) ;
uint32_t process_MJX() ;
void initialize_mjx_txid() ;
[... 77 lines cut ...]
    AUX7,  // (CH11) calibrate X (V2x2), roll trim (H7)
    AUX8,  // (CH12) Reset / Rebind
};

#define PPM_MIN 1000
#define PPM_SAFE_THROTTLE 1050
#define PPM_MID 1500
#define PPM_MAX 2000
#define PPM_MIN_COMMAND 1300
#define PPM_MAX_COMMAND 1700
#define GET_FLAG(ch, mask) (ppm[ch] > PPM_MAX_COMMAND ? mask : 0)
#define GET_FLAG_INV(ch, mask) (ppm[ch] < PPM_MIN_COMMAND ? mask : 0)

static uint16_t ppm[12] = {PPM_MIN,PPM_MIN,PPM_MIN,PPM_MIN,PPM_MID,PPM_MID,
                           PPM_MID,PPM_MID,PPM_MID,PPM_MID,PPM_MID,PPM_MID,};
extern bool reset;
#endif
#endif

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

```

jantje, 2017-04-30:
Can I close this issue?

JohnInWI, 2017-05-01:
Yes thanks!

closed 2017-05-01 by jantje