Sloeber/arduino-eclipse-plugin #726: How to add ESP32 to sloeber

abkglobal, 2017-04-25:
Please can anyone explain the steps of how you added ESP32 to sloeber and got it working?

jantje, 2017-04-25:
Based on these instructions https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md
it looks quite easy to do in Sloeber (git is integrated in Sloeber) but if yo are not used to it there are plenty of possibilities to go wrong.
It is basically the same as I do here for local debugging https://youtu.be/ZN-rLSyhBN8?t=400

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-04-29:
Did you get it to work?

abkglobal, 2017-05-01:
Hello,Thanks for your email. I'm struggling to instal sloeber successfully. After lanching it showed the message on the right.I updated my java, but still the same message.

jantje, 2017-05-01:
I don't know what you mean with "After lanching it showed the message on the right"
On windows a typical issue is installing the 64 bit sloeber and not having a 64 bit java properly installed.
If the arduino IDE is working for you I propose to install the 32 bit version of sloeber to get you going

abkglobal, 2017-05-02:
Thanks,I installed the 32bit version. And it compiles and works well with the arduino.I have imported the repository for the ESP32 just as I did for the arduino. ButI get the following message below. What do I need to get the ESP32 compiling correctly.Thanks.
/////////////////////////////////////////////Compile message////////////////////////////////////////////////////////////////////////////////////////////////////////////
08:57:42 **** Clean-only build of configuration Release for project esp32_blink ****"C:\\unitTest\\sloeber\\arduinoPlugin\\tools\\make\\make" clean make: *** No rule to make target 'clean'.  Stop.
08:57:42 Build Finished (took 320ms)
08:57:44 **** Build of configuration Release for project esp32_blink ****"C:\\unitTest\\sloeber\\arduinoPlugin\\tools\\make\\make" all 'Building file: ..\.ino.cpp''Starting C++ compile'"C:/unitTest/git/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/config" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/bluedroid" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/app_update" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/bootloader_support" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/bt" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/driver" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/esp32" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/ethernet" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/fatfs" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/freertos" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/log" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/mdns" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/mbedtls_port" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/nvs_flash" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/openssl" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/soc" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/spi_flash" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/sdmmc" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/tcpip_adapter" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/ulp" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/vfs" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/xtensa-debug-module" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/coap" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/wpa_supplicant" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/expat" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/json" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/nghttp" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -c -DF_CPU=240000000L -DARDUINO=10802 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ARDUINO-ESP32 -DESP32 -DCORE_DEBUG_LEVEL=3  -I"C:\unitTest\git\arduino-esp32\cores\esp32" -I"C:\unitTest\git\arduino-esp32\variants\esp32" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"The system cannot find the path specified.subdir.mk:24: recipe for target '.ino.cpp.o' failedmake: *** [.ino.cpp.o] Error 1
08:57:45 Build Finished (took 722ms)

abkglobal, 2017-05-02:
Hello,I just found out that I had to go into the  esp32  folder  and locatea file called 'get' and run the  file. This generated some extra files and folder in the ESP32 folder. It now compiles succesfully. I am now looking at debugging it on the hardware demo board I've got.Regards,Louis.

      From: Louis Abraka <abkglobalresearch@yahoo.com>
 To: Sloeber/arduino-eclipse-plugin <reply@reply.github.com> 
 Sent: Tuesday, 2 May 2017, 9:02
 Subject: Re: [Sloeber/arduino-eclipse-plugin] How to add ESP32 to sloeber (#726)
   
Thanks,I installed the 32bit version. And it compiles and works well with the arduino.I have imported the repository for the ESP32 just as I did for the arduino. ButI get the following message below. What do I need to get the ESP32 compiling correctly.Thanks.
/////////////////////////////////////////////Compile message////////////////////////////////////////////////////////////////////////////////////////////////////////////
08:57:42 **** Clean-only build of configuration Release for project esp32_blink ****"C:\\unitTest\\sloeber\\arduinoPlugin\\tools\\make\\make" clean make: *** No rule to make target 'clean'.  Stop.
08:57:42 Build Finished (took 320ms)
08:57:44 **** Build of configuration Release for project esp32_blink ****"C:\\unitTest\\sloeber\\arduinoPlugin\\tools\\make\\make" all 'Building file: ..\.ino.cpp''Starting C++ compile'"C:/unitTest/git/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/config" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/bluedroid" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/app_update" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/bootloader_support" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/bt" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/driver" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/esp32" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/ethernet" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/fatfs" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/freertos" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/jsmn" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/log" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/mdns" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/mbedtls" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/mbedtls_port" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/nvs_flash" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/openssl" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/soc" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/spi_flash" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/sdmmc" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/tcpip_adapter" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/ulp" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/vfs" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/xtensa-debug-module" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/newlib" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/coap" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/wpa_supplicant" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/expat" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/json" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/nghttp" "-IC:/unitTest/git/arduino-esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -c -DF_CPU=240000000L -DARDUINO=10802 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ARDUINO-ESP32 -DESP32 -DCORE_DEBUG_LEVEL=3  -I"C:\unitTest\git\arduino-esp32\cores\esp32" -I"C:\unitTest\git\arduino-esp32\variants\esp32" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"The system cannot find the path specified.subdir.mk:24: recipe for target '.ino.cpp.o' failedmake: *** [.ino.cpp.o] Error 1
08:57:45 Build Finished (took 722ms)

jantje, 2017-05-02:
great so this issue can be closed

abkglobal, 2017-05-02:
Hello,I am still struggling the debug the ESP32. It seems I am limited to debugging only the ESP6288. I can't seem to find the gbd library to add. Although ,I have already included it in the ino file.See below a screen shot of my work.

jantje, 2017-05-02:
For debugging I imported stuff from http://gnuarmeclipse.github.io/  for esp32 you probably need to install other stuff.
Do not forget to do a write up on how to do it ;-)

jantje, 2017-05-25:
I'm closing this issue as I do not support debugging esp32 as part of sloeber. If I did it would be under the jantje/hardware repository

closed 2017-09-01 by jantje