Sloeber/arduino-eclipse-plugin #727: Add support for multiple library json files

jantje, 2017-04-25:
As implemented #724.
Note that after this upgrade you need to add http://downloads.arduino.cc/libraries/library_index.json to windows->preferences->third party index url's

jantje, 2017-04-25:
@martinmikula do you have links to library json files I can use to test?

martinmikula, 2017-04-26:
I just created one for tests: www.lmt.sk/arduino/library_mikula_index.json (please let me know if it works for you...)

jantje, 2017-04-30:
I fixed the multiple version issue here locally so unit testing is getting lots better.
But there is still a problem with libs that are in multiple json files

I'm not sure what the intend of this json file is but if we want to make this a official Sloeber library json file (and that would be a good thing I think) the following libraries need some fixing
A4963 needs include guards (I'll raise pull request)
Adafruit_Motor_Shield_library not found based on "AFMotor.h"  (I consider making a mapping table for known libraries)
ACE128 error in include guard
many libraries have folders hanging around that need to be filtered out. (I think I know a solution to this)
Not sure what to think about the ACROBOTIC_SSD1306 library and examples
Same for AD7193 library

jantje, 2017-05-01:
I also found a another case for multiple libraries.
Your json file contains keypad and arduino keyPad
As the windows filesystem is case insensitive this ends up as the same

jantje, 2017-05-09:
Upgrade instructions:
add http://downloads.arduino.cc/libraries/library_index.json to arduino->preferences->arduino->Third party index url's

closed 2017-05-25 by jantje