Sloeber/arduino-eclipse-plugin #730: Project build if called function from both time.h and Time.h

dpeddi, 2017-04-29:
If you use function exopsed  by Time library and function exposed by time.h you may get into trouble.

Under windows you get Time.o and since windows is case insensitive you wont get time.o build. Probably on linux all would work...

Arduino ide seems to differentiate the obj name of core and library so doesn't have this issue.

Other ide have same issue like sloeber (platformio).. I've opened an issue to Time developer but simply answered... On Arduino it works...

jantje, 2017-04-29:
Can you provide the console output so I can understand better?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

dpeddi, 2017-04-30:
Actually I can't reproduce it

closed 2017-04-30 by dpeddi

jantje, 2017-04-30:
:-)