Sloeber/arduino-eclipse-plugin #731: Unable to include downloaded Arduino libraries

mtwheatley, 2017-04-29:
I have downloaded a bunch of Arduino libraries and installed them using the Arduino IDE library manager into a folder: c:\users\xxxx\documents\Arduino\libraries.  Included are libraries that include things like Timezone.h and DallasTemerature.h.

In Sloeber Eclipse I added the Arduino library folder (described above) to Window->Preferences->Arduino->Private Library Path.  

However, when I #include header files from libraries in this folder, they remain unresolved in the Arduino Eclipse sketch.

Help!

Michael Wheatley

jantje, 2017-04-29:
this is probably regression due to #729 
I will try to look at it tomorrow

jantje, 2017-04-30:
Seems like your issue is not related to #729 
Your workflow is simply not supported. You can not use libraries downloaded with the Arduino IDE. You have to download and install the libraries with the library manager in Sloeber.
Here is a video explaining how to do so https://www.youtube.com/watch?v=MGAyIOC24lU

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-04-30 by jantje

mtwheatley, 2017-04-30:
Thank you.  I am now a patron.

Michael