Sloeber/arduino-eclipse-plugin #736: .ino files that include headers that are "defined out" are included in the .ino.cpp file

jantje, 2017-05-03:
following code will make the include to be included in the .ino.cpp file, while it should not
```
#define A
#ifndef A
#include include.h
#endif
```

jantje, 2017-05-09:
Work around: disable auto lib import in preferences->arduino and do the lib import manually

Gerry33, 2017-05-10:
Hi Jantje,
I observed the same problem, but it  has nothing to do with imports.

myFile.ino:
#define A 

... compile ...

// #define A  comment out cause not needed
... compile ...

-> 'A' is still defined  (in '.ino.cpp') and code sections with '#ifdef A' are therefore still compiled. 
A very annoying and error prone bug. 

br
gerry

jantje, 2017-05-10:
I meant add libraries not import.
Thinking about this that will not fix the issue as you will get a error on not finding the header.
I think using .cpp files will.

uzi18, 2017-06-14:
if use external .h files in project, example in ino file:
```
int data_max=10;
#include "test.h"
```
in test.h:
```
#define something_usefull 3
int buffer[data_max];
```
Sloeber creates .ino.cpp:

```
#include "test.h"
#include "project.ino"
```

couses compilation error as data_max is unknown, but Arduino IDE compilation works ok.

Why not add to .ino.cpp only known Libraries and .ino files?

jantje, 2017-06-14:
did you test with the latest nightly?

closed 2017-06-17 by jantje