Sloeber/arduino-eclipse-plugin #737: Add option to add #pragma once to header files automatically

jantje, 2017-05-03:
When libraries are installed and projects are created from sample code the #pragma once can be added to all .h files
Doing so will get lots of examples to work "out of the box"
For those who (like me before today) do not know what #pragma once is all about.
It is the same thing as the #include guards but far better
1) less error prone as you do not risk to have the same define 2 times
2) should be faster
3) easier to add as it only needs 1 line to be added somewhere (as to 2 lines at the top and 1 at the bottom

closed 2017-06-17 by jantje

ArminJo, 2019-09-18:
Hi,
how can I suppress adding of more than one #pragma once.
I have *.h files which look like:
`#endif /* SRC_ADCUTILS_H_ */

#pragma once

//Added by Sloeber 
#pragma once

//Added by Sloeber 
#pragma once`

erdemuncuoglu, 2019-09-18:
Hello @ArminJo,

Would you consider opening a new issue? This one seems to be rather old to be related.

Also can you please check this [ISSUE_TEMPLATE](https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/.github/ISSUE_TEMPLATE/bug_report.md) so the mighty developers can solve it more easily and more quickly.

Sorry in advance if you get annoyed of my comment.