Sloeber/arduino-eclipse-plugin #739: Some private libraries refuse to be added.

far5893, 2017-05-09:
I have e private folder where i put libraries i create or downloaded, there are two libraries that refuses to be automatically added to project. All other libraries are ok, where can i find a log or info why they are discarded ?

far5893, 2017-05-09:
[image]

jantje, 2017-05-09:
These 2 video's contain the (possible ) answers to your questions
Why is my lib not found: https://www.patreon.com/posts/7891687 (note I'm working on a more advanced way right now)
How to use libraries from github (or any other source) in Sloeber https://www.youtube.com/watch?v=d-Vt21BIHl0&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF&index=5

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-05-09 by jantje

far5893, 2017-05-09:
But there is a bug! If i put files in SMTPClient in cron (a library i wrote than it import automatically) and do a library refresh it work ok.

far5893, 2017-05-09:
Ok it's a bug in indexer, if i move #includes there are changes in included  libraries!

far5893, 2017-05-09:
Ok for now i solved moving header and source of un-included libraries in included libraries, and 
adding some #include where indexer like them (and not where they should be).

jantje, 2017-05-09:
If you think it is a bug please describe step by step procedure to reproduce the issue.