Sloeber/arduino-eclipse-plugin #747: cannot change location of ino file.

cmedianu, 2017-05-22:
It would be useful if projects could be shared with platform io. To do this, it should be possible to relocate the ino file to the src directory. This would also make the sloeber / arduino project less messy.

Thanks.

jantje, 2017-05-22:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2023-07-10:
I'm working on this one right now