Sloeber/arduino-eclipse-plugin #748: Problem with directory locations after 'sharing' project with Git.

judge2005, 2017-05-22:
Hi,
I 'shared' my project using eGet, which very kindly relocated everything to my git directory. Now my sloeber build doesn't work because when it links it looks in the old location for the Release directory rather than the new location:

C:\Users\mpand\eclipse\java-neon\eclipse\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar: **C:\Users\mpand\Documents\74141\K155ID1/Release/arduino.ar**: No such file or directory
makefile:85: recipe for target '**C:/Users/mpand/git/nclock1/K155ID1/Release/arduino.ar**' failed
make: *** [C:/Users/mpand/git/nclock1/K155ID1/Release/arduino.ar] Error 1

jantje, 2017-05-22:
I'm not sure what you did but it looks like you also shared your release folder.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jensriebold, 2017-05-24:
I just become a $5/month Patreon because you made a great job and I hope that cross platform project sharing will work in the next release of the project!

Thanks in advance

jantje, 2017-05-24:
Thanks :-)
Git sharing works and 4.1 (needs to be released but issues keep popping up) will do a far better job at the cross platform level detection and remediation. 
However you should only check in "input" of the build process and not "output". The way you scan tell git to ignore files/folders is by adding them to the .ignore file.
 
In Sloeber by default everything in the "release" folder is output and should not be put in the version control system. 
Everything in the libraries folder are linked resources so again this folder should be added to the .ignore file.
Therefore you should add the release and libraries folders to the .ignore file.

jantje, 2017-06-15:
Did you succeed to get it to work?

judge2005, 2017-06-15:
I haven't tried since. I switched to subversion for my private projects, but at some point I will have to use GIT again :(

jantje, 2017-06-15:
Ok I'll close the defect. You can always reopen if needed. But I'm pretty confident that it works.

closed 2017-06-15 by jantje