Sloeber/arduino-eclipse-plugin #749: Cannot build library examples

Mrnt, 2017-05-22:
This may be by design, but if I'm developing an example sketch as part of a custom library it is not possible to build the example inside Eclipse. When the custom library is attached to the example sketch (Arduino->Add a library to the selected project) a dialog appears with the following error:

>Could not create folderlibraries/MyCustomLib
>Cannot create a link to '/Users/bob/Development/ArduinoDevArea/libraries/MyCustomLib' because it overlaps the location of the project that contains the linked resource.

The directory structure I am using is:

    // base directory for custom lib:
    ArduinoDevArea/libraries/MyCustomLib
    // lib files: 
    ArduinoDevArea/libraries/MyCustomLib/MyCustomLib.h
    ArduinoDevArea/libraries/MyCustomLib/MyCustomLib.cpp
    // example1 sketch:
    ArduinoDevArea/libraries/MyCustomLib/examples/example1.ino

I have example1.ino as an Arduino project and it looks something like:

    #include "MyCustomLib.h"
    //etc

jantje, 2017-05-22:
Try creating a new sketch and use your example and select "link to code" 

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-05-25:
@Mrnt 
Are you ok with this answer?
Can you close this issue?

closed 2017-05-29 by jantje