Sloeber/arduino-eclipse-plugin #750: Problem with generated utf-8 .ino file

nicoverduin, 2017-05-23:
Not shure this has been reported before but I noticed the following issue:
**Situation**
I use Sloeber versie 4.0. All my code is generated using Software Ideas Modeler 10.68. This means the code for  a .ino file and c++ classes as well. ****
**Problem**
If I generate the code in UTF-8 format including a sketch file of type .cpp there is no problem. All are completely compiled by both Sloeber and the Arduino IDE.
If I generate the sketch as .ino file UTF-8 format I get an error from the compiler stating that my includes are missing the last double-quotes. If I rename this sketch file back to .cpp all compiles fine.
If I generate the sketch the sketch in ascii format as .ino all compiles well. 

Based on the above the problem seems to arise from where Sloeber copies the .ino into a .ino.cpp file?
I did check the generated files and all have the 3 byte UTF-8 signature at the beginning. If I blank out these 3 bytes (using a hex editor) as spaces the problem is gone. 

Regards
Nico

jantje, 2017-05-23:
It  is quite possible that the parsing doesn't work properly when using "other file formats"
Mine is set to CP1252.
I'm actually surprised you want to use ino format if you have cpp at your disposal.
To investigate look at the .ino.cpp file. SOmething must be "incorrect there"

nicoverduin, 2017-05-23:
The only reason still spending effort for .ino is for those morons that persistently stick to the arduino ide..... it seems the .ino.cpp is missing all the last double quotes of the includes. But as i am the only one having this issue (why would you want to generate all the code after all :) ) i'll close this one as the workaround is in any case plain ascii

jantje, 2017-05-23:
txs. It is probably more work to setup a test environment for me than to fix the problem.
The problem is here (and 3 lines below).
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/PdePreprocessor.java#L185.
As you have everything at your disposal it should be easy to set a breakpoint and have a look.

nicoverduin, 2017-05-23:
I have setup the environment in the past. Of course that is no longer existing.... Is there a setup procedure for Windows?

jantje, 2017-05-23:
Short version: install eclipse java add cdt and gnuarm and import the project from github

jantje, 2017-05-23:
och you also need to add nebula

nicoverduin, 2017-05-23:
It seems a bit more complicated than jus removing the first 3 bytes. Ther could be specia characters inside the file as well as the user of SIM can type anything in. Exporting to ascii solves that problem. I will leave it as is.

closed 2017-05-23 by nicoverduin

jantje, 2017-05-23:
ok. 
Thinking about it I think the problem is that a external tool is saving a file in a format different than eclipse is told to save in.
So It might work with utf-8 if you set the text file encoding under windows->preferences->general->workspace to utf-8.

nicoverduin, 2017-05-24:
Unfortunately makes no difference.

jantje, 2017-05-24:
then debugging is the only way to find the root cause