Sloeber/arduino-eclipse-plugin #751: Serial plotter issue : Failed to create the part's control (RELOADED)

DareDevil73, 2017-05-23:
Hi, I cannot post my problem on the closed issue #650 so I created a new one.

I have just installed, from scratch, by the TAR.GZ, the latest release available on the web site, and I got the same issue.

I searched a lot for the _plugin.xml_ file, but I didn't find it.

Could you help me please?

jantje, 2017-05-23:
please state

-  the sloeber version you use
- the os
- The stack strace
- 32 or 64 bit
- And the plugin.xml as you can find according to this comment
- https://github.com/Sloeber/arduino-eclipse-plugin/issues/650#issuecomment-278471973

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

DareDevil73, 2017-05-23:
Hi, I'll be a supporter ASAP be sure! :)

The answers to your questions are:

- Sloeber 4.0 (the latest availble on the web site)
- OS Linux Mint 18.1 64 bit on VMWare 12.5 for Windows 10 PRO 64 bit
- 64 bit for HOST and GUEST OS

Unfortunately I cannot find the plugin.xml, these are the onyl I found:

./programs/sloeber/configuration/org.eclipse.osgi/18/0/.cp/plugin.xml
./programs/sloeber/plugins/org.eclipse.ui.themes_1.1.300.v20161107-1827/plugin.xml
./programs/sloeber/plugins/org.eclipse.platform_4.6.2.v20161124-1400/plugin.xml

The stack trace is:

java.lang.IllegalArgumentException: Argument not valid
	at org.eclipse.swt.SWT.error(SWT.java:4514)
	at org.eclipse.swt.SWT.error(SWT.java:4448)
	at org.eclipse.swt.SWT.error(SWT.java:4419)
	at org.eclipse.swt.widgets.Widget.error(Widget.java:483)
	at org.eclipse.swt.widgets.Control._setBackground(Control.java:4193)
	at org.eclipse.swt.widgets.Control.setBackground(Control.java:4183)
	at org.eclipse.nebula.widgets.oscilloscope.multichannel.Plotter.<init>(Plotter.java:189)
	at io.sloeber.monitor.views.MyPlotter.<init>(MyPlotter.java:35)
	at io.sloeber.monitor.views.PlotterView.createPartControl(PlotterView.java:77)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.createPartControl(CompatibilityPart.java:150)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityView.createPartControl(CompatibilityView.java:143)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.create(CompatibilityPart.java:340)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(java.base@9-internal/Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(java.base@9-internal/NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(java.base@9-internal/DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(java.base@9-internal/Method.java:531)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:966)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:931)
	at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:151)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:375)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:294)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:162)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.createFromBundle(ReflectionContributionFactory.java:105)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.doCreate(ReflectionContributionFactory.java:74)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.create(ReflectionContributionFactory.java:56)
	at org.eclipse.e4.ui.workbench.renderers.swt.ContributedPartRenderer.createWidget(ContributedPartRenderer.java:129)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:975)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:651)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$1.run(PartRenderingEngine.java:536)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:520)
	at org.eclipse.e4.ui.workbench.renderers.swt.ElementReferenceRenderer.createWidget(ElementReferenceRenderer.java:70)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:975)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:651)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:757)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:728)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:722)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:706)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.subscribeTopicToBeRendered(PartRenderingEngine.java:149)
	at sun.reflect.GeneratedMethodAccessor63.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(java.base@9-internal/DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(java.base@9-internal/Method.java:531)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.di.internal.extensions.EventObjectSupplier$DIEventHandler.handleEvent(EventObjectSupplier.java:80)
	at org.eclipse.equinox.internal.event.EventHandlerWrapper.handleEvent(EventHandlerWrapper.java:201)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:197)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)
	at org.eclipse.equinox.internal.event.EventAdminImpl.dispatchEvent(EventAdminImpl.java:135)
[... cut ...]
	at sun.reflect.NativeMethodAccessorImpl.invoke(java.base@9-internal/NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(java.base@9-internal/DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(java.base@9-internal/Method.java:531)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1492)

Thank you so much!
Antonio

jantje, 2017-05-23:
> Sloeber 4.0 (the latest availble on the web site)

From the stack trace I would assume this the latest stable and not the latest nightly. 
@wimjongman is Bug 510847 https://bugs.eclipse.org/bugs/show_bug.cgi?id=510847 fixed in nebula?

DareDevil73, 2017-05-23:
Yes, it is the latest stable.

How could I get rud of this bug?

jantje, 2017-05-23:
The bug in in nebula and I think it is fixed. So upgrading nebula should work. The nebula update site is http://download.eclipse.org/nebula/snapshot
I think a better idea right now is upgrading to the sloeber nightly http://eclipse.baeyens.it/update/V4/nightly
Both will only fix the issue if Bug 510847 is fixed in nebula

wimjongman, 2017-05-23:
> nebula/snapshot

No it is not fixed in Nebula AFAIK.

jantje, 2017-05-23:
@wimjongman 
Shall I create a pull request?

jantje, 2017-05-23:
@wimjongman 
I pushed a commit. I made bg a local variable and removed the dispose

DareDevil73, 2017-05-24:
Thank you! Can I download this update in some way?

DareDevil73, 2017-05-24:
During update I got this error:

> An error occurred while collecting items to be installed
session context was:(profile=DefaultProfile, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
HTTP Server 'Service Unavailable': http://download.eclipse.org/nebula/snapshot/plugins/org.eclipse.nebula.widgets.ganttchart_1.0.0.201705221618.jar
HttpComponents connection error response code 503.

I restarted the IDE, but the Plotter shoe me the same exception message. :(

jantje, 2017-05-24:
> Thank you! I inspected your commit and added the softwre store. I'm going to check it...

I'm not following here

>During update I got this error:

Strange Sloeber is not using/referencing ganttchart. If the install continued you should be fine.

>I restarted the IDE, but the Plotter show me the same exception message. :(

As Wim stated 

> No it is not fixed in Nebula AFAIK.

I pushed a commit to nebula and the review pointed out more issues of the same type. I'm not sure how long it takes to get it on the server. But I'm working on it.

jantje, 2017-05-24:
I'm not sure why but this is not happening on my windows machine. So I'm asuming it is a linux only problem

DareDevil73, 2017-05-24:
I tried to update to nightly build, but the issue persists.

wimjongman, 2017-05-24:
For people interested in the intermediate patch (not for the weak of heart):

Add this update site to Eclipse:

https://hudson.eclipse.org/nebula/job/nebula.stable.gerrit/lastSuccessfulBuild/artifact/releng/org.eclipse.nebula.updatesite/target/repository/

[image: Inline image 1]

[image: Inline image 3]

[image: Inline image 4]

DareDevil73, 2017-05-24:
Hi, I suspect you missed a file :)

> Unable to read repository at https://hudson.eclipse.org/nebula/job/nebula.stable.gerrit/lastSuccessfulBuild/artifact/releng/org.eclipse.nebula.updatesite/target/repository/content.xml.
Unable to read repository at https://hudson.eclipse.org/nebula/job/nebula.stable.gerrit/lastSuccessfulBuild/artifact/releng/org.eclipse.nebula.updatesite/target/repository/content.xml.
Unable to make member of class sun.security.ssl.SSLSocketImpl accessible:  module java.base does not export sun.security.ssl to unnamed module @4b78028

DareDevil73, 2017-05-25:
Hi @wimjongman , could you publish the _content.xml_ missing file? Thank you!

jantje, 2017-05-25:
The suspected issue has been fixed and accepted now in nebula.
I'm not sure how long it will take to be available at http://download.eclipse.org/nebula/snapshot but there is a version of today. As I don't have the problem I can't see if it contains the fix nor if the fix works

wimjongman, 2017-05-25:
> nebula/job/nebula.stable.gerrit/lastSuccessfulBuild/
> artifact/releng/org.eclipse.nebula.updatesite/target/
> repository/content.xml.

The Eclipse fix was merged so it should be available in the nightly.

Cheers,

Wim

jantje, 2017-05-25:
I'm currently releasing V4.1 which has been build this afternoon.
It should contain the fix.

DareDevil73, 2017-05-25:
Thank  you so much, but could I kindly ask you which are the only needed software update sites to add to my installation? Due to several attempts to fix the issue I added a lot of garbage to the list.

Regards,
Antonio

jantje, 2017-05-25:
This is what the V4.1 fresh install has
[image]

DareDevil73, 2017-05-26:
Hi, I just removed my old release, downloaded a fresh 4.1 tar.gz and installed it.

Now I don't get any exception, but the plotter view is a black canvas with a vertical yellow line on the right from the top to the bottom of the view.

Did I miss something?

Thank you!

jantje, 2017-05-26:
I just tested here.
This is what I see
[image]
What do your colour settings look like?
[image]

DareDevil73, 2017-05-26:
Hi, attached my settings,the default ones! :(
[image]

jantje, 2017-05-26:
Any errors in the error log?
Did you try to send data to the plotter and see wether you can see it?

DareDevil73, 2017-05-26:
In the error log there are no errors, and I see the first image I attached to this thread (3 post above).

jantje, 2017-05-26:
The yellow line is normal.
The message about connecting the serial monitor is only there f the serial monitor is not connected.
But the crosshairs should be there.
Does the plotter react to double click?
Please use the simplot library example to see wether you can see any output https://github.com/jantje/ArduinoLibraries/tree/master/simplot

jantje, 2017-05-26:
Also try to change the background color to white and see whether you see more then. Note you need to close and reopen the plotter to see the changes.

DareDevil73, 2017-05-26:
> Does the plotter react to double click?

Yes, it asks me to selec a CSV file from the file system.

> Also try to change the background color to white and see whether you see more then. Note you need to close and reopen the plotter to see the changes.

I see a white canvas.

Sorry fot the stupid question. I'm use this great tool (Sloeber) since few days, so I expect to see onto the plotter view the data got from the serial as for the Arduino IDE. Isn't it?!

Thank you for your patience and assitance!

jantje, 2017-05-26:
> Is it right?

Yes
The simplot library packs data (maximum 6 integer values) and sends it over the serial connection in such a way that sloeber can identify what is to be plotted on the plotter and what is to be printed in the serial monitor.

jantje, 2017-05-29:
@DareDevil73 Would you mind trying to debug this issue?
As I can not reproduce it there are very little other options to find the problem.

jantje, 2017-07-04:
Any update on this issue?

wimjongman, 2017-11-22:
I think it is a user error as he indicated in one of his comments. There are no further reports of this issue. Please reopen if it happens again.

closed 2017-11-22 by wimjongman