Sloeber/arduino-eclipse-plugin #755: can no longer add a library to the project !!

ghislainborremans, 2017-05-26:
Hello 
for some reason i can no longer add a library to the project ( the dialogues does not appear anymore) 
Because i am in an urgency, i reinstalled Arduino and then installed teh new sloever V4.1,. I added the arduino/esp8266 url and selected the 2.1.0 ( this is what i had before ) 
Still no solution, the dialogue does not appear?? 
what to do

jantje, 2017-05-26:
I just tested here with a V4.1 download 32bit on windows
[image]
Do you see any errors in the error log?

chfoo, 2017-06-12:
I can reproduce this by:

1. Install a library, eg "Arduino Low Power" using Preferences->Arduino->Library Manger
2. Open "Add a library to selected project" from Preferences. (It works here.)
3. Remove "Arduino Low Power" by unticking it in Library Manager
4. Try opening "Add a library to selected project" from Preferences.  Window does not open anymore.

I get this error:

```
!ENTRY org.eclipse.ui 4 0 2017-06-11 22:42:18.040
!MESSAGE Unhandled event loop exception
!STACK 0
org.eclipse.e4.core.di.InjectionException: java.lang.ArrayIndexOutOfBoundsException: 0
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:65)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:282)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:264)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:132)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:152)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:494)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:488)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:210)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:433)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:454)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem$3.handleEvent(AbstractContributionItem.java:482)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5227)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1340)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4561)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4151)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:1121)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1022)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:150)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:693)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:138)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
Caused by: java.lang.ArrayIndexOutOfBoundsException: 0
	at io.sloeber.core.tools.Libraries.findAllArduinoManagerLibraries(Libraries.java:117)
	at io.sloeber.core.tools.Libraries.getAllInstalledLibraries(Libraries.java:155)
	at io.sloeber.core.api.Sketch.getAllAvailableLibraries(Sketch.java:191)
	at io.sloeber.ui.Import_Libraries_Page.createSourceGroup(Import_Libraries_Page.java:85)
	at io.sloeber.ui.Import_Libraries_Page.createControl(Import_Libraries_Page.java:58)
	at org.eclipse.jface.wizard.WizardDialog.showStartingPage(WizardDialog.java:1234)
	at org.eclipse.jface.wizard.WizardDialog.createContents(WizardDialog.java:590)
	at org.eclipse.jface.window.Window.create(Window.java:426)
	at org.eclipse.jface.dialogs.Dialog.create(Dialog.java:1095)
	at org.eclipse.jface.window.Window.open(Window.java:783)
	at io.sloeber.ui.actions.AddLibraryAction.execute(AddLibraryAction.java:48)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:295)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:90)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	... 37 more
```
I can workaround this by deleting the empty library directory in `sloeber/arduinoPlugin/libraries`.

jantje, 2017-06-12:
Thanks. I'll have a look at this as I'm currently working on improving library support :-)

jantje, 2017-06-12:
I see I already catered for the issue which I think is the root cause of the exception.
Are you sure sloeber/arduinoPlugin/libraries was empty? Looking at the code it should have contained a empty folder.
This is important because if it is a lib folder remaining the root cause is not yet fixed! I can't try it out now as my eclipse instance is running tests with all the libraries installed.

chfoo, 2017-06-13:
Hello, this is how my directory structure looks:

Before installing library:
```
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ ls -hal
total 60K
drwxr-xr-x 15 chris chris 4.0K Jun 11 22:48 .
drwxr-xr-x  6 chris chris 4.0K Jun 11 20:54 ..
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 CapacitiveSensor
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 Ethernet
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 Firmata
[ ... data snipped out ... ]
```

After installing library Arduino Low Power:
```
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ ls -hal
total 64K
drwxr-xr-x 16 chris chris 4.0K Jun 12 22:16 .
drwxr-xr-x  6 chris chris 4.0K Jun 11 20:54 ..
drwxr-xr-x  3 chris chris 4.0K Jun 12 22:17 Arduino_Low_Power
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 CapacitiveSensor
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 Ethernet
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 Firmata
[ ... data snipped out ... ]
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 WiFi
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ ls -hal Arduino_Low_Power/
total 12K
drwxr-xr-x  3 chris chris 4.0K Jun 12 22:17 .
drwxr-xr-x 16 chris chris 4.0K Jun 12 22:16 ..
drwxr-xr-x  4 chris chris 4.0K Jun 12 22:17 1.1.0
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ ls -hal Arduino_Low_Power/1.1.0/
total 76K
drwxr-xr-x 4 chris chris 4.0K Jun 12 22:17 .
drwxr-xr-x 3 chris chris 4.0K Jun 12 22:17 ..
drwxr-xr-x 6 chris chris 4.0K Jun 12 22:17 examples
-rw-r--r-- 1 chris chris  797 Nov 30  1979 keywords.txt
-rw-r--r-- 1 chris chris  332 Nov 30  1979 library.properties
-rw-r--r-- 1 chris chris  26K Nov 30  1979 LICENSE
drwxr-xr-x 4 chris chris 4.0K Jun 12 22:17 src
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ 
```

After removing library Arduino Low Power:
```
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ ls -hal
total 64K
drwxr-xr-x 16 chris chris 4.0K Jun 12 22:16 .
drwxr-xr-x  6 chris chris 4.0K Jun 11 20:54 ..
drwxr-xr-x  2 chris chris 4.0K Jun 12 22:18 Arduino_Low_Power
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 CapacitiveSensor
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 Ethernet
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 Firmata
[ ... data snipped out ... ]
drwxr-xr-x  3 chris chris 4.0K Jun 11 20:54 WiFi
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ ls -hal Arduino_Low_Power/
total 8.0K
drwxr-xr-x  2 chris chris 4.0K Jun 12 22:18 .
drwxr-xr-x 16 chris chris 4.0K Jun 12 22:16 ..
chris@kaiju13:~/software/sloeber/arduinoPlugin/libraries$ ls -hal Arduino_Low_Power/1.1.0/
ls: cannot access 'Arduino_Low_Power/1.1.0/': No such file or directory
```
At this point, the error is thrown trying to open Add a library to project. I said earlier I manually deleted a directory to workaround the problem, the directory was `Arduino_Low_Power/`.

jantje, 2017-06-13:
Ok txs.
Thanks to eismeraldo the nightly should fix the symptom (handling empty library folder) I still need to fix the root cause (the fact that the folder is still there after a delete)

jantje, 2017-06-15:
Strange this issue is fixed but I run the latest nightly and it does not seem to be fixed.

closed 2017-06-17 by jantje