Sloeber/arduino-eclipse-plugin #760: V4.1 fails to install in Neon 4.6.3

DrObscure, 2017-06-04:
When installing from the nightly, I get issues from Eclipse concerning the dependencies as follows:
I have installed CDT 9.2.1.201704050430, and Sloeber lists 9.2.1.201703062208 as required, Eclipse aborts stating that the newer version is in place..
Also the plugin is requiring GNU ARM C/C++ J-Link Debugging, as well as the OpenOCD, PyOCD and QEMU debugging equivalents...  and failing to install them as well..  

Eclipse aborts the install without Sloeber in place..

DrObscure, 2017-06-04:
BTW: Just noticed your warning on the nightly page about running V3 and V4 on the same Eclipse.. 
Does this mean I should uninstall my V3 version before trying to complete the V4 install??

jantje, 2017-06-04:
>Does this mean I should uninstall my V3 version before trying to complete the V4 install??

Yes do not install V3 and V4 in the same eclipse

DrObscure, 2017-06-04:
That doesn't solve my issues, but glad i saw it before completion, and created others :) ...
However, you might want to make that more clear in your install information for those of us following up the from the V3 tree..  it was not obvious..

DrObscure, 2017-06-04:
BTW: I develop several tool chains in Eclipse, PERL, Php, etc. .. not just Arduino.. so having multiple instance of Eclipse around is not a desirable way for me to go..  and having to bring the others over to a new instance every time something new arrives seems a very clumsy process..  having and IDE for each manufacturer is bad enough.. maybe I just need more laptops to work from :)

jantje, 2017-06-04:
I'm in the same situation. java, php, html, lua, bash,c++ ....
What I do in case of a incompatible upgrade like this is copy my install remove V3 add V4
However this does not work with the product version as you can not remove the basis tht is why I do not document this procedure.
I know these incompatibilities are a pain, so I try to limit the incompatibilities to the minimum; but eclipse legal forced us to rename the plugin so all baeyens.it references changed to sloeber.io so there is no way to "upgrade" that.

DrObscure, 2017-06-04:
Ok.. I've uninstalled V3, so willing to take it on the chin this one time to get synced back with the team..
Tried the nightly V4.1 install again, with same results:
------------------------------
Your original request has been modified.
  "C/C++ Development Tools" will be ignored because a newer version is already installed. 
Cannot complete the install because one or more required items could not be found.
  Software being installed: GNU ARM C/C++ OpenOCD Debugging 4.1.4.201705270258 (ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group 4.1.4.201705270258)
  Missing requirement: GNU ARM C/C++ Debugging GDB 3.2.1.201705270258 (ilg.gnuarmeclipse.debug.gdbjtag 3.2.1.201705270258) requires 'bundle org.eclipse.cdt.debug.gdbjtag.core 8.2.0' but it could not be found
  Cannot satisfy dependency:
    From: GNU ARM C/C++ OpenOCD Debugging 4.1.4.201705270258 (ilg.gnuarmeclipse.debug.gdbjtag.openocd 4.1.4.201705270258)
    To: bundle ilg.gnuarmeclipse.debug.gdbjtag 0.0.0
  Cannot satisfy dependency:
    From: GNU ARM C/C++ OpenOCD Debugging 4.1.4.201705270258 (ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group 4.1.4.201705270258)
    To: ilg.gnuarmeclipse.debug.gdbjtag.openocd [4.1.4.201705270258]

jantje, 2017-06-04:
Please check your update sites:
These are the ones I have in the new install
[image]

DrObscure, 2017-06-05:
Thanks for this list.. also, for some strange reason, occasionally Eclipse deletes all my entries in the software update list after an update,, and i have lost some context because of that..  I'll add these and see if it can find all the pieces this time :)

DrObscure, 2017-06-05:
Ok Jantje, with the addition of the sites (likely the CDT 9.2 pointer) Sloeber 4.1 installed.. immediately giving me the well know error #705, which i will now visit,  and again watch the video to begin to upgrade my various workspace projects ..:)

jantje, 2017-06-05:
On windows stay to the root as close as possible.

DrObscure, 2017-06-05:
I see and realize the 40 character limit and moving the plugin is no problem, but how do I inform Eclipse of that new location? is there a property somewhere to handle it? didn't find it under the Arduino preferences..

Apparently I need some direction on how to accomplish this.. I tried creating a C:\Sloeber dir and added the tree eclipse\pulgins and copied the arduinoPlugin into this..  deleted the neon\plugin\arduinoPlugin dir and restarted Eclipse.. it came up with the #705 error and still shows the original dir that fails..  sigh..

DrObscure, 2017-06-05:
Ok.. moved the arduinoPlugin dir to c:\Sloeber..  
Also noticed that although I have removed the original neon\arduuinoPlugin dir, when Eclipse starts up it recreates it and then tells me the path is too long, etc..  and of course, does not look in the Sloeber dir instead..

jantje, 2017-06-05:
Why not simply moving the eclipse install folder?

DrObscure, 2017-06-05:
That is also easy to do.. just not clear that Eclipse bases operation relative to its launch dir... 
If so, then, that is a lot easier than fighting the system :)
I'll move the entire Eclipse dir tree from 'c:\program files (X86)' to c:\Sloeber..

DrObscure, 2017-06-05:
Well that was easy enough and Eclipse came up without all the complaining about the path length, or much of anything else...  
So now the task of updating the config for the projects & workspace..  
 --thanks again..

closed 2017-06-05 by DrObscure

jantje, 2017-06-05:
c:\program files (X86) is a bad place to install as it contains a space.
c:\Sloeber is a good place
Good behaving eclipse applications can be moved without any drawbacks. This is why Sloeber saves all its info in arduinoPlugin folder (the name is pre sloeber name, now I would have used Sloeber)
But I'm happy you are going again