Sloeber/arduino-eclipse-plugin #761: Update site not available

DareDevil73, 2017-06-05:
Hi, I cannot get updates by  the site update site http://eclipse.baeyens.it/update/V4/stable configured inside my IDE.

The response, got from a web browser, is:

> If you ended up here you took a shortcut and failed. Go back to start. Start is here installAdvice.shtml

Could you help me please?

jantje, 2017-06-05:
Thanks for the allert
I disabled http on the server 
So the update sites should point to https.
I'll reenable http for now

DareDevil73, 2017-06-05:
Now it works!

jantje, 2017-06-05:
I'm reopening this to make the update site point to https

jantje, 2017-06-05:
Main reason is that google requires https for all sites that requests passwords which I do for my gallerij and blog so I will be tagged as "untrustworthy" by google.
So I want to disable http on the site and serve only https and standard eclipse does not seem to be able to redirect (which seems ok to me from a security point of view)

jantje, 2017-06-05:
I also need to point to the https variant for the "what is new page"

DareDevil73, 2017-06-06:
Today the GNUARM repository site is not reachable.

Thank you!

jantje, 2017-06-17:
Strange, all my update sites are gone after upgrading to the last nightly :-(

jantje, 2017-06-29:
These are the "needed" update sites
jantje
For sloeber
https://eclipse.baeyens.it/update/V4/stable
https://eclipse.baeyens.it/update/V4/nightly

For Eclipse
http://download.eclipse.org/eclipse/updates/4.6
http://download.eclipse.org/releases/neon

For CDT
http://download.eclipse.org/tools/cdt/releases/9.2

For debugging functionality
http://gnuarmeclipse.sourceforge.net/updates

closed 2018-03-06 by jantje