Sloeber/arduino-eclipse-plugin #763: removing V3 from eclipse..

DrObscure, 2017-06-05:
I have moved to the V4 plugin, and I use the internal eclipse uninstall to remove V3.. however, the uninstall still left vestiges of the old V3 in the form of several "it.baeyens.arduino.xxx" files in the tree...
Can we assume that all these can be physically removed, even though the uninstall left them in place??

I'm getting an error from eclipse referencing one as creating a CoreException since the dir tree move..

jantje, 2017-06-05:
I think starting eclipse with -clean should clean these up

DrObscure, 2017-06-06:
-clean didn't remove much, so I just deleted them manually...
also, it appears I missed an old V3 project that was likely generating the Eclipse error..  once I upgraded its config the error went away as well...

closed 2017-06-06 by DrObscure

MaxRusse11, 2018-02-02:
Is there a means to remove this altogether from eclipse?