Sloeber/arduino-eclipse-plugin #764: example build fails in V4.1

DrObscure, 2017-06-06:
To test that I had V4.1 running correctly I tried to build a minimal project.  I choose the .cpp template, and attempted to build it as given.

The build failed in two ways.. 
1) 'make clean' failed, claiming no rule.
2) the build of the .cpp file failed due to a path munging of the '\' to a '/' at the arduinoPlugin dir and down the tree to the compiler... causing the compile to fail.
-------
22:33:25 **** Clean-only build of configuration Release for project sloeber_t2 ****
"C:\\Sloeber\\Eclipse\\neon\\arduinoPlugin\\tools\\make\\make" clean 
make: *** No rule to make target 'clean'.  Stop.

22:33:25 Build Finished (took 91ms)

22:33:26 **** Build of configuration Release for project sloeber_t2 ****
"C:\\Sloeber\\Eclipse\\neon\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber_t2.cpp'
'Starting C++ compile'
"C:\Sloeber\Eclipse\neon\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"C:\Sloeber\Eclipse\neon\arduinoPlugin\packages\arduino\hardware\avr\1.6.14\cores\arduino" -I"C:\Sloeber\Eclipse\neon\arduinoPlugin\packages\arduino\hardware\avr\1.6.14\variants\standard" -MMD -MP -MF"sloeber_t2.cpp.d" -MT"sloeber_t2.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber_t2.cpp"  -o  "sloeber_t2.cpp.o"
subdir.mk:18: recipe for target 'sloeber_t2.cpp.o' failed
process_begin: CreateProcess(NULL, C:\Sloeber\Eclipse\neon\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Sloeber\Eclipse\neon\arduinoPlugin\packages\arduino\hardware\avr\1.6.14\cores\arduino -IC:\Sloeber\Eclipse\neon\arduinoPlugin\packages\arduino\hardware\avr\1.6.14\variants\standard -MMD -MP -MFsloeber_t2.cpp.d -MTsloeber_t2.cpp.o -D__IN_ECLIPSE__=1 -x c++ ..\sloeber_t2.cpp -o sloeber_t2.cpp.o, ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [sloeber_t2.cpp.o] Error 2

22:33:26 Build Finished (took 613ms)

jantje, 2017-06-06:
Did you do project properties->arduino->apply->ok?

harisjmd, 2017-06-06:
Same thing happens to me..@jantje did that too but after Apply, OK button wasn't acting normally, wasn't closing the properties window.. I had either press cancel or exit button.

jantje, 2017-06-06:
I understand now.
 you will have to delete the arduinoplugin folder. The file structure between V3 and V4 has changed completely. I made it in line with arduino ide.
You can keep the downloads folder under arduinoplugin if you want to avoid redownloading

DrObscure, 2017-06-06:
ok, that was the problem.. I deleted the arduinoPlugin folder, restarted eclipse and it rebuilt the tree properly, Then as you stated above, I went back to the properties of each project and re-applied the Arduino configuration and board..  and we are good to go :)

closed 2017-06-06 by DrObscure

harisjmd, 2017-06-06:
(For my understanding) So by deleting arduinoplugin folder soebler will use arduino ide?
Thnx

jantje, 2017-06-06:
@harisjmd no

jantje, 2017-06-06:
To summarize the V3 to V4 plugin upgrade process that has been done here
1) move the eclipse location close enough to the root of the disk (windows only)
2) start eclipse and remove V3 plugin
3) Stop eclipse delete the ArduinoPlugin folder (you can keep the download folder)
4) start eclipse and install V4
5) reinstall boards and libraries
6) for all existing projects in the workspace open project properties and select the correct toolchain
7) for all existing projects in the workspace open project properties and select Arduino apply->ok

This is why install new V4 product is so much easier than upgrade