Sloeber/arduino-eclipse-plugin #767: Make not found in

Besama, 2017-06-13:
Hello,
I download Windows 32 bit bundle OK, unzip with  error: make....corrupt. 
In C:\C\sloeber\arduinoPlugin\tools\make they have 2 dll but not make.exe
I install eclipse OK but now I can't "verify". 
Please you will send me make.exe file.

Thank's

Mauricio

jantje, 2017-06-13:
There should be a make.zip in C:\C\sloeber\arduinoPlugin\downloads containing 3 dll's and make.exe. 
Extract all files to C:\C\sloeber\arduinoPlugin\tools\make
If the local zip file is corrupt you can download it from http://eclipse.baeyens.it/download/make.zip

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Besama, 2017-06-14:
Ok, I download it, extract and replace. Perfect.
Thank you very much, great plugin!

I don't know  Patreon  but I'll find this, to contribute with you. 
Best regards from Argentina,

Mauricio

closed 2017-06-14 by jantje

danobot, 2019-01-18:
Why is this not automated?

jantje, 2019-01-18:
> 
> 
> Why is this not automated?

It is automated. Something went wrong somewhere.

danobot, 2019-01-18:
I went back to stable, uninstalled all platforms and reinstalled platforms. I am now able to build and upload successfully. 
Many thanks for this. Been looking wanting to switch for a while