Sloeber/arduino-eclipse-plugin #772: Failed execution of avrdude for Adafruit Feather 32u4 (core=arduino:arduino)

ramonpoca, 2017-06-17:
Libraries are found ok in nightly (2017-06-17), but execution of avrdude fails with a:

    java.io.IOException: Cannot run program "avrdude" (in directory "/Users/ramon/Documents/sloeber-   
    workspace"): error=2, No such file or directory

[log.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/1082570/log.txt)

jantje, 2017-06-17:
I have D:\embedded\V4\sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude -CD:\embedded\V4\sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino8/etc/avrdude.conf -patmega32u4 -cavr109 -Pcom4 -b57600 -D -Uflash:w:D:\workspaces\embedded\adafruit_wiced/Release/adafruit_wiced.hex:i 

The fact you miss the path may be related to
1) not having arduino avr instaled in the sloeber boardsmanager
2) not having done project properties->arduino->apply->ok after a change (like installing arduino avr or upgrading to the nightly)
3) not having adafruit avr installed via the sloeber boardsmanager
4) others I don not know of or didn't think of.

Best regards 
Jantje
And thanks for being a patron :-)

jantje, 2017-06-17:
Not sure why I clicked close

ramonpoca, 2017-06-18:
1.- check, 2.- check, 3.- check.

There are two versions of avrdude installed under the arduino toolchain:
/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/arduino/tools/avrdude/6.0.1-arduino5/bin/avrdude
/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude

Might it happen just like with the libraries path, that there should be a executable tool path inheritance?

ramonpoca, 2017-06-18:
Pasting project org.eclipse.cdt.core.prefs:

```
eclipse.preferences.version=1
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ALT_SIZE_COMMAND/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ALT_SIZE_COMMAND/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ALT_SIZE_COMMAND/value="${A.COMPILER.PATH}${A.COMPILER.SIZE.CMD}" --format\=avr --mcu\=${A.BUILD.MCU} "${A.BUILD.PATH}/${A.BUILD.PROJECT_NAME}.elf"
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ARCHIVE_FILE/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ARCHIVE_FILE/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ARCHIVE_FILE/value=arduino.ar
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ARCHIVE_FILE_PATH/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ARCHIVE_FILE_PATH/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.ARCHIVE_FILE_PATH/value=${A.BUILD.PATH}/${A.ARCHIVE_FILE}
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.EXTENDED_FUSES/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.EXTENDED_FUSES/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.EXTENDED_FUSES/value=0xcb
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.FILE/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.FILE/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.FILE/value=caterina/Caterina-Feather32u4.hex
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.HIGH_FUSES/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.HIGH_FUSES/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.HIGH_FUSES/value=0xd8
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.LOCK_BITS/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.LOCK_BITS/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.LOCK_BITS/value=0x2F
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.LOW_FUSES/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.LOW_FUSES/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.LOW_FUSES/value=0xff
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.TOOL/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.TOOL/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.TOOL/value=arduino\:avrdude
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.UNLOCK_BITS/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.UNLOCK_BITS/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BOOTLOADER.UNLOCK_BITS/value=0x3F
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.ARCH/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.ARCH/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.ARCH/value=AVR
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.BOARD/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.BOARD/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.BOARD/value=AVR_FEATHER32U4
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.CORE.PATH/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.CORE.PATH/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.CORE.PATH/value=${A.RUNTIME.PLATFORM.PATH}/cores/${A.BUILD.CORE}
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.CORE/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.CORE/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.CORE/value=arduino\:arduino
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.EXTRA_FLAGS/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.EXTRA_FLAGS/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.EXTRA_FLAGS/value=${A.BUILD.USB_FLAGS}
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.F_CPU/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.F_CPU/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.F_CPU/value=8000000L
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.MCU/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.MCU/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.MCU/value=atmega32u4
environment/project/io.sloeber.core.toolChain.release.1570554354/A.BUILD.PATH/delimiter=\:
[... cut ...]
environment/project/io.sloeber.core.toolChain.release.1570554354/JANTJE.WORKSPACE_LOCATION/value=
environment/project/io.sloeber.core.toolChain.release.1570554354/PATH/delimiter=\:
environment/project/io.sloeber.core.toolChain.release.1570554354/PATH/operation=replace
environment/project/io.sloeber.core.toolChain.release.1570554354/PATH/value=${A.COMPILER.PATH}${PathDelimiter}${A.BUILD.GENERIC.PATH}${PathDelimiter}${PATH}
environment/project/io.sloeber.core.toolChain.release.1570554354/append=true
environment/project/io.sloeber.core.toolChain.release.1570554354/appendContributed=true
```

jantje, 2017-06-18:
I have the impression there are old values remaining in the settings.
Seems like they obstruct the "working".
Make a copy of org.eclipse.cdt.core.prefs (or better the whole workspace to be sure)
Goto project properties->C/C++ build->environment and delete all entries starting with JANTJE
select ok (I got some errors)
Goto project properties->arduino and reconfigure

ramonpoca, 2017-06-19:
I tried but didn't work. Next I tried deleting all entries and reconfiguring. Again it didn't work.
I uninstalled all the adafruit boards and configured the project for an Arduino Uno. It built but avrdude failed the same way.
Environment for avrdude seems coherent.

Creating a blank project and uploading it also fails.

jantje, 2017-06-19:
What I think is strange is that nowhere in the provided settings file avrdude is on its own. So I don't understand how you have only the avrdude command ant not something like bin/avrdude

jantje, 2017-06-19:
The uno story is strange. There is no redirection when using uno so that should work fine.
Is this a pre V4.0 upgraded eclipse?
Would it be possible to connect to your system remotely to have a look?

ramonpoca, 2017-06-19:
Nay. I've even installed the last complete nightly (io.sloeber.product-4.1.0-20170616.121957-42-macosx.cocoa.x86_64.tar.gz), installed Arduino boards, Adafruit boards, created an empty .ino, clicked upload with the same result. Also using Arduino/Genuino UNO as the board.

Only one avrdude at: arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude. 
It executes from the command line. I've even tried to manually set in C/C++ Environment:
    A.TOOLS.AVRDUDE.CMD.PATH=/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude

And the same error happens.

java.io.IOException: Cannot run program "avrdude" (in directory "/Users/ramon/Documents/sloeber-workspace"): error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at io.sloeber.core.tools.ExternalCommandLauncher.launch(ExternalCommandLauncher.java:294)
	at io.sloeber.core.tools.uploaders.GenericLocalUploader.RunConsoledCommand(GenericLocalUploader.java:42)
	at io.sloeber.core.tools.uploaders.arduinoUploader.uploadUsingPreferences(arduinoUploader.java:77)
	at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:178)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 5 more

jantje, 2017-06-19:
I have been looking into the code here and I really don't see where it goes wrong.
I suspect a casing issue (given windows doesn't case) but with the uno ....
Could you do a debug session?
The problem must be in BoardDescriptor.getUploadCommand

jantje, 2017-06-19:
I have changed some code to go to uppercase.
I'm setting up a linux to see whether the same problem exists there

Steffen-Engel, 2017-06-20:
I have the same problem.

as some additional information:
loading firmware with an older sloeber release works.
then loading with 4.1 works.

loading again with 4.1 it does not work:

version 1, same Port number for bootloader and application:

```
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM6
PORTS { COM3, COM5, COM6,} / { COM3, COM5, COM6,} => {}
PORTS { COM3, COM5, COM6,} / { COM3, COM5, COM6,} => {}

[...]

PORTS { COM3, COM5, COM6,} / { COM3, COM5, COM6,} => {}
PORTS { COM3, COM5, COM6,} / { COM3, COM5, COM6,} => {}
PORTS { COM3, COM5, COM6,} / { COM3, COM5, COM6,} => {}
PORTS { COM3, COM5, COM6,} / { COM3, COM5,} => {}
PORTS { COM3, COM5, COM6,} / { COM3, COM5, COM6,} => {}
Comport appeared and disappeared with same name
Using comportCOM6 from now onwards
Ending reset

LaunchingC:\usr\sloeber_4_1\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino2/bin/avrdude -CC:\usr\sloeber_4_1\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino2/etc/avrdude.conf -patmega32u4 -cavr109 -PCOM6 -b57600 -D -Uflash:w:C:\Daten\Desktop\src\Ardora\Apollo/Release/Apollo.hex:i 
Output:

Connecting to programmer: .

```

When port number is not the same, it even does not use the changed port for upload:

```
using arduino loader
Forcing reset using 1200bps open/close on portCOM9
PORTS { COM3, COM5, COM9,} / { COM3, COM5, COM9,} => {}
PORTS { COM3, COM5, COM9,} / { COM3, COM5, COM6,} => { COM6,}
Comport reset took500ms
Using comportCOM6 from now onwards
Ending reset

LaunchingC:\usr\sloeber_4_1\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CC:\usr\sloeber_4_1\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -PCOM9 -b57600 -D -Uflash:w:C:\Daten\Desktop\src\Ardora\Apollo/Release/Apollo.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM9": Das System kann die angegebene Datei nicht finden.

```

Steffen-Engel, 2017-06-20:
Ok, for me it seems to be a half Arduino half Sloeber item: the test code without proper reboot was a sketch without use of the serial port.

Adding serial.begin in setup() made it work.

The problem with a change of the port number still exist

ramonpoca, 2017-06-20:
Success!
Current nightly fixed the problem!

jantje, 2017-06-20:
@ramonpoca 
So it seemed a casing problem. As such linux and mac only.

@Steffen-Engel 
Your problem is different as in your case the command is there but the used com port is wrong. Can you create a new issue specifying the board and json file?

closed 2017-06-20 by jantje