Sloeber/arduino-eclipse-plugin #773: Arduino Star OTTO

petus, 2017-06-19:
Hi,
at first, thanks for your sloeber. 
At second, can you update library for Arduino STAR OTTO (STM32F4) ? Your version of library doesn't work. After uploading I see "device descriptor read/64, error -110", I have to manualy switch to bootloader mode.
In Arduino IDE by Arduino.org works well. Thanks a lot.

jantje, 2017-06-19:
I have no clue what you are talking about.

petus, 2017-06-20:
OK, I try to explain it better.
I have Arduino Star Otto (arduino board with STM32F4) and I tried Arduino IDE by Arduino.org and sloeber on Linux Mint 64 bit.
The board worked fine in Arduino IDE - when I connected the board to computer I see connected board in terminal, when I uploaded the code, the board is switched to bootloader mode and the code is successfuly uploaded.
When I tried sloeber, the eclipse console showed "the code is uploaded", but the same code didn't work. 
So I checked linux terminal and I saw this error message "_device descriptor read/64, error -110_".
And Otto board is not connected to my laptop.
When I wanted to repair connection with my laptop, I had to upload the code from Arduino IDE by Arduino.org and also manualy switch the board to bootloader mode.
I compared files for otto board on Arduino-org github with files which sloeber downloaded and the "library" of board is not actual. So I think that the problem is old version of otto library.

jantje, 2017-06-20:
Which json did you use to install this board?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

petus, 2017-06-20:
1.0.1 version from Sloeber (v4.1) -> Preference -> Arduino -> Board manager (I think)

I wrote article about sloeber available on https://time4ee.com/news.php?readmore=137 and also on chiptron.cz - for czech/slovak makers.
And I want to write articles about OTTO with Sloeber, because that's better than Arduino IDE. So, Sloeber will be mentioned of course.

jantje, 2017-06-20:
I'm asking for the json as I can't find this OTTO board. As there are 400+ boards supported I don't feel like searching for it.

petus, 2017-06-20:
Sorry, but I don't know what do you think. I install Arduino star otto through sloeber. Preference -> Arduino -> Platform and Boards -> arduino -> Arduino STM32F4 Boards -> 1.0.1
I did not have to use json file.
Arduino library is on https://github.com/arduino-org/arduino-core-stm32f4

jantje, 2017-06-29:
Sorry for the delay. I missed your previous message.
That is indeed the info I needed to continue.
I installed the board and compiled a project (took 2 minutes to build which is very long). I can not test the upload but I see that -on my system- sloeber tries to reset the board and runs following upload command
`D:\embedded\sloeber\/arduinoPlugin/packages/arduino/tools/arduinoSTM32load/2.0.0/arduinoSTM32load -dfu D:\embedded\sloeber\/arduinoPlugin/packages/arduino/tools/dfu-util/0.9.0-arduino1 -bin D:\workspaces\embedded\hh/Release/hh.bin -port="com1" -v `

Can you share your the upload comand on sloeber and on the arduino IDE?

jantje, 2017-07-04:
Any input for me?

petus, 2017-07-04:
Sorry for delay, but I'm out of computer. I will write you later.

jantje, 2017-07-08:
I have been doing upload test lately and I found no more issues. So I hope this one can be marked as closed soon as well.

petus, 2017-07-09:
Here are outputs of sloeber and arduino IDE.
Sloeber:

> Starting upload
using arduino loader
Forcing reset using 1200bps open/close on port/dev/ttyACM0
Continuing to use/dev/ttyACM0
Ending reset
Launching/home/petus/Plocha/sloeber//arduinoPlugin/packages/arduino/tools/arduinoSTM32load/2.0.0/arduinoSTM32load -dfu /home/petus/Plocha/sloeber//arduinoPlugin/packages/arduino/tools/dfu-util/0.9.0-arduino1 -bin /media/petus/Data/OwnCloud/IDE-workspace/Sloeber_arduino_workspace/OTTO_LED/Release/OTTO_LED.bin -port="/dev/ttyACM0" -v 
Output:
arduinoSTM32load 2.0.0 - compiled with go1.2.1
Starting download script...
Serial Port: "/dev/ttyACM0"
BIN FILE /media/petus/Data/OwnCloud/IDE-workspace/Sloeber_arduino_workspace/OTTO_LED/Release/OTTO_LED.bin
Unix/Linux type OS detected
/bin/stty: '"/dev/ttyACM0"': Adresář nebo soubor neexistuje
Waiting for device...
Device found!
dfu-util 0.9
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 29524
Download	[                         ]   0%            0 bytes
Download	[                         ]   0%            0 bytes
Download	[=                        ]   6%         2048 bytes
Download	[===                      ]  13%         4096 bytes
Download	[=====                    ]  20%         6144 bytes
Download	[======                   ]  27%         8192 bytes
Download	[========                 ]  34%        10240 bytes
Download	[==========               ]  41%        12288 bytes
Download	[============             ]  48%        14336 bytes
Download	[=============            ]  55%        16384 bytes
Download	[===============          ]  62%        18432 bytes
Download	[=================        ]  69%        20480 bytes
Download	[===================      ]  76%        22528 bytes
Download	[====================     ]  83%        24576 bytes
Download	[======================   ]  90%        26624 bytes
Download	[======================== ]  97%        28672 bytes
Download	[=========================] 100%        29524 bytes
Download done.
File downloaded successfully
Resetting STM32, starting firmware at address 0x08000000...
Successfully reset STM32
SUCCESS: Sketch will execute in about 5 seconds.
arduinoSTM32load finished
upload done

Arduino IDE
> arduinoSTM32load 2.0.0 - compiled with go1.2.1
Starting download script...
Serial Port: "/dev/ttyACM0"
BIN FILE /tmp/arduino_build_832220/Blink.ino.bin
Unix/Linux type OS detected
/bin/stty: '"/dev/ttyACM0"': Adresář nebo soubor neexistuje
Waiting for device...
Device found!
dfu-util 0.9
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 34112
Download	[                         ]   0%            0 bytes
Download	[=                        ]   6%         2048 bytes
Download	[===                      ]  12%         4096 bytes
Download	[====                     ]  18%         6144 bytes
Download	[======                   ]  24%         8192 bytes
Download	[=======                  ]  30%        10240 bytes
Download	[=========                ]  36%        12288 bytes
Download	[==========               ]  42%        14336 bytes
Download	[============             ]  48%        16384 bytes
Download	[=============            ]  54%        18432 bytes
Download	[===============          ]  60%        20480 bytes
Download	[================         ]  66%        22528 bytes
Download	[==================       ]  72%        24576 bytes
Download	[===================      ]  78%        26624 bytes
Download	[=====================    ]  84%        28672 bytes
Download	[======================   ]  90%        30720 bytes
Download	[======================== ]  96%        32768 bytes
Download	[=========================] 100%        34112 bytes
Download done.
File downloaded successfully
Resetting STM32, starting firmware at address 0x08000000...
Successfully reset STM32
SUCCESS: Sketch will execute in about 5 seconds.

petus, 2017-07-09:
hmm, interesting. When STAR OTTO board is working and I try upload the code through sloeber, I see this error message. 
[image]

But the code is uploaded by sloeber successfully. At the end of uploading, the board is not working. I have to upload the code by Arduino IDE and then I can work with STAR OTTO board.

jantje, 2017-07-09:
If I understand correctly there are 2 issues
1) the reset of the STAR OTTO fails in sloeber. It seems this does not impact the upload though.
2) A project build by sloeber uploaded to STAR OTTO does not run on the START OTTO
I can't investigate the first issue as I don't have the hardware.
I can investigate the second issue but it is low priority for me right now.

petus, 2017-07-10:
That's a pity. I plan to write a few parts of tutorial how to use arduino star otto, because star otto is not on sell.
I tried to upload the code only through arduinostm32load utility and the problem looks like compilation by sloeber. I will try other way how to find workaroind of this problem. Thanks for your support and your time.

jantje, 2017-07-10:
If you find the compile difference between arduino ide and sloeber I will look at it.

jantje, 2017-11-23:
Is there a update on this issue?

petus, 2017-11-24:
No. But I didn't try the latest version of sloeber. I tried other STM32 nucleo boards with github.com/stm32duino repository and it didn't work too.

jantje, 2017-11-24:
> I tried other STM32 nucleo boards with github.com/stm32duino repository and it didn't work too.

With the latest version 4.2?

petus, 2017-11-24:
With v4.1

jantje, 2017-11-24:
my paypal account is jan@baeyens.it

petus, 2017-12-02:
Ok, I tried the newest Sloeber (4.2) and arduino boards with STM32 didn't work.

wimjongman, 2017-12-04:
[image]

wimjongman, 2017-12-04:
I have installed a Nucleo-64 (NUCLEO-F401RE) and it apart from the problems mentioned earlier it uploads and runs fine.

[image]

closed 2018-01-24 by jantje