Sloeber/arduino-eclipse-plugin #774: Saving changes to Project->Properties->Arduino->Board Selection gets serialization error and corrupts entire workspace

yaoyogi, 2017-06-28:
Eclipse Neon.3 Release (4.6.3) - Build id: 20170314-1500
Plugin: Sloeber	4.1.0.201705252335	io.sloeber.feature.feature.group	Jan Baeyens and Others
MAC OS X Sierra 10.12.2

The project is an "Arduino Sketch Project"

Right click project -> Properties -> Arduino.  I modify the "Arduino board selection" tabs' values and click "Apply", got: Internal error serialization.  From this point forward, I can no longer apply/update the projects Arduino properties. 

This also, somehow broke the ability for the project to be built and got obscure error regarding make:
    Errors occurred during the build.
    Errors running builder 'Scanner Configuration Builder' on project 'testproj'.
    0
[image]

The build was working prior to this error.   I had to abandon the project since it can no longer get updated or built.

[image]

The problem seems to have corrupted the entire Eclipse workspace.  If I try to create a new Arduino Sketch Project I get a "Multiple problems have occurred" error dialog, error value: 0 containing 5 entries of: "Serializing CDT Project settings.."

Even if I close the project that had the original serialization error, I cannot edit/save the properties of all my other Arduino Sketch projects.  
Even after deleting the project with the original serialization error the workspace is still corrupted and am unable to edit/save any sketches Arduino properties.

I have no other option than to delete/abandon the entire workspace.

jantje, 2017-06-29:
I have seen this before but I can't recall exactly what causes the issue.
You get a "serialization error " for each sloeber project.
If you show me the details I probably can remember what needs to be done to fix the issue.
But try: for all projects int eh workspace : properties->arduino->apply->ok (note you need to do apply and ok)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

yaoyogi, 2017-06-29:
Thank your for quick reply.  I did try apply/ok for each project but did not help.  Note that even deleting the projects so there are no projects in the workspace at all did not fix the problem.  Creating a new project would result in the same errors.   I think the problem was related to processing the added libraries vs. just updating the Arduino properties which I was initially able to do a few times successfully "before" adding some libraries.  Also, when adding the libraries, I needed to do an Eclipse Restart to get them properly processed/recognized.  I see from searches that this has confused people.

When folks submit issues they should submit the Eclipse ../.metadata/.log file to help you diagnose problems.  I did copy a few of the errors and will include them with this update, but unfortunately I deleted the physical workspace files.

Allowing developers to use their existing Eclipse C/C++ environment and just use your cool plugin will allow you to get more adopters vs. having to use the packaged "product" version.  I understand the packaged version allows greater control from your standpoint, minimizes environment variations, etc... but for some shops that won't work out.

I appreciate your efforts and understand you're an independent open source project and will look at supporting via Patreon.

[slober_eclipse_log_fragments.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/1113206/slober_eclipse_log_fragments.txt)

jantje, 2017-06-29:
>Allowing developers to use their existing Eclipse C/C++ environment and just use your cool plugin will allow you to get more adopters vs. having to use the packaged "product" version.

As explained on the website http://eclipse.baeyens.it/installAdvice.shtml you can install as a plugin or as the product. I advice people new to Sloeber to install the product so they at least know what to look for.

As the workspace is gone there is little interest in keeping this issue open so I close it.

closed 2017-06-29 by jantje

yaoyogi, 2017-06-29:
I did send fragments from the .log file which show some issues like ArrayIndexOutOfBoundsException and other failures which seem possibly relevant.  Looking at the plugin source, the reattachLibrariesToProject(Project) triggered the out of bounds exception.  I did use the reattach libs to project thinking it might help the recently added libs to show in the UI.  This was close in time to when I saw the serialization errors but other interleaving operations may have occurred.

I don't see the same errors so far in the "product" bundling that I did with plugin which is another hint possibly to what's going on.

The stack traces I am seeing in the .log are not matching up with what's currently on the head in GIT so I don't have the exact same source for what I am seeing in the plugin or the product builds I am using.

What info/details would help you diagnose such issues?  Do you want a zip of the entire workspace?

Thank you for pointing out the advice link but I am already running as plugin and product.  My comment was that if people can't use the plugin due to the entire workspace being corrupted it could limit people adopting your work; perhaps I was just stating the obvious.

Thank you for efforts and response.

     MrP

jantje, 2017-06-30:
The indexoutofbounds is fixed in #755 
I see no other issues.