Sloeber/arduino-eclipse-plugin #775: .ino.cpp raises an error that the Arduino IDE is happy with

danilius, 2017-06-29:
Sorry if this is the wrong place to ask but I've not found anywhere more likely to find an answer and I'm a noob to Arduino, C++ and sloeber so it's probably a bit silly. Do please point me to somewhere better if it exists!

The following was declared in the .ino:
`struct time_slot {`
`  int aTime;`
`  bool onOff;`
`};`

This function uses it:

`void readtime_slot(unsigned int aSlotPos, time_slot &aSlot){`
`  //do something fascinating`
`}`

The Arduino IDE compiles it quite happily. However sloeber complains:

`'time_slot' has not been declared	.ino.cpp	/sk_v1	line 18	C/C++ Problem`
`'time_slot' was not declared in this scope	.ino.cpp	/sk_v1	line 17	C/C++ Problem`
`make: *** [.ino.cpp.o] Error 1	sk_v1		 	C/C++ Problem`
`recipe for target '.ino.cpp.o' failed	subdir.mk	/sk_v1/Release	line 24	C/C++ Problem`
`variable or field 'writetime_slot' declared void	.ino.cpp	/sk_v1	line 17	C/C++ Problem`

Where have I gone wrong?

jantje, 2017-06-29:
You didn't do anything wrong. The ino support in Sloeber is not as good as it is in Arduino IDE.
Even though I did lots of improvements in the nightly for ino support this situation is to complex to solve in a -what I call- good way.
The work around for you isd to work in a c/c++ way and put your declarations in a header file.
In other words: to make a headerfile named myproject.h with following content:
```
#pragma once
#include "Arduino.h"
struct time_slot {
int aTime;
bool onOff;
};
```

and add 
`#include "myproject.h"`
to you .ino file
and remove
```
struct time_slot {
int aTime;
bool onOff;
};
```
from your .ino file

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

danilius, 2017-06-29:
Well, that's a comprehensive and impressive response. Will give that a whirl. Many thanks for that.

danilius, 2017-06-29:
And of course it worked a treat, thanks again!

jantje, 2017-06-29:
Please consider supporting.
Best regards
Jantje

closed 2017-06-29 by jantje

uzi18, 2017-07-23:
@jantje : it works like a charm and also it not break compilation in Arduino IDE.
do You remember why sloeber.ino.cpp is not simple file with just include of project.ino?
Why it redefines all #define from ino file and double include all #include "something.h"?
Is it a bug?

pischky, 2022-04-09:
Please note:
You will have the same problems when using "using a=b" or "typedef" and with return types declared in the *.ino file.