Sloeber/arduino-eclipse-plugin #776: There are no projects selected in the project explorer or all selected projects are not arduino projects. Are you in the Arduino perspective?

jief666, 2017-06-29:
Hi,

I got this message, whatever I do (selecting different project, restart Sloeber.app, closing and reopening project). Try to look in logs but nothing more than this message.

Anyone an idea ?

jantje, 2017-06-29:
Are you in the Arduino perspective?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-06-29 by jantje

jief666, 2017-06-29:
Yes, of course I am. It was written in the message, so I checked.

jief666, 2017-06-29:
I've deleted sloeber.app and re-unzip, to restart fresh, guessing something went wrong in eclipse.
I've reimported my projetcs. But the 1.6.18 library is gone and I got a lot of ......Contents/Eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.18/cores no such file or directory

jief666, 2017-06-29:
Ok, I found where to download the previous 1.6.18. If I want to migrate my project to 1.6.19 how should I do ?

jief666, 2017-06-29:
Ok, I found where to download 1.6.18 and where to select which one I want.
In the end, I don't know what happened when Sloeber.app starts not to recognize my projects. Yesterday was fine. I relaunched Sloeber today and it was broken. I didn't move or rename anything, didn't install or update. No idea...

jantje, 2017-06-29:
When you reinstall you get the latest versions so that will be 1.6.19 so it is normal 1.6.18 is not found.
Did you upgrade from 3.X to 4.X?

jief666, 2017-06-29:
No upgrade. I am on the exact same version. I always keep the zip I've downloaded. So no change of version at all.

Yes, I guess it was ok that it gets the latest version. When I imported the project, it could have asked "Your project use 1.6.18, I have 1.6.19, would you like to download 1.6.18 or migrate project to 1.6.19 ?"

Or at least when I tried to build, it could have said that it can't build because of missing version of Arduino SDK.

But I guess there's plenty of other things to think of...
Thanks for answering.
Regards.

jantje, 2017-06-29:
good it is solved :-)

jief666, 2017-06-29:
I wouldn't really called that "solved" as it can happen again anytime. Projects weren't not seen as arduino project even if they were. Sorry to say, but that means a design problem somewhere.

jantje, 2017-06-30:
I really have no clue what happened and I never heard something similar before.
So unless we can reproduce the issue I wouldn't know how to fix the bug.

Defozo, 2017-09-28:
I think I have the same problem. Compilation works just after I create a new Arduino project but after I restart Sloeber I get the error and I don't know what to do?

jantje, 2017-09-28:
@Defozo can you make a video of this issue? I think that will be the easiest way to get this issue going.

HBH-THU, 2018-11-08:
@jief666 i solve this problem, just because i select the project item in "Project Explorer" not in "Navigator". you can try if this solution works.

hansmuc, 2023-03-21:
had another"not slober project" in the same workspace", After deletion it works again.