Sloeber/arduino-eclipse-plugin #777: Cannot create a new arduino project

gitgit3, 2017-06-30:
Installed the eclipse plugin for Arduino IDE using this link - http://eclipse.baeyens.it/update/V4/stable.
Tried to create a new Arduino project. Went to File->New->Project, and then Arduino->'New Arduino Sketch'. Had to click the 'Next' button twice to get through to the next dialog. Chose a project name, 'Generator_Controller' using the default location and then click 'Next'. The 'Finish' button is disabled. No matter what I tried, I cannot get past this dialog. So unable to create a new sketch.

Eclipse version:
Version: Luna Service Release 2 (4.4.2)
Build id: 20150219-0600
(64-bit)

Running on Linux Mint
Linux johnp-Aspire-V5-552P 3.13.0-24-generic #47-Ubuntu SMP Fri May 2 23:30:00 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

jantje, 2017-06-30:
Can you show what the sceen looks like when you can't get a finish?

gitgit3, 2017-06-30:
[image]

jantje, 2017-06-30:
clik the next button?

gitgit3, 2017-06-30:
Tried that, it does nothing.

jantje, 2017-06-30:
strange something is there something in the error log?

gitgit3, 2017-06-30:
Could this be the problem?

!ENTRY io.sloeber.arduino.core 1 0 2017-06-30 21:45:06.877
!MESSAGE The folder/home/johnp/Arduino/hardware' does not contain any files.

gitgit3, 2017-06-30:
I tried uninstalling the plugin, installing the CDT (which I thought I already had but didn't) and then installed Sloeber again. Still the same problem.

gitgit3, 2017-06-30:
I found a stack trace in the log which occurs for each press of the next button:

!ENTRY org.eclipse.ui 4 0 2017-06-30 23:11:47.638
!MESSAGE Unhandled event loop exception
!STACK 0
java.lang.NoSuchMethodError: org.eclipse.swt.widgets.Composite.requestLayout()V
	at io.sloeber.ui.project.properties.BoardSelectionPage.enableControls(BoardSelectionPage.java:306)
	at io.sloeber.ui.project.properties.BoardSelectionPage.draw(BoardSelectionPage.java:255)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizardBoardPage.createControl(NewSketchWizardBoardPage.java:51)
	at org.eclipse.jface.wizard.WizardDialog.updateForPage(WizardDialog.java:1276)
	at org.eclipse.jface.wizard.WizardDialog.access$4(WizardDialog.java:1268)
	at org.eclipse.jface.wizard.WizardDialog$8.run(WizardDialog.java:1257)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:70)
	at org.eclipse.jface.wizard.WizardDialog.showPage(WizardDialog.java:1254)
	at org.eclipse.jface.wizard.WizardDialog.nextPressed(WizardDialog.java:942)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:434)
	at org.eclipse.jface.dialogs.Dialog$2.widgetSelected(Dialog.java:619)
...more

jantje, 2017-06-30:
have you tried the product before installing the plugin?
That is lots easier

jantje, 2017-07-04:
Any progress here?

gitgit3, 2017-07-04:
Well, I having logged it as bug, I was waiting for it to be assigned to a developer to reproduce and ultimately fix! Much as I would love to get involved, I currently have very little time.

And I'm not sure what you mean by the 'product' ? If you mean the Arduino IDE, then yes I've been using it for a month or two. Currently, I'm using Eclipse with a C++ project and including some of the Arduino libs along with my source code. I'm using the Arduino IDE to compile and upload. I'll probably muddle along like that until someone fixes the issue :)

jantje, 2017-07-04:
I mean did you download your version from here? http://eclipse.baeyens.it/stable.php?OS=Linux
This is one download inclusive eclipse and ready to roll.
I fear you are missing some components.

gitgit3, 2017-07-05:
Yes I went to that page but chose the install plugin link just below the download buttons. I would not normally have more than one Eclipse install. I have my installation setup with the correct bench-layout, syntax highlighting, hot-keys etc.. It would be more useful to know which components I am missing.

jantje, 2017-07-05:
We consider the plugin install advanced usage. Advanced users are supposed to know eclipse and code well enough to find out what which components are missing.
As to trouble shooting having seen sloeber working and understanding what to expect is a great way to help analyse the missing parts. So installing sloeber in a test location and getting a feel is the advised way to go.
Don't forget:

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-07-08:
I'm closing this issue as a usage error. Please reopen if you can reproduce the problem with the product.

closed 2017-07-08 by jantje