Sloeber/arduino-eclipse-plugin #778: Can't upload to Leonardo from V4.1 plugin, but upload from Arduino IDE 1.6.11 is working OK.

MAWMN, 2017-07-02:
I am not able to uplad a sketch to my Leonardo board, with the default programmer. (AVRDUDE)
Monitor port is COM10, changes to COM9 when the reset is performed for upload. 
I am using Windows 7 (x64) See info below..

[image]

**Error Message:**
eclipse.buildId=unknown
java.version=1.8.0_131
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=nl_NL
Framework arguments:  -perspective io.sloeber.application.perspective -perspective io.sloeber.application.perspective -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective -perspective io.sloeber.application.perspective -data file:/E:/Eclipse_Sloeber_WS/ -perspective io.sloeber.application.perspective

io.sloeber.arduino.core
Error
Sun Jul 02 18:06:45 CEST 2017
Serial port COM10 not found. Did you select the right one from the project properties -> Arduino -> Arduino?

jssc.SerialPortException: Port name - COM10; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at io.sloeber.core.api.Serial.connect(Serial.java:169)
	at io.sloeber.ui.monitor.views.SerialMonitor.ResumePort(SerialMonitor.java:546)
	at io.sloeber.core.api.SerialManager.StartSerialMonitor(SerialManager.java:48)
	at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:189)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

**Console output:**
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM10
PORTS { CNCA0, CNCB0, COM1, COM5, COM7, COM8, COM10, COM25,} / { CNCA0, CNCB0, COM1, COM5, COM7, COM8, COM10, COM25,} => {}
PORTS { CNCA0, CNCB0, COM1, COM5, COM7, COM8, COM10, COM25,} / { CNCA0, CNCB0, COM1, COM5, COM7, COM8, COM9, COM25,} => { COM9,}
Comport reset took500ms
Using comportCOM9 from now onwards
Ending reset

LaunchingE:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CE:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -PCOM10 -b57600 -D -Uflash:w:E:\Eclipse_Sloeber_WS\First_Test/Release/First_Test.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM10": Het systeem kan het opgegeven bestand niet vinden.

avrdude done.  Thank you.

/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude finished
upload done

hope this can be solved..

Marcel

jantje, 2017-07-02:
Have you tried the nightly? I did some fixes in this area.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

MAWMN, 2017-07-02:
I don't know, I downloaded the complete package "V4.1_win64.2017-05-25_16-52-16.tar.gz" from https://github.com/Sloeber/arduino-eclipse-plugin/releases.

If this is not the correct one how can I change / update or fix this, or get and install the nightly build.

Marcel

jantje, 2017-07-02:
Add the nightly update site to the update sites and do help->check for updates
The nightly update site is http://eclipse.baeyens.it/update/V4/nightly

MAWMN, 2017-07-03:
Hello,
  after updating and restart the problem is still the same..
[image]
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM10
PORTS { CNCA0, CNCB0, COM1, COM7, COM8, COM10, COM25,} / { CNCA0, CNCB0, COM1, COM7, COM8, COM10, COM25,} => {}
PORTS { CNCA0, CNCB0, COM1, COM7, COM8, COM10, COM25,} / { CNCA0, CNCB0, COM1, COM7, COM8, COM9, COM25,} => { COM9,}
Comport reset took500ms
Using comportCOM9 from now onwards
Ending reset

LaunchingE:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CE:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -PCOM10 -b57600 -D -Uflash:w:E:\Eclipse_Sloeber_WS\First_Test/Release/First_Test.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM10": Het systeem kan het opgegeven bestand niet vinden.

avrdude done.  Thank you.

/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude finished
upload done

Marcel

jantje, 2017-07-04:
Should be fixed in the nightly tomorrow

MAWMN, 2017-07-06:
Hello,

   just updated with the nightly build, everything is working fine now..

thanks for the support..

Marcel

closed 2017-07-06 by MAWMN

jantje, 2017-07-06:
Thanks for the confirmation.