Sloeber/arduino-eclipse-plugin #779: Can't upload the code: wait for the 'arduino installer job' to finish

FlorisVO, 2017-07-03:
I can no longer upload the code to the Arduino. (In the previous version everything was fine)
[image]
But when i look at progress tab or the right bottom of the eclipse window, the 'arduino installer job' is not running/pending.

**-- Error Details --**
Date: Mon Jul 03 14:48:56 CEST 2017
Message: Please wait for the 'arduino installer job' to finish. Look at right bottom of the eclipse window to see if it is still running.
Severity: Error
Product: Eclipse 4.6.3.20170314-1500 (org.eclipse.epp.package.java.product)
Plugin: io.sloeber.core.ui
Sloeber version: 4.1.0.201705252335
Session Data:
eclipse.buildId=4.6.3.M20170301-0400
java.version=1.8.0_131
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=nl_NL
Framework arguments:  -product org.eclipse.epp.package.java.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.java.product

I hope someone can solve this.... 
Thanks in advance!

jantje, 2017-07-03:
The installer job starts when the plugin is started and will download new json files if you set so in windows->preferences->arduino->Third party index url's->update local json files
The installer job also starts when you install new platforms in windows->preferences->arduino->Platforms and boards
So if this doesn't go away after the plugin has been running for a while check these settings and restart.

As this is not a normal situation: Can you look into the error log for errors that may cause this situation and report them here?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-07-05:
Any update here?

FlorisVO, 2017-07-06:
Hi currently I'm very busy with a huge project, so i just switched to Linux (Ubuntu) (where everything works fine). When i have time (i hope next week), i will solve the issues.

crv38, 2017-07-22:
Hi,
I got exactly the same issue yesterday on Windows, not sure it's the same root cause, but in case.
I fix it by changing the path of my eclipse folder.
The problem was on startup of eclipse and this plugin, there was just a warning message that the path may be  too long for some platforms, but when closing this window, looks like some update process of
this plugin (update of the board manager) stop as well, do not complete, and remain in busy state.

After changing the eclipse folder location, could see the update completed at startup and use the plugin
like a charm.

Thanks by the way on the great job with this plugin.

jantje, 2017-07-22:
It indeed could be that a long path in windows makes the installer stop.

jantje, 2017-09-01:
Any update here? Can this be closed?

crv38, 2017-09-04:
no more issue on my side after changing the path, but if the issue of the long path name is confirmed, I would suggest to stop properly the processes to avoid such errors and blocking the system. Then could close it.

jantje, 2017-09-07:
@crv38 
I moved the test to the startup code (it was in the install code) and added a test for spaces in the workspace and the install path.
I think this is a good improvement to the inexperienced user experience.

closed 2017-09-07 by jantje

crv38, 2017-09-08:
thanks Jantje, that will for sure help to make it more robust and avoid misused by newbies (as me ;-) )